espressif/esp-bsp #73: i80_controller demo with esp_lcd_ra8875 - artifacts and shifted image

dmartauz, 2022-10-12:
I tried to adapt i80_controller demo to use esp_lcd_ra8875 component. It seems to work but there are artifacts and shifted image. It is most visible on LVGL FPS counter as it is redrawn periodically. Sometimes the artifacts are also visible outside of the FPS counter like on this picture:
[image]

LVGL FPS counter image shifts:
[image]
[image]

After pressing "Show again" touchscreen button the FPS counter is displayed correctly for 2-3 seconds, than it gets corrupted again.
Everything is displayed correctly after hard reset until the logo animation is finished therefore I don't think this is a PCLK frequency issue or FPS issue.

My code can be found here: 
[i80_controller_ra8875_20221012.zip](https://github.com/espressif/esp-bsp/files/9763240/i80_controller_ra8875_20221012.zip)

Hardware used:
- ESP32-S3-DEVKITC-1U-N8R8
- Waveshare 7inch Capacitive Touch LCD (C) 800x480

Software components:
- esp-idf 5.0
- lvgl 8.3.0
- esp_lcd_ra8875 1.0.0
- esp_lcd_touch 1.0.1
- esp_lcd_touch_gt911 1.0.1

espzav, 2022-10-14:
Hello @dmartauz,
Thank you for reporting this issue. I haven't got any problem on my example application with this display. I am sorry, I cannot try it now, I haven't got this display here. I will have it after a week at the latest. 
I checked your code and compared it with my example code. I have never tried to put the LVGL buffer to PSRAM with this display, do you have enabled or disabled macro `CONFIG_EXAMPLE_LCD_I80_COLOR_IN_PSRAM`?

dmartauz, 2022-10-15:
Hi @espzav, thank you for your reply and greetings to Brno from Trencin, SK ;-)
Results posted above are with CONFIG_EXAMPLE_LCD_I80_COLOR_IN_PSRAM enabled.
When disabled, the code fails at 
`assert failed: app_main i80_controller_example_main.c:257 (buf2)`
When line 257 `assert(buf2);` is commented out the code runs but with all visual artifacts as when running in PSRAM.

Could you please share your code so I can make a test on my HW?

espzav, 2022-10-17:
Hi @dmartauz, 
greetings to Trencin ;-) 
The internal RAM has limited size and the screen is big. I am using only one buffer (`buf2 = NULL`). If it isn't enough, you can decrease the size of the buffer for example from `100` to `70`.

My code is a complex example and it is not public now. I am working on some more components to help using LVGL. After that, the example will be published.

dmartauz, 2022-10-17:
@espzav I tried the single buffer solution, it now runs without assert error in the internal RAM but the artifacts are still present. I also tried different buffer sizes. Changing buffer size seems to have impact on how often and what artifacts are present but they are still there.
Could you please run my code once you have the display available?
When do you expect the public example to be available?

espzav, 2022-10-18:
Hi @dmartauz, I tried now your example on HW and it is working after some modifications, which I told you before. Here is working project. I disabled PSRAM and disabled second buffer. I don't see any artifacts. 
I will test it with PSRAM later. 
[i80_controller_ra8875.zip](https://github.com/espressif/esp-bsp/files/9810241/i80_controller_ra8875.zip)

espzav, 2022-10-19:
@dmartauz I know about pins connected to SPIRAM. My HW isn't good to use with octal PSRAM. I tried the display with PSRAM (QUAD) too and it is working. I can use a double buffer and more RAM. I used the example, which I shared with you before. 

If you use my example, have you got same issue?

dmartauz, 2022-10-20:
@espzav I tried your example with GPIO numbers modified to match my HW setup and the issue is still there. I even replaced GPIO19 and GPIO20 from my original configuration with GPIO2 and GPIO1 because according to the documentation GPIO19 and GPIO20 are used for USB-JTAG by default, but it did not help.

Tomorrow I will try to rewire everything to achieve the same GPIO configuration as you use with exception of GPIO37 which I cannot use because of octal SPIRAM.

Btw I noticed that you are not using CS signal -> `#define EXAMPLE_PIN_NUM_CS -1 `
How do you have CS pin of your LCD wired? Or did you left it floating?

espzav, 2022-10-20:
@dmartauz Here is my schematic. Hope, it helps. 
[Schematic_ESP32S3_7inch_tablet_2022-10-20.pdf](https://github.com/espressif/esp-bsp/files/9830477/Schematic_ESP32S3_7inch_tablet_2022-10-20.pdf)

dmartauz, 2022-10-20:
@espzav Once I grounded CS pin of the LCD (as you have in your schematic) and set `#define EXAMPLE_PIN_NUM_CS -1` everything is much more stable. I think that there are LCD CS pin timing issues in regards to timing requirements of RA8875.

espzav, 2022-10-25:
@dmartauz I like to hear, that it is better. Do you need the CS line or is it ok use it in ground?

dmartauz, 2022-10-25:
@espzav for me it is ok to ground the CS pin as I am not sharing the parallel bus between multiple devices. I propose to at least mention somewhere in the documentation that there are issues if CS pin is used with RA8875 and that it is recommended not to use it.

espzav, 2022-10-25:
@dmartauz yes, we will check the CS line and fix it or add the note to the documentation. I only needed to know, if we have some time or is it blocking for you.

dmartauz, 2022-10-25:
@espzav Now I only get artifacts when UART is connected to a PC but I think this is caused by some interference or ground loop. Right now my setup is on prototyping board. I will be designing proper PCB soon which should include isolated DC-DC converter so hopefully this issue will be resolved as well.

espzav, 2022-10-25:
@dmartauz I am using my board with display only on USB (connected to UART) and using console. I haven't got any artifacts there. Have you got connected RD to Vcc?

dmartauz, 2022-10-26:
@espzav The problem was I left RD pin of the LCD floating. When connected to Vcc everything runs ok. I even made short test with CS pin enabled (mapped to GPIO2) and there were no issues.

espzav, 2022-10-26:
@dmartauz Great! Thank you for info. I will add note about RD pin to Readme.