espressif/esp-bsp #87: error: unknown type name 'portMUX_TYPE'

Lzw655, 2022-10-28:
@espzav Hi, when I didn't declare `esp_lcd_touch.h` before `freertos/FreeRTOS.h` and `freertos/semphr.h`, I got an error as belows:
```
error: unknown type name 'portMUX_TYPE'
   69 |     portMUX_TYPE lock; /*!< Lock for read/write */
      |     ^~~~~~~~~~~~
```
I think maybe it's better to include `freertos/FreeRTOS.h` and `freertos/semphr.h` in the file `esp_lcd_touch.h`.

AI-Tipto, 2022-12-28:
same error!

/managed_components/espressif__esp_lcd_touch/include/esp_lcd_touch.h:69:5: error: 'portMUX_TYPE' does not name a type; did you mean 'portBASE_TYPE'?
   69 |     portMUX_TYPE lock; /*!< Lock for read/write */
      |     ^~~~~~~~~~~~
      |     portBASE_TYPE
In file included from /Users/asura/app/Esp32s3/LvglUI/managed_components/lvgl__lvgl/src/core/lv_obj.h:140,

espzav, 2023-01-03:
Fixed and merged on master.

closed 2023-01-03 by espzav