espressif/esp-bsp #95: assert failed: esp_lcd_touch_new_i2c_gt911 esp_lcd_touch_gt911.c:51 (io != NULL)

srufeng, 2022-12-06:
`esp_lcd_panel_io_i2c_config_t io_config = ESP_LCD_TOUCH_IO_I2C_GT911_CONFIG();

esp_lcd_touch_config_t tp_cfg = {
    .x_max = CONFIG_LCD_HRES,
    .y_max = CONFIG_LCD_VRES,
    .rst_gpio_num = -1,
    .int_gpio_num = -1,
    .levels = {
        .reset = 0,
        .interrupt = 0,
    },
    .flags = {
        .swap_xy = 0,
        .mirror_x = 0,
        .mirror_y = 0,
    },
};

esp_lcd_touch_handle_t tp;
esp_lcd_touch_new_i2c_gt911(io_handle, &tp_cfg, &tp);
`
使用上面的代码时，提示assert failed: esp_lcd_touch_new_i2c_gt911 esp_lcd_touch_gt911.c:51 (io != NULL)
idf版本用的是master

Lzw655, 2022-12-06:
@srufeng 你要先初始化 io_handle

Lzw655, 2022-12-06:
可以参考这里 https://github.com/espressif/esp-bsp/blob/57c5f3a2da79edf711d620438c997d1e035e88d7/esp-box/esp-box.c#L329-L347

srufeng, 2022-12-06:
> 可以参考这里
> 
> https://github.com/espressif/esp-bsp/blob/57c5f3a2da79edf711d620438c997d1e035e88d7/esp-box/esp-box.c#L329-L347

谢谢，我先试试

Lzw655, 2023-01-31:
@srufeng 若问题已解决，请关闭此 issue，谢谢！

closed 2023-02-01 by srufeng