espressif/esp-bsp #109: Adding ISR support to MPU6050 sensor module (FEATURE)

Frozen-Burrito, 2022-12-29:
I have used the MPU6050 component in a custom project, and it works great. However, I believe that there are a few capabilities of the sensor not yet implemented by the component, such as interrupt configuration and handling.

One of the features of my project involved waking up the ESP32 from deep sleep whenever the MPU6050 detected motion. Initially, I wanted to implement it using the ULP to communicate directly with the MPU6050 via I2C, but then I found out about "Motion detection" interrupts which can be enabled on the sensor to trigger a wakeup from EXT0 or EXT1. Based on the [MPU6050 datasheet](https://netsonic.fi/en/files/MPU6050-datasheet.pdf) and [this post from the Arduino Stack Exchange](https://arduino.stackexchange.com/questions/48424/how-to-generate-hardware-interrupt-in-mpu6050-to-wakeup-arduino-from-sleep-mode), I learned that registers `0x1F`, `0x37`, `0x38 `and `0x3A `can be used to configure interrupt behavior of the MPU6050. Yet, as far as I know, the component provided in this repo does not support reading or writing to these registers. In the end, I implemented an extension module to support `DATA_READY` and `MOTION` interrupts for my project. 

Does it make sense to enhance the MPU6050 component to make it capable of enabling interrupts in the MPU6050, registering ISR handlers and getting the interrupt status? If so, I would like to contribute to the component with the work that I've already implemented. I would greatly appreciate your opinion on this. 

(I am opening this issue since the CONTRIBUTING.md mentions that an issue should be created before contributing large(ish) scale changes)

tore-espressif, 2023-01-02:
Hello @Frozen-Burrito , such a new feature would be very useful!

We didn't implement it in the initial version, but the driver is indeed incomplete without it. You're more than welcome to raise a PR with your changes. Thank you

Frozen-Burrito, 2023-01-04:
Alright, I'll start working on integrating my changes on the driver. 

Originally, I was considering about implementing motion detect interrupts. Yet, while I was looking at the register map for the MPU6050, I noticed that revisions up to 4.0 do include information about motion detect registers and features, but revision 4.2 (the most recent revision I've found so far) doesn't event mention motion detect. I will do more research to find why this feature is not mentioned in the most recent revision. I will probably just include regular ISR capabilities in my first PR, without motion detect.

tore-espressif, 2023-01-08:
> I noticed that revisions up to 4.0 do include information about motion detect registers and features, but revision 4.2

Do you mean revision of the sensor or revision of the datasheet?

Frozen-Burrito, 2023-01-08:
I mean the revision of the register map. For example, the description of the INT_ENABLE register changes:

[image]
This is revision 4.0, which does mention the MOT_EN bit (it also has the "preliminary" watermark, which is what made me doubt in the first place)

[image]
Meanwhile, revision 4.2 does not mention the MOT_EN bit and instead marks bit 6 as "reserved".

I haven't yet found out if this difference between the register maps is present in different revisions of the actual chip. I have implemented motion detection on the MPU6050 I have and it works, so I have to do more research on that.

PR #117 (2023-01-16): Add MPU6050 ISR support (BSP-251)
# Change description

- Include new functions to setup and enable interrupts produced by the MPU6050 on different events (mainly `DATA_READY` at the moment, see notes below).
- Add new definitions for registers, types and enumerations used to configure and handle interrupts triggered by the MPU6050.
- Update component README.md with more information about the sensor, mention ISR support and add links to the register map and the sensors example.

# Notes
Since the driver does not yet support FiFo or I2C sensor slaves, interrupts by `FIFO_OVERFLOW` and `I2C_MST_ [... cut]
closed 2023-01-16 by tore-espressif