espressif/esp-bsp #137: Port expander TCA 9555/9539

hephaisto, 2023-02-25:
I'm planning to integrate either a TCA 9555 or a TCA 9539 (16bit port expander) into my project.
There currently is a driver for TCA9554, which is the 8-bit equivalent.
I would like to contribute the expansion to 16 bit, but I would like some feedback first on what way to go:
* build a separate component for 16bit (basically a carbon copy of everything except a few lines)
* use the same component
  * with multiple h/c-files and structs
  * with only one h/c file, but separate functions
  * extend functions to take a parameter (probably worst option because of backwards compatibility)
* just keep my stuff locally and don't push to this repo

tore-espressif, 2023-03-08:
@Lzw655 could you please help with this issue?

tore-espressif, 2023-03-08:
@Lzw655 Agreed, I just wanted to hear your opinion as you are the author of `io_expander`

Lzw655, 2023-03-08:
@hephaisto Please build a separate component for TCA9555 (or TCA9539) and push to this repo. And If there are some bugs or needed features in `esp_io_expander` component, please let me know. Thanks for your contribution.

hephaisto, 2023-04-10:
@Lzw655 TCA9539 and TCA9555 only differ in the I2C address (from the software perspective), so I think it would be easier to handle if the actual code is shared between those two. The address needs to get passed in anyway:
`ZCA9539_ADDRESS_00 = 0b1110100,TCA9555_ADDRESS_000 = 0b0100000,`
Not sure how to name the component then. "tca9539_tca9555"? "tca95xx" would be too general, since there are also e.g. "tca9554" port expanders.

Lzw655, 2023-04-11:
@hephaisto You are right. I checked the TCA95xx series datasheets and compared them as shown [here](https://www.ti.com/interface/i2c/general-purpose-ios-gpios/overview.html). The difference between them is generally the address and the number of IO bits. So I think you can name them "esp_io_expander_tca95xx_16bit" and list all supported chips in README. 
Hi @tore-espressif, I think maybe I should create a new component named "esp_io_expander_tca95xx_8bit". What do you think?

PR #174 (2023-05-31): TCA95xx_16bit (BSP-326)
closes #137

# ESP-BSP Pull Request checklist

> Note: For new BSPs create a PR with this [link](https://github.com/espressif/esp-bsp/compare/main...my-branch?quick_pull=1&template=pr_template_bsp.md).

- [x] Version of modified component bumped
- [ ] CI passing

# Change description

Added TCA9539 and TCA9555 16bit port expanders.
closed 2023-05-31 by tore-espressif