espressif/esp-bsp #138: component esp_lvgl_port

espressif2022, 2023-03-06:
hello @espzav 
The component has been modified recently, and a navigation button has been added here, but I see that the registered event is LV_INDEV_TYPE_ENCODER.
I want to know, is the design here according to the requirements of ESP32-S3-BOX-Lite?
Because I want to add an enode knob event, so I want to confirm here.

espzav, 2023-03-06:
Hello @espressif2022,
yes, the latest changes in LVGL port are for ESP32-S3-BOX-Lite. 
The encoder API will be very nice! I think, it will be simillar. But Encoder API in LVGL can take count of steps. If you need any help, feel free to contact me.

tore-espressif, 2023-07-03:
@espzav @espressif2022 Can we consider this issue as resolved?

espressif2022, 2023-07-03:
> @espzav @espressif2022 Can we consider this issue as resolved?

Yes, it support both encoder and button now.

closed 2023-07-03 by espressif2022