espressif/esp-bsp #151: esp-box v1.1.0 issues and requirement

donghengqaz, 2023-04-03:
I use esp-idf v4.4.x, so have to use esp-box v1.1.0 or older versions, but this version issue. I refer to esp-box old code and add semaphore to `lvgl_port_flush_ready` and `lvgl_port_flush_callback` and it works well. Besides this I would like to make LVGL task's stack size configurable.
I have patch, how could I add PR to patch an esp-box of older version?

donghengqaz, 2023-04-04:
My patch is attached, PTAL.

```patch
diff --git a/esp-box/esp-box.c b/esp-box/esp-box.c
index 8cbaf93..85d1502 100644
--- a/esp-box/esp-box.c
+++ b/esp-box/esp-box.c
@@ -24,6 +24,7 @@ static lv_disp_draw_buf_t disp_buf; // Contains internal graphic buffer(s) calle
 static lv_disp_drv_t disp_drv;      // Contains callback functions
 static esp_lcd_touch_handle_t tp;   // LCD touch handle
 static SemaphoreHandle_t lvgl_mux;  // LVGL mutex
+static SemaphoreHandle_t lvgl_sem;
 
 void bsp_i2c_init(void)
 {
@@ -123,6 +124,7 @@ static bool lvgl_port_flush_ready(esp_lcd_panel_io_handle_t panel_io, esp_lcd_pa
 {
     lv_disp_drv_t *disp_driver = (lv_disp_drv_t *)user_ctx;
     lv_disp_flush_ready(disp_driver);
+    xSemaphoreGive(lvgl_sem);
     return false;
 }
 
@@ -133,6 +135,9 @@ static void lvgl_port_flush_callback(lv_disp_drv_t *drv, const lv_area_t *area,
     const int offsetx2 = area->x2;
     const int offsety1 = area->y1;
     const int offsety2 = area->y2;
+
+    xSemaphoreTake(lvgl_sem, portMAX_DELAY);
+
     // copy a buffer's content to a specific area of the display
     esp_lcd_panel_draw_bitmap(panel_handle, offsetx1, offsety1, offsetx2 + 1, offsety2 + 1, color_map);
 }
@@ -222,6 +227,9 @@ static lv_disp_t *lvgl_port_display_init(void)
     };
     ESP_ERROR_CHECK(spi_bus_initialize(BSP_LCD_SPI_NUM, &buscfg, SPI_DMA_CH_AUTO));
 
+    vSemaphoreCreateBinary(lvgl_sem);
+    assert(lvgl_sem);
+
     ESP_LOGD(TAG, "Install panel IO");
     esp_lcd_panel_io_handle_t io_handle = NULL;
     const esp_lcd_panel_io_spi_config_t io_config = {
@@ -369,9 +377,9 @@ lv_disp_t *bsp_display_start(void)
     lv_disp_t *disp = lvgl_port_display_init();
     lvgl_port_indev_init();
     lvgl_port_tick_init();
-    lvgl_mux = xSemaphoreCreateMutex();
+    lvgl_mux = xSemaphoreCreateRecursiveMutex();
     assert(lvgl_mux);
-    xTaskCreate(lvgl_port_task, "LVGL task", 4096, NULL, CONFIG_BSP_DISPLAY_LVGL_TASK_PRIORITY, NULL);
+    xTaskCreate(lvgl_port_task, "LVGL task", 4096 * 4, NULL, CONFIG_BSP_DISPLAY_LVGL_TASK_PRIORITY, NULL);
     return disp;
 }
 
-- 
2.25.1

```

espressif2022, 2023-04-04:
@donghengqaz 
`lv_disp_flush_ready` deals with the flag of `draw_buf->flushing`. 
Before the flushing flag is cleared, func `draw_buf_flush` will wait and won't execute `draw_bitmap`. 
I think it's not necessary to add semaphore here.

donghengqaz, 2023-04-04:
@espressif2022 Without this semaphore I test with esp-idf v4.4.3, and the function `draw_buf_flush` blocks deadly, you can try this esp-box v1.1.0 with related [example](https://github.com/espressif/esp-box/tree/master/examples/lv_demos), just modify the [idf_component.yml](https://github.com/espressif/esp-box/blob/master/examples/lv_demos/main/idf_component.yml) to use esp-box v1.1.0.

tore-espressif, 2023-04-05:
Hello @donghengqaz , thank you for the proposed fix! 

I'll get my esp-box next week and try it

donghengqaz, 2023-04-20:
Hi @tore-espressif Is there any in-progress about this ?

tore-espressif, 2023-04-20:
Hello @donghengqaz I'm sorry for the delay. 

I prepared #162 , that updates esp-box BSP to our `esp_lvgl_port` implementation, that already solver the mutex issues you are using. Could you please have a look?

If it is too big change for your, we can accept the change as you proposed it

donghengqaz, 2023-04-21:
> Hello @donghengqaz I'm sorry for the delay.
> 
> I prepared #162 , that updates esp-box BSP to our `esp_lvgl_port` implementation, that already solver the mutex issues you are using. Could you please have a look?
> 
> If it is too big change for your, we can accept the change as you proposed it

Hi @tore-espressif, I see this PR use an extra package, my purpose is to use esp-idf v4.4.3, and remove BSP source code from my project. So the big change is also OK for me. When this PR is done, I will integrate and test.

tore-espressif, 2023-05-03:
Hello @donghengqaz , I'll merge #162 and release esp-box v1.2.0.

FYI, the plan for future is to make all latest BSP version backward compatible with IDFv4.4.x

donghengqaz, 2023-05-04:
> Hello @donghengqaz , I'll merge #162 and release esp-box v1.2.0.
> 
> FYI, the plan for future is to make all latest BSP version backward compatible with IDFv4.4.x

It's great, I will update to this one.

donghengqaz, 2023-05-04:
> > Hello @donghengqaz , I'll merge #162 and release esp-box v1.2.0.
> > FYI, the plan for future is to make all latest BSP version backward compatible with IDFv4.4.x
> 
> It's great, I will update to this one.

Hi @tore-espressif I integrate this branch to my repo, but the LVGL flushing also blocks deadly, then I add semaphore in flush and flush_ready, it works well...

closed 2023-07-26 by espzav