espressif/esp-bsp #152: bsp/esp-bsp.h: No such file or directory

13koma, 2023-04-03:
Trying to use driver to run on esp32 in esp-idf env in platformio. I've added in libs folders from your repository and trying to run example code. When I build it i got error when including "bsp/esp-bsp.h", but there is no such folder in your project. How can I fix it

tore-espressif, 2023-04-05:
Hello @13koma , 
I'll try to help, but I don't have vast experience with platformio. Components from this repository are distributed by [IDF component manager](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-component-manager.html).

What is your project structure? How did you add the libs to your project?

Could you please try running `pip install idf-component-manager` and then building the project again?

tore-espressif, 2023-04-19:
Can I ask for any update about this issue?

tore-espressif, 2023-04-26:
@13koma could you please check that you don't have IDF component manager disabled? https://github.com/espressif/idf-component-manager/blob/main/README.md#disabling-the-component-manager