espressif/esp-bsp #154: Component esp_lvgl_port, can SIMD instructions be used to speed up LVGL swapping bytes?

modi12jin, 2023-04-05:
I am more curious whether SIMD instructions can speed up the LVGL drawing UI process?

igrr, 2023-04-05:
They definitely can be used to speed up color fill and similar operations. LVGL project is implementing the parallel drawing framework, which will allow the port layer to define handlers for specific drawing tasks ([see this discussion](https://github.com/lvgl/lvgl/issues/3700#issuecomment-1450921020)). According to @kisvegabor the release of LVGL with these changes is planned in autumn (https://github.com/lvgl/lvgl/issues/4011#issuecomment-1491501750). Then we will implement some of the drawing tasks using SIMD instructions.

kisvegabor, 2023-04-05:
I confirm it! :slightly_smiling_face:

kisvegabor, 2024-03-14:
Since v9 LVGL has some built-in ASM accelerators. See an example for the usage [here](https://github.com/lvgl/lvgl/blob/master/src/draw/sw/blend/lv_draw_sw_blend_to_rgb565.c).

Somewhere Espressif described that maintaining the ESP-ASM on ESP's side would be more beneficial, but I don't know if there is a final decision about it. Having them in LVGL could be beneficial to avoid compatibility issues though.

closed 2024-11-05 by modi12jin