espressif/esp-bsp #157: Mute function for ES7210

christian-nils, 2023-04-09:
Hello,

I am fiddling with my new ESP32-S3-BOX, and I wanted to mute the microphone via software (instead of cutting power to the ES7210 chip) as the physical mute button does. I could not find the implementation of the mute function in your ESP component (using the registries 0x15 and 0x16).

Do you plan to add this feature in the near future? Thank you,

Greetings,

tore-espressif, 2023-04-12:
Hello @christian-nils , thank you for the issue.
We can add the mute function to es7210 component. However, the preferred way is to use latest [esp_codec_dev](https://components.espressif.com/components/espressif/esp_codec_dev) component that is shipped with latest [ESP-BOX BSP](https://github.com/espressif/esp-bsp/blob/master/esp-box/idf_component.yml#L17). This component already contains the mute feature.

We plan to discontinue the `es7210` component and use `esp_codec_dev` in all future BSP's and examples. Is this solution acceptable for you?

christian-nils, 2023-04-12:
Hi @tore-espressif!

Thanks for your quick and informative response. It sounds like your proposed approach is the most reasonable and sustainable one. I will have a try. Thank you, have a great day.

closed 2023-04-12 by christian-nils