espressif/esp-bsp #172: Touch rotated but not display Waveshare 7inch hardware rotation

hayschan, 2023-05-15:
## Machine

Computer: Windows 11
MCU: ESP32-S3 N8R8
LVGL: v8.3.3

## Symptom

I would like to rotate the display by 180 degree.

After setting `disp_drv.rotated = LV_DISP_ROT_180;` , only touch input is rotated. The UI is not rotated.

## Extra info

By setting the `rotated` flag, the screen could perform hardware rotation. Hardware rotation has better performance. 

According to LVGL documentation:

> LVGL supports rotation of the display in 90 degree increments. You can select whether you'd like software rotation or hardware rotation.  
If you select software rotation (`sw_rotate` flag set to 1), LVGL will perform the rotation for you. Your driver can and should assume that the screen width and height have not changed. Simply flush pixels to the display as normal. Software rotation requires no additional logic in your `flush_cb` callback.  
There is a noticeable amount of overhead to performing rotation in software, which is why hardware rotation is also available. In this mode, LVGL draws into the buffer as though your screen now has the width and height inverted. You are responsible for rotating the provided pixels yourself.  
The default rotation of your display when it is initialized can be set using the `rotated` flag. The available options are `LV_DISP_ROT_NONE`, `LV_DISP_ROT_90`, `LV_DISP_ROT_180`, or `LV_DISP_ROT_270`. The rotation values are relative to how you would rotate the physical display in the clockwise direction. Thus, `LV_DISP_ROT_90` means you rotate the hardware 90 degrees clockwise, and the display rotates 90 degrees counterclockwise to compensate.

I implemented by adding this line:
```c
disp_drv.rotated = LV_DISP_ROT_180;
```

The rotated algorithm is in the following lines:

```c
static void lvgl_port_update_callback(lv_disp_drv_t *drv)
{
    esp_lcd_panel_handle_t panel_handle = (esp_lcd_panel_handle_t) drv->user_data;

    switch (drv->rotated) {
    case LV_DISP_ROT_NONE:
        // Rotate LCD display
        esp_lcd_panel_swap_xy(panel_handle, false);
        esp_lcd_panel_mirror(panel_handle, false, false);
        // Rotate LCD touch
        esp_lcd_touch_set_mirror_y(tp, false);
        esp_lcd_touch_set_mirror_x(tp, false);
        break;
    case LV_DISP_ROT_90:
        // Rotate LCD display
        esp_lcd_panel_swap_xy(panel_handle, true);
        esp_lcd_panel_mirror(panel_handle, false, true);
        // Rotate LCD touch
        esp_lcd_touch_set_mirror_y(tp, false);
        esp_lcd_touch_set_mirror_x(tp, false);
        break;
    case LV_DISP_ROT_180:
        // Rotate LCD display
        esp_lcd_panel_swap_xy(panel_handle, false);
        esp_lcd_panel_mirror(panel_handle, true, true);
        // Rotate LCD touch
        esp_lcd_touch_set_mirror_y(tp, false);
        esp_lcd_touch_set_mirror_x(tp, false);
        break;
    case LV_DISP_ROT_270:
        // Rotate LCD display
        esp_lcd_panel_swap_xy(panel_handle, true);
        esp_lcd_panel_mirror(panel_handle, true, false);
        // Rotate LCD touch
        esp_lcd_touch_set_mirror_y(tp, false);
        esp_lcd_touch_set_mirror_x(tp, false);
        break;
    }
}
```

hayschan, 2023-05-15:
### `lvgl_port_update_callback` not executed at all

Weirdly, it seems that the function `lvgl_port_update_callback(lv_disp_drv_t *drv)` is not executed at all. I added `ESP_LOG` commend to indicate excecution, but none of them prints out anything at all.

```c
static void lvgl_port_update_callback(lv_disp_drv_t *drv)
{
    ESP_LOGI("Rotation", "Enter lvgl_port_update_callback()")
    esp_lcd_panel_handle_t panel_handle = (esp_lcd_panel_handle_t) drv->user_data;

    switch (drv->rotated) {
    case LV_DISP_ROT_NONE:
        // Rotate LCD display
        esp_lcd_panel_swap_xy(panel_handle, false);
        esp_lcd_panel_mirror(panel_handle, false, false);
        // Rotate LCD touch
        esp_lcd_touch_set_mirror_y(tp, false);
        esp_lcd_touch_set_mirror_x(tp, false);
        break;
    case LV_DISP_ROT_90:
        // Rotate LCD display
        esp_lcd_panel_swap_xy(panel_handle, true);
        esp_lcd_panel_mirror(panel_handle, false, true);
        // Rotate LCD touch
        esp_lcd_touch_set_mirror_y(tp, false);
        esp_lcd_touch_set_mirror_x(tp, false);
        break;
    case LV_DISP_ROT_180:
        // Rotate LCD display
        esp_lcd_panel_swap_xy(panel_handle, false);
        esp_lcd_panel_mirror(panel_handle, true, true);
        // Rotate LCD touch
        esp_lcd_touch_set_mirror_y(tp, false);
        esp_lcd_touch_set_mirror_x(tp, false);
        ESP_LOGI("Rotation", "Enter LV_DISP_ROT_180")
        break;
    case LV_DISP_ROT_270:
        // Rotate LCD display
        esp_lcd_panel_swap_xy(panel_handle, true);
        esp_lcd_panel_mirror(panel_handle, true, false);
        // Rotate LCD touch
        esp_lcd_touch_set_mirror_y(tp, false);
        esp_lcd_touch_set_mirror_x(tp, false);
        break;
    }
}
```

hayschan, 2023-05-15:
> Hello @hayschan, please, could you try call this function `bsp_display_rotate(display, LV_DISP_ROT_180);` after the display was initialized?

I could confirm that `bsp_display_rotate(display, LV_DISP_ROT_180);` works flawlessly in terms of screen rotation. However, is this hardware rotation or software rotation? I am worried about performance issue.

> Note: My recommendation is to use the `esp_lvgl_port` component (https://github.com/espressif/esp-bsp/tree/master/components/esp_lvgl_port) which initialize LVGL and rotation and after that you can use function `bsp_display_rotate`. It is still HW rotation. It should work, I tested it some months ago.

Thanks. I'll take a look and try this ESP component.

espzav, 2023-05-15:
> I could confirm that `bsp_display_rotate(display, LV_DISP_ROT_180);` works flawlessly in terms of screen rotation. However, is this hardware rotation or software rotation? I am worried about performance issue.
It is HW rotation. It is using the `lvgl_port_update_callback` for rotate screen by HW.

hayschan, 2023-05-15:
> Thanks. I'll take a look and try this ESP component.

Just found out that `lvgl/lvgl` is replaced by `esp_lvgl_port` in a commit three months ago https://github.com/espressif/esp-bsp/commit/1a088da5ba927ce5d7afa30ee282d8888c0087de

I should have updated my Custom Waveshare 7inch component in the first place. 😭

hayschan, 2023-05-18:
> @hayschan the BSP component for 7inch screen is here: https://github.com/espressif/esp-bsp/tree/master/SquareLine/boards/custom_waveshare_7inch/components it is using the `esp_lvgl_port`

I have changed to the latest Waveshare 7inch screen component which is `v1.0.1`. 

## Symptom

### Before

Before the change, the UI works normally as expected. 

### After

Right after the change, there are **snowflakes** on the screen. At the very top, there is a few lines of pixel that display normally. But other parts of the screen remains white and flickering.

[image]

## Error log

I have also check the serial monitor.

```
I (1447) LVGL: Starting LVGL task
I (1447) gpio: GPIO[9]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 

I (1477) PWM_var: PWM_var = 0.00%
I (1487) gpio: GPIO[11]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1517) gpio: GPIO[10]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2 
I (1517) GT911: TouchPad_ID:0x39,0x31,0x31
I (1517) GT911: TouchPad_Config_Version:91
I (1787) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1787) ESP-EXAMPLE: Example initialization done.
E (6957) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (6957) task_wdt:  - IDLE (CPU 0)
E (6957) task_wdt: Tasks currently running:
E (6957) task_wdt: CPU 0: LVGL task
E (6957) task_wdt: CPU 1: IDLE
E (6957) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x42045D0E:0x3FC94600 0x42045EC4:0x3FC94620 0x40377139:0x3FC94640 0x4201BABC:0x3FCF7BA0 0x4201C0C5:0x3FCF7BC0 0x4201C21B:0x3FCF7C30 0x4201C328:0x3FCF7C60 0x4201C4D0:0x3FCF7C80 0x4202C041:0x3FCF7CB0 0x4202C0EF:0x3FCF7CD0 0x42010FDC:0x3FCF7CF0 0x4037E955:0x3FCF7D10
0x42045d0e: task_wdt_timeout_handling at E:/esp_idf_5.0/Espressif/frameworks/esp-idf-v5.0/components/esp_system/task_wdt.c:486 (discriminator 3)

0x42045ec4: task_wdt_isr at E:/esp_idf_5.0/Espressif/frameworks/esp-idf-v5.0/components/esp_system/task_wdt.c:586

0x40377139: _xt_lowint1 at E:/esp_idf_5.0/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1117

0x4201babc: draw_buf_flush at E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/core/lv_refr.c:1198
0x4201c0c5: refr_area_part at E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/core/lv_refr.c:712

0x4201c21b: refr_area at E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/core/lv_refr.c:594 (discriminator 2)

0x4201c328: refr_invalid_areas at E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/core/lv_refr.c:532

0x4201c4d0: _lv_disp_refr_timer at E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/core/lv_refr.c:324

0x4202c041: lv_timer_exec at E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/misc/lv_timer.c:313 
(discriminator 2)

0x4202c0ef: lv_timer_handler at E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/misc/lv_timer.c:109

0x42010fdc: lvgl_port_task at E:/esp_fryer_highend/ESP-fryer-highend/managed_components/espressif__esp_lvgl_port/esp_lvgl_port.c:441

0x4037e955: vPortTaskWrapper at E:/esp_idf_5.0/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151

E (11957) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (11957) task_wdt:  - IDLE (CPU 0)
E (11957) task_wdt: Tasks currently running:
E (11957) task_wdt: CPU 0: LVGL task
E (11957) task_wdt: CPU 1: IDLE
E (11957) task_wdt: Print CPU 0 (current core) backtrace

[... 78 lines cut ...]

0x4201c4d0: _lv_disp_refr_timer at E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/core/lv_refr.c:324

0x4202c041: lv_timer_exec at E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/misc/lv_timer.c:313 
(discriminator 2)

0x4202c0ef: lv_timer_handler at E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/misc/lv_timer.c:109

0x42010fdc: lvgl_port_task at E:/esp_fryer_highend/ESP-fryer-highend/managed_components/espressif__esp_lvgl_port/esp_lvgl_port.c:441

0x4037e955: vPortTaskWrapper at E:/esp_idf_5.0/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151

ClearCommError failed (PermissionError(13, '???????????????????????????', None, 22))
Waiting for the device to reconnect...
I (865) esp_image: segment 1: paddr=00418ac0 vaddr=3fc93700 size=02c30h ( 11312) load
I (868) esp_image: segment 2: paddr=0041b6f8 vaddr=40374000 size=04920h ( 18720) load
I (875) esp_image: segment 3: paddr=00420020 vaddr=42000020 size=65aach (416428) map
I (954) esp_image: segment 4: paddr=00485ad4 vaddr=40378920 size=0ad38h ( 44344) load
I (965) esp_image: segment 5: paddr=00490814 vaddr=50000000 size=00010h (    16) load
I (971) boot: Loaded app from partition at offset 0x10000
I (972) boot: Disabling RNG early entropy source...
I (985) cpu_start: Pro cpu up.
I (985) cpu_start: Starting app cpu, entry point is 0x40375408
0x40375408: call_start_cpu1 at E:/esp_idf_5.0/Espressif/frameworks/esp-idf-v5.0/components/esp_system/port/cpu_start.c:142
```

closed 2023-05-19 by espzav