espressif/esp-bsp #173: Screen shows Snowflakes but the first few lines

hayschan, 2023-05-19:
## TL;DR

TL;DR: After updating to the latest Waveshare 7-inch screen component, the UI started displaying snowflakes and flickering. The error log indicates a task watchdog timeout, with the LVGL task failing to reset the watchdog in time.

## Version

Waveshare 7inch screen component: `v1.0.1`

## Symptoms

### Before

When I was using Waveshare 7inch screen component `v1.0.0`, the UI works normally as expected. 

### After

Right after the change to `v1.0.1`, there are **snowflakes** on the screen. At the very top, there is a few lines of pixel that display normally. But other parts of the screen remains white and flickering.

[image]

The UI does not respond to touch screen either.

## Code

`main.c`
```c
/*
 * SPDX-FileCopyrightText: 2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 */

#include "esp_log.h"
#include "bsp/ws_7inch.h"
#include "lvgl.h"
#include "ui/ui.h"

#define TAG "ESP-EXAMPLE"

/*******************************************************************************
* Private functions
*******************************************************************************/

// *INDENT-OFF*
void app_lvgl_display(void)
{
    bsp_display_lock(0);

    ui_init();

    bsp_display_unlock();
}

void app_main(void)
{
    /* Initialize I2C (for touch) */
    bsp_i2c_init();

    /* Initialize display and LVGL */
    bsp_display_start();

    /* Turn on display backlight */
    bsp_display_backlight_on();

    /* Add and show objects on display */
    app_lvgl_display();

    ESP_LOGI(TAG, "Example initialization done.");
}
// *INDENT-ON*
[... 133 lines cut ...]

0x4201c4d0: _lv_disp_refr_timer at E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/core/lv_refr.c:324

0x4202c041: lv_timer_exec at E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/misc/lv_timer.c:313 
(discriminator 2)

0x4202c0ef: lv_timer_handler at E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/misc/lv_timer.c:109

0x42010fdc: lvgl_port_task at E:/esp_fryer_highend/ESP-fryer-highend/managed_components/espressif__esp_lvgl_port/esp_lvgl_port.c:441

0x4037e955: vPortTaskWrapper at E:/esp_idf_5.0/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151

ClearCommError failed (PermissionError(13, '???????????????????????????', None, 22))
Waiting for the device to reconnect...
I (865) esp_image: segment 1: paddr=00418ac0 vaddr=3fc93700 size=02c30h ( 11312) load
I (868) esp_image: segment 2: paddr=0041b6f8 vaddr=40374000 size=04920h ( 18720) load
I (875) esp_image: segment 3: paddr=00420020 vaddr=42000020 size=65aach (416428) map
I (954) esp_image: segment 4: paddr=00485ad4 vaddr=40378920 size=0ad38h ( 44344) load
I (965) esp_image: segment 5: paddr=00490814 vaddr=50000000 size=00010h (    16) load
I (971) boot: Loaded app from partition at offset 0x10000
I (972) boot: Disabling RNG early entropy source...
I (985) cpu_start: Pro cpu up.
I (985) cpu_start: Starting app cpu, entry point is 0x40375408
0x40375408: call_start_cpu1 at E:/esp_idf_5.0/Espressif/frameworks/esp-idf-v5.0/components/esp_system/port/cpu_start.c:142
```

hayschan, 2023-05-19:
## Summary

Based on the provided log, the LVGL task watchdog reset is being triggered. Here's the relevant information from the log:

```python
E (6612) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (6612) task_wdt:  - IDLE (CPU 0)
E (6612) task_wdt: Tasks currently running:
E (6612) task_wdt: CPU 0: LVGL task
E (6612) task_wdt: CPU 1: IDLE
```

According to the log, the LVGL task on CPU 0 is not resetting the watchdog in time. The watchdog is a mechanism that ensures tasks are executing within the expected timeframe. When a task fails to reset the watchdog, it indicates that the task is taking too long to complete, which can be an indication of a potential issue in the code.

The backtrace information provides additional details about the code execution at the time the watchdog reset occurred:

```python
Backtrace: 0x42043A8A:0x3FC93EF0 0x42043C40:0x3FC93F10 0x40377101:0x3FC93F30 0x4201AA86:0x3FCEADD0 0x4201B0BD:0x3FCEADF0 0x4201B213:0x3FCEAE60 0x4201B320:0x3FCEAE90 0x4201B4EC:0x3FCEAEB0 0x4202B961:0x3FCEAEE0 0x4202BA1F:0x3FCEAF00 0x42010184:0x3FCEAF20 0x4037E29D:0x3FCEAF40
```

The backtrace shows the sequence of function calls leading up to the watchdog reset. Here are some of the relevant functions in the backtrace related to LVGL:

- `draw_buf_flush` at `E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/core/lv_refr.c:1198`
- `refr_area_part` at `E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/core/lv_refr.c:712`
- `refr_area` at `E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/core/lv_refr.c:594`
- `refr_invalid_areas` at `E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/core/lv_refr.c:532`
- `_lv_disp_refr_timer` at `E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/core/lv_refr.c:324`
- `lv_timer_exec` at `E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/misc/lv_timer.c:313`
- `lv_timer_handler` at `E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/misc/lv_timer.c:109`
- `lvgl_port_task` at `E:/esp_fryer_highend/ESP-fryer-highend/managed_components/espressif__esp_lvgl_port/esp_lvgl_port.c:441`

These functions are part of the LVGL library and are involved in the rendering and refreshing of the LVGL UI. The issue may be related to the LVGL task being overloaded or not completing its tasks within the allocated time.

To identify the specific part of the code causing the LVGL task watchdog reset, you can review the relevant LVGL functions mentioned in the backtrace. Look for any potential bottlenecks or long-running operations within those functions that could be causing the task to take too long.

Additionally, enable LVGL debug logging to get more detailed information about the execution flow and any potential errors or warnings that may help pinpoint the problem area.

Keep in mind that troubleshooting complex issues like this often requires a deep understanding of the codebase and the specific hardware setup. If you're unable to identify the cause of the issue, consider reaching out to the LVGL community or the developers of the ESP-fryer-highend project for further assistance.

## Error Log

```python
I (0) cpu_start: App cpu up.
I (996) cpu_start: Pro cpu start user code
I (997) cpu_start: cpu freq: 160000000 Hz
I (997) cpu_start: Application information:
I (1000) cpu_start: Project name:     ESP-fryer-highend
I (1005) cpu_start: App version:      v1.2.0-12-g939ccbb-dirty
I (1012) cpu_start: Compile time:     May 19 2023 16:18:03
I (1018) cpu_start: ELF file SHA256:  e6573f853a2b17f6...
I (1024) cpu_start: ESP-IDF:          v5.0-dirty
I (1030) heap_init: Initializing. RAM available for dynamic allocation:
I (1037) heap_init: At 3FC97AA0 len 00051C70 (327 KiB): D/IRAM
I (1043) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (1050) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (1056) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (1064) spi_flash: detected chip: generic
I (1068) spi_flash: flash io: dio
W (1072) pcnt(legacy): legacy driver is deprecated, please migrate to `driver/pulse_cnt.h`
W (1081) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
I (1089) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (10) LVGL: Starting LVGL task
I (1112) gpio: GPIO[9]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1122) gpio: GPIO[11]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1152) gpio: GPIO[10]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2 
I (1152) GT911: TouchPad_ID:0x39,0x31,0x31
I (1152) GT911: TouchPad_Config_Version:91
I (1422) ESP-EXAMPLE: Example initialization done.
E (6612) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
[... 6 lines cut ...]
Backtrace: 0x42043A8A:0x3FC93EF0 0x42043C40:0x3FC93F10 0x40377101:0x3FC93F30 0x4201AA86:0x3FCEADD0 0x4201B0BD:0x3FCEADF0 0x4201B213:0x3FCEAE60 0x4201B320:0x3FCEAE90 0x4201B4EC:0x3FCEAEB0 0x4202B961:0x3FCEAEE0 0x4202BA1F:0x3FCEAF00 0x42010184:0x3FCEAF20 0x4037E29D:0x3FCEAF40
0x42043a8a: task_wdt_timeout_handling at E:/esp_idf_5.0/Espressif/frameworks/esp-idf-v5.0/components/esp_system/task_wdt.c:486 (discriminator 3)

[... cut ...]
0x4202b961: lv_timer_exec at E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/misc/lv_timer.c:313 
(discriminator 2)

0x4202ba1f: lv_timer_handler at E:/esp_fryer_highend/ESP-fryer-highend/managed_components/lvgl__lvgl/src/misc/lv_timer.c:109

0x42010184: lvgl_port_task at E:/esp_fryer_highend/ESP-fryer-highend/managed_components/espressif__esp_lvgl_port/esp_lvgl_port.c:441

0x4037e29d: vPortTaskWrapper at E:/esp_idf_5.0/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151
```

espzav, 2023-05-19:
Hi @hayschan, I tried today the latest version of the component `ws_7inch` and everything is ok. 
ws_7inch version: 1.0.1
LVGL port version: 1.1.0
GT911 version: 1.0.7~1
LCD touch version: 1.0.4
RA8875 version: 1.0.0

hayschan, 2023-05-19:
I was wondering if you would be open to allowing me to give it a try as well? I understand that you've already tested it and I would be grateful for the opportunity to experiment with it on my own machine. 

Please let me know if it's okay with you and if there are any specific instructions or considerations I should keep in mind while trying out the code. I truly appreciate your time and willingness to share your work.

Thank you in advance! 😃

hayschan, 2023-05-19:
> ws_7inch version: 1.0.1
> LVGL port version: 1.1.0
> GT911 version: 1.0.7~1
> LCD touch version: 1.0.4
> RA8875 version: 1.0.0

I just checked. My versions of the components are excatly the same as you.

espzav, 2023-05-25:
HI @hayschan,
I am sorry for the late reply. Here is my testing project from SquareLine with latest `ws_7inch` component.
[test_800.zip](https://github.com/espressif/esp-bsp/files/11563983/test_800.zip)

VojtechBartoska, 2023-06-06:
Any updates on this @hayschan? Thanks

hayschan, 2023-06-09:
Sorry for the late reply. I will try the code tomorrow and update the findings.

VojtechBartoska, 2023-06-13:
@hayschan Thanks a lot, looking forward for your feedback

hayschan, 2023-06-14:
> I am sorry for the late reply. Here is my testing project from SquareLine with latest `ws_7inch` component.
[test_800.zip](https://github.com/espressif/esp-bsp/files/11563983/test_800.zip)

I have tried running `test_800.zip`. Runs flawlessly. 👍🏼 

I should now dig into my code. 

Thanks a lot for the help. ❤️

VojtechBartoska, 2023-06-20:
@hayschan Great, closing this issue as solved.

closed 2023-06-20 by VojtechBartoska