espressif/esp-bsp #178: CST816 Touch driver support (IDFGH-10255)

krupis, 2023-06-07:


krupis, 2023-06-13:
How is it going regarding the CST816 touch driver? Has there been any updates ?

Highly appreciate!

Lzw655, 2023-06-13:
@krupis Sorry for being late. I have an LCD with CST816 touch in the home and I'll use it to test the driver tomorrow. Please wait another day.

krupis, 2023-06-13:
> i2c_cmd_handle_t cmd;
>     for (int i = 0; i < 0x7f; i++) {
>         cmd = i2c_cmd_link_create();
>         i2c_master_start(cmd);
>         i2c_master_write_byte(cmd, (i << 1) | I2C_MASTER_WRITE, true);
>         i2c_master_stop(cmd);
>         if (i2c_master_cmd_begin(BSP_I2C_NUM, cmd, portMAX_DELAY) == ESP_OK) {
>             ESP_LOGW(TAG, "%02X", i);
>         }
>         i2c_cmd_link_delete(cmd);
>     }

Not a problem at all.

Do you have a T-Display-S3 from Lilygo or some custom board?

I have tried to run the command that you have suggested. I have a public repository for testing CST816 that here:
https://github.com/krupis/T-Display-S3-esp-idf

I have a lvgl_setup function that is being called in main.c to initialise the whole display:

```
void lvgl_setup()
{

    gpio_config_t pwr_gpio_config =
        {
            .mode = GPIO_MODE_OUTPUT,
            .pin_bit_mask = 1ULL << EXAMPLE_PIN_NUM_POWER};
    ESP_ERROR_CHECK(gpio_config(&pwr_gpio_config));
    gpio_set_level(EXAMPLE_PIN_NUM_POWER, EXAMPLE_LCD_BK_LIGHT_ON_LEVEL);

    gpio_config_t input_conf =
        {
            .mode = GPIO_MODE_INPUT,
            .pull_up_en = GPIO_PULLUP_ENABLE,
            .pin_bit_mask = 1ULL << PIN_LCD_RD};
    ESP_ERROR_CHECK(gpio_config(&input_conf));

    gpio_config_t bk_gpio_config =
        {
            .mode = GPIO_MODE_OUTPUT,
            .pin_bit_mask = 1ULL << EXAMPLE_PIN_NUM_BK_LIGHT};
    ESP_ERROR_CHECK(gpio_config(&bk_gpio_config));
    gpio_set_level(EXAMPLE_PIN_NUM_BK_LIGHT, EXAMPLE_LCD_BK_LIGHT_ON_LEVEL);

    static lv_disp_draw_buf_t disp_buf; // contains internal graphic buffer(s) called draw buffer(s)
    static lv_disp_drv_t disp_drv;      // contains callback functions

    ESP_LOGI(TAG, "Initialize Intel 8080 bus");
    esp_lcd_i80_bus_handle_t i80_bus = NULL;
    esp_lcd_i80_bus_config_t bus_config = {
        .clk_src = LCD_CLK_SRC_DEFAULT,
        .dc_gpio_num = EXAMPLE_PIN_NUM_DC,
        .wr_gpio_num = EXAMPLE_PIN_NUM_PCLK,
        .data_gpio_nums = {
            EXAMPLE_PIN_NUM_DATA0,
            EXAMPLE_PIN_NUM_DATA1,
            EXAMPLE_PIN_NUM_DATA2,
            EXAMPLE_PIN_NUM_DATA3,
            EXAMPLE_PIN_NUM_DATA4,
            EXAMPLE_PIN_NUM_DATA5,
            EXAMPLE_PIN_NUM_DATA6,
            EXAMPLE_PIN_NUM_DATA7,
        },
        .bus_width = 8,
        .max_transfer_bytes = LVGL_LCD_BUF_SIZE * sizeof(uint16_t)
        //.psram_trans_align = EXAMPLE_PSRAM_DATA_ALIGNMENT,
        //.sram_trans_align = 4,
    };
[... 202 lines cut ...]
    };

    ESP_LOGI(TAG,"esp_lcd_touch_new_i2c_cst816s");
    esp_lcd_touch_new_i2c_cst816s(tp_io_handle, &tp_cfg, &tp);
```

I am getting errors:

```
[1B][0;32mI (539) LVGL_SETUP: Register display driver to LVGL[1B][0m
[1B][0;32mI (539) LVGL_SETUP: Install LVGL tick timer[1B][0m
[1B][0;32mI (539) LVGL_SETUP: Initializing I2C for display touch[1B][0m
[1B][0;32mI (539) LVGL_SETUP: esp_lcd_new_panel_io_i2c[1B][0m
[1B][0;32mI (549) LVGL_SETUP: esp_lcd_touch_new_i2c_cst816s[1B][0m
[1B][0;32mI (549) gpio: GPIO[16]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (559) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2 [1B][0m
[1B][0;31mE (589) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (589) CST816S: read_id(174): I2C read failed[1B][0m
[1B][0;31mE (589) CST816S: esp_lcd_touch_new_i2c_cst816s(84): Read version failed[1B][0m
[1B][0;32mI (589) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (599) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [1B][0m
[1B][0;31mE (609) CST816S: Initialization failed![1B][0m
```

If you want, you can try use my example project to get going from your side. The display seems to work fine (I can display images and other things on the display with that example project). It is just the Display that I am struggling with.

C-basstien, 2023-06-13:
Hello,
I have the exact same issue with also the T-Display-S3, I have tested your code suggestion for swapping pin and it does not work. It also work fine with Arduino IDE on my side 
Seb

Lzw655, 2023-06-13:
@krupis @C-basstien Get it. Let me take a look at the code in Arduino.

krupis, 2023-06-13:
@Lzw655 
 
**UPDATE**
As I was testing out the I2C, I have found out a few things:
There is a RES (IO 21) pin on the T-Display-S3 display.
[image]

I have toggled it HIGH before scanning the i2c devices and it detected 0x15 I2C device.

My lvgl_setup looks like: 
(Notice I added setting IO 21 high and also reorganized where I call the I2C functions)
```
void lvgl_setup()
{
    gpio_config_t pwr_gpio_config =
        {
            .mode = GPIO_MODE_OUTPUT,
            .pin_bit_mask = 1ULL << EXAMPLE_PIN_NUM_POWER};
    ESP_ERROR_CHECK(gpio_config(&pwr_gpio_config));
    gpio_set_level(EXAMPLE_PIN_NUM_POWER, EXAMPLE_LCD_BK_LIGHT_ON_LEVEL);

    gpio_config_t input_conf =
        {
            .mode = GPIO_MODE_INPUT,
            .pull_up_en = GPIO_PULLUP_ENABLE,
            .pin_bit_mask = 1ULL << PIN_LCD_RD};
    ESP_ERROR_CHECK(gpio_config(&input_conf));

    gpio_config_t bk_gpio_config =
        {
            .mode = GPIO_MODE_OUTPUT,
            .pin_bit_mask = 1ULL << EXAMPLE_PIN_NUM_BK_LIGHT};
    ESP_ERROR_CHECK(gpio_config(&bk_gpio_config));
    gpio_set_level(EXAMPLE_PIN_NUM_BK_LIGHT, EXAMPLE_LCD_BK_LIGHT_ON_LEVEL);

    //CST816 TOUCH TESTING FUNCTION
    gpio_config_t touch_gpio_config =
        {
            .mode = GPIO_MODE_OUTPUT,
            .pin_bit_mask = 1ULL << PIN_TOUCH_RES};
    ESP_ERROR_CHECK(gpio_config(&touch_gpio_config));
    gpio_set_level(PIN_TOUCH_RES, 1);
    //END OF CST816 TOUCH TESTING FUNCTION

    static lv_disp_draw_buf_t disp_buf; // contains internal graphic buffer(s) called draw buffer(s)
    static lv_disp_drv_t disp_drv;      // contains callback functions

    ESP_LOGI(TAG, "Initialize Intel 8080 bus");
    esp_lcd_i80_bus_handle_t i80_bus = NULL;
    esp_lcd_i80_bus_config_t bus_config = {
        .clk_src = LCD_CLK_SRC_DEFAULT,
        .dc_gpio_num = EXAMPLE_PIN_NUM_DC,
        .wr_gpio_num = EXAMPLE_PIN_NUM_PCLK,
        .data_gpio_nums = {
            EXAMPLE_PIN_NUM_DATA0,
            EXAMPLE_PIN_NUM_DATA1,
            EXAMPLE_PIN_NUM_DATA2,
            EXAMPLE_PIN_NUM_DATA3,
            EXAMPLE_PIN_NUM_DATA4,
            EXAMPLE_PIN_NUM_DATA5,
            EXAMPLE_PIN_NUM_DATA6,
            EXAMPLE_PIN_NUM_DATA7,
        },
        .bus_width = 8,
        .max_transfer_bytes = LVGL_LCD_BUF_SIZE * sizeof(uint16_t)
        //.psram_trans_align = EXAMPLE_PSRAM_DATA_ALIGNMENT,
        //.sram_trans_align = 4,
    };
    ESP_ERROR_CHECK(esp_lcd_new_i80_bus(&bus_config, &i80_bus));
    esp_lcd_panel_io_handle_t io_handle = NULL;
[... 233 lines cut ...]
            .swap_xy = 0,
            .mirror_x = 0,
            .mirror_y = 0,
        },
    };
```

The serial console output:
```
[1B][0;32mI (382) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (392) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (392) gpio: GPIO[38]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (402) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (412) LVGL_SETUP: Initialize Intel 8080 bus[1B][0m
[1B][0;32mI (422) LVGL_SETUP: Install LCD driver of st7789[1B][0m
[1B][0;32mI (422) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (552) LVGL_SETUP: Initializing I2C for display touch[1B][0m
[1B][0;32mI (552) LVGL_SETUP: esp_lcd_new_panel_io_i2c[1B][0m
[1B][0;32mI (552) LVGL_SETUP: esp_lcd_touch_new_i2c_cst816s[1B][0m
[1B][0;32mI (552) CST816S: IC id: 181[1B][0m
[1B][0;32mI (552) LVGL_SETUP: Register display driver to LVGL[1B][0m
[1B][0;32mI (562) LVGL_SETUP: Install LVGL tick timer[1B][0m
```

There are no more error messages. I am very confused. What was the issue with my initial esp_lcd_touch_config_t declaration.

Lzw655, 2023-06-13:
@krupis Thanks for your patient test. I just notice that the `RST` should be set to high first.

And according to your results, I find an error in the source code. Line 71 should be placed to line 64. It may cause the level of `RST` not right.
https://github.com/espressif/esp-bsp/blob/22aa83177ed0affa79853081cc113fac09dd83cc/components/lcd_touch/esp_lcd_touch_cst816s/esp_lcd_touch_cst816s.c#L60-L80

Can you help me to verify it? If success, I'll fix it now.

krupis, 2023-06-13:
For this test, do you want me to remove the below from my code?
```
     gpio_config_t touch_gpio_config =
        {
             .mode = GPIO_MODE_OUTPUT,
             .pin_bit_mask = 1ULL << PIN_TOUCH_RES};
     ESP_ERROR_CHECK(gpio_config(&touch_gpio_config));
     gpio_set_level(PIN_TOUCH_RES, 1);
```
For the time being, i didint remove the above.

Since I made changes in the managed component, I can no longer build the project

```
ERROR: Some components (espressif/esp_lcd_touch_cst816s) in the
  "managed_components" directory were modified on the disk since the last run
  of the CMake.  Content of this directory is managed automatically.

  If you want to keep the changes, you can move the directory with the
  component to the "components"directory of your project.

  I.E.  for "espressif__esp_lcd_touch_cst816s" run:

  mv
  C:\Users\petrikas.lu\Desktop\WORK\ESP32\T-Display-S3-test\T-Display-S3-esp-idf\managed_components\espressif__esp_lcd_touch_cst816s
  C:\Users\petrikas.lu\Desktop\WORK\ESP32\T-Display-S3-test\T-Display-S3-esp-idf\components\espressif__esp_lcd_touch_cst816s

  Or, if you want to discard the changes remove the ".component_hash" file
  from the component's directory.

  I.E.  for "espressif__esp_lcd_touch_cst816s" run:

  rm
  C:\Users\petrikas.lu\Desktop\WORK\ESP32\T-Display-S3-test\T-Display-S3-esp-idf\managed_components\espressif__esp_lcd_touch_cst816s\.component_hash
```

It tells me to copy the cst816s managed component and place it to my components folder. I just manually cloned the whole espressif__esp_lcd_touch_cst816s folder and moved it into my components folder.

[image]

 After building and flashing the device, It seems that it still does not work with the below esp_lcd_touch_config_t

```
    esp_lcd_touch_config_t tp_cfg = {
        .x_max = 320,
        .y_max = 170,
        .rst_gpio_num = 21,
        .int_gpio_num = 16,
        .levels = {
            .reset = 1,
            .interrupt = 0,
        },
        .flags = {
            .swap_xy = 0,
            .mirror_x = 0,
            .mirror_y = 0,
        },
    };
```

```
[1B][0;32mI (557) gpio: GPIO[16]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2 [1B][0m
[1B][0;32mI (567) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
[1B][0;31mE (597) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (597) CST816S: read_id(176): I2C read failed[1B][0m
[1B][0;31mE (597) CST816S: esp_lcd_touch_new_i2c_cst816s(86): Read version failed[1B][0m
[1B][0;32mI (597) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (607) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [1B][0m
[1B][0;31mE (617) CST816S: Initialization failed![1B][0m
```

Lzw655, 2023-06-13:
@krupis Please remove:

```
#define PIN_TOUCH_RES                  21
    gpio_config_t touch_gpio_config =
        {
            .mode = GPIO_MODE_OUTPUT,
            .pin_bit_mask = 1ULL << PIN_TOUCH_RES};
    ESP_ERROR_CHECK(gpio_config(&touch_gpio_config));
    gpio_set_level(PIN_TOUCH_RES, 1);
```

And try the code (.reset = 0):

```
    esp_lcd_touch_config_t tp_cfg = {
        .x_max = 320,
        .y_max = 170,
        .rst_gpio_num = 21,
        .int_gpio_num = 16,
        .levels = {
            .reset = 0,
            .interrupt = 0,
        },
        .flags = {
            .swap_xy = 0,
            .mirror_x = 0,
            .mirror_y = 0,
        },
    };
```

krupis, 2023-06-13:
I tried your suggestion. My lvgl_setup:

```

void lvgl_setup()
{
    gpio_config_t pwr_gpio_config =
        {
            .mode = GPIO_MODE_OUTPUT,
            .pin_bit_mask = 1ULL << EXAMPLE_PIN_NUM_POWER};
    ESP_ERROR_CHECK(gpio_config(&pwr_gpio_config));
    gpio_set_level(EXAMPLE_PIN_NUM_POWER, EXAMPLE_LCD_BK_LIGHT_ON_LEVEL);

    gpio_config_t input_conf =
        {
            .mode = GPIO_MODE_INPUT,
            .pull_up_en = GPIO_PULLUP_ENABLE,
            .pin_bit_mask = 1ULL << PIN_LCD_RD};
    ESP_ERROR_CHECK(gpio_config(&input_conf));

    gpio_config_t bk_gpio_config =
        {
            .mode = GPIO_MODE_OUTPUT,
            .pin_bit_mask = 1ULL << EXAMPLE_PIN_NUM_BK_LIGHT};
    ESP_ERROR_CHECK(gpio_config(&bk_gpio_config));
    gpio_set_level(EXAMPLE_PIN_NUM_BK_LIGHT, EXAMPLE_LCD_BK_LIGHT_ON_LEVEL);

    //CST816 TOUCH TESTING FUNCTION
    // gpio_config_t touch_gpio_config =
    //     {
    //         .mode = GPIO_MODE_OUTPUT,
    //         .pin_bit_mask = 1ULL << PIN_TOUCH_RES};
    // ESP_ERROR_CHECK(gpio_config(&touch_gpio_config));
    // gpio_set_level(PIN_TOUCH_RES, 1);
    //END OF CST816 TOUCH TESTING FUNCTION

    static lv_disp_draw_buf_t disp_buf; // contains internal graphic buffer(s) called draw buffer(s)
    static lv_disp_drv_t disp_drv;      // contains callback functions

    ESP_LOGI(TAG, "Initialize Intel 8080 bus");
    esp_lcd_i80_bus_handle_t i80_bus = NULL;
    esp_lcd_i80_bus_config_t bus_config = {
        .clk_src = LCD_CLK_SRC_DEFAULT,
        .dc_gpio_num = EXAMPLE_PIN_NUM_DC,
        .wr_gpio_num = EXAMPLE_PIN_NUM_PCLK,
        .data_gpio_nums = {
            EXAMPLE_PIN_NUM_DATA0,
            EXAMPLE_PIN_NUM_DATA1,
            EXAMPLE_PIN_NUM_DATA2,
            EXAMPLE_PIN_NUM_DATA3,
            EXAMPLE_PIN_NUM_DATA4,
            EXAMPLE_PIN_NUM_DATA5,
            EXAMPLE_PIN_NUM_DATA6,
            EXAMPLE_PIN_NUM_DATA7,
        },
        .bus_width = 8,
        .max_transfer_bytes = LVGL_LCD_BUF_SIZE * sizeof(uint16_t)
        //.psram_trans_align = EXAMPLE_PSRAM_DATA_ALIGNMENT,
        //.sram_trans_align = 4,
    };
    ESP_ERROR_CHECK(esp_lcd_new_i80_bus(&bus_config, &i80_bus));
    esp_lcd_panel_io_handle_t io_handle = NULL;
    esp_lcd_panel_io_i80_config_t io_config = {
        .cs_gpio_num = EXAMPLE_PIN_NUM_CS,
        .pclk_hz = EXAMPLE_LCD_PIXEL_CLOCK_HZ,
        .trans_queue_depth = 20,
        .dc_levels = {
            .dc_idle_level = 0,
            .dc_cmd_level = 0,
            .dc_dummy_level = 0,
[... 125 lines cut ...]
    xTaskCreatePinnedToCore(lvgl_timer_task, "lvgl Timer", 10000, NULL, 4, NULL, 1);

}
```

Unfortunately same results:
```
[1B][0;32mI (401) LVGL_SETUP: Initialize Intel 8080 bus[1B][0m
[1B][0;32mI (411) LVGL_SETUP: Install LCD driver of st7789[1B][0m
[1B][0;32mI (411) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (541) LVGL_SETUP: Initializing I2C for display touch[1B][0m
[1B][0;32mI (551) LVGL_SETUP: esp_lcd_new_panel_io_i2c[1B][0m
[1B][0;32mI (551) LVGL_SETUP: esp_lcd_touch_new_i2c_cst816s[1B][0m
CUSTON COMPONENT CST816S config 
[1B][0;32mI (551) gpio: GPIO[16]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2 [1B][0m
[1B][0;32mI (561) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
[1B][0;31mE (591) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (591) CST816S: read_id(177): I2C read failed[1B][0m
[1B][0;31mE (591) CST816S: esp_lcd_touch_new_i2c_cst816s(87): Read version failed[1B][0m
[1B][0;32mI (591) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (601) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [1B][0m
[1B][0;31mE (611) CST816S: Initialization failed![1B][0m
[1B][0;32mI (621) LVGL_SETUP: Register display driver to LVGL[1B][0m
[1B][0;32mI (631) LVGL_SETUP: Install LVGL tick timer[1B][0m
```

krupis, 2023-06-13:
@Lzw655 
Il do a scope measurement tommorow. In the meantine, can you try with your custom CST816 device to see if yours is working?

Lzw655, 2023-06-14:
Nice work! I'll fix it according to your code.

krupis, 2023-06-14:
> Nice work! I'll fix it according to your code.

I am glad you guys have managed to find a fix but I still cannot get the touch to work properly (perhaps this time it is no longer the driver issue but my code issue?)

After making a modification :
```
static esp_err_t reset(esp_lcd_touch_handle_t tp)
{
    if (tp->config.rst_gpio_num != GPIO_NUM_NC) {
        ESP_RETURN_ON_ERROR(gpio_set_level(tp->config.rst_gpio_num, tp->config.levels.reset), TAG, "GPIO set level failed");
        vTaskDelay(pdMS_TO_TICKS(200));
        ESP_RETURN_ON_ERROR(gpio_set_level(tp->config.rst_gpio_num, !tp->config.levels.reset), TAG, "GPIO set level failed");
        vTaskDelay(pdMS_TO_TICKS(200));
    }

    return ESP_OK;
}
```

In my custom esl_lcd_touch_cst816s.c component (I cloned it from the managed component to make changes).
I was able to build the code without any issues with the following code commented out:

```
    // gpio_config_t touch_gpio_config =
         {
             .mode = GPIO_MODE_OUTPUT,
             .pin_bit_mask = 1ULL << PIN_TOUCH_RES};
     ESP_ERROR_CHECK(gpio_config(&touch_gpio_config));
     gpio_set_level(PIN_TOUCH_RES, 1);
```

and using the following esp_lcd_touch_config_t:

```
        esp_lcd_touch_config_t tp_cfg = {
        .x_max = 320,
        .y_max = 170,
        .rst_gpio_num = 21,
        .int_gpio_num = 16,
        .levels = {
            .reset = 0,
            .interrupt = 0,
        },
        .flags = {
            .swap_xy = 0,
            .mirror_x = 0,
            .mirror_y = 0,
        },
    };
```

I have then added the following callback:
```
static void example_lvgl_touch_cb(lv_indev_drv_t * drv, lv_indev_data_t * data)
{
    uint16_t touchpad_x[1] = {0};
    uint16_t touchpad_y[1] = {0};
    uint8_t touchpad_cnt = 0;
    /* Read touch controller data */
    esp_lcd_touch_read_data(drv->user_data);

    /* Get coordinates */
    bool touchpad_pressed = esp_lcd_touch_get_coordinates(drv->user_data, touchpad_x, touchpad_y, NULL, &touchpad_cnt, 1);

    if (touchpad_pressed && touchpad_cnt > 0) {
        data->point.x = touchpad_x[0];
        data->point.y = touchpad_y[0];
        data->state = LV_INDEV_STATE_PRESSED;
    } else {
        data->state = LV_INDEV_STATE_RELEASED;
[... 62 lines cut ...]
[1B][0;31mE (1463) CST816S: read_data(112): I2C read failed[1B][0m
[1B][0;31mE (1493) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (1493) CST816S: read_data(112): I2C read failed[1B][0m
[1B][0;31mE (1523) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (1523) CST816S: read_data(112): I2C read failed[1B][0m
[1B][0;31mE (1553) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (1553) CST816S: read_data(112): I2C read failed[1B][0m
[1B][0;31mE (1583) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (1583) CST816S: read_data(112): I2C read failed[1B][0m
[1B][0;31mE (1613) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (1613) CST816S: read_data(112): I2C read failed[1B][0m
[1B][0;31mE (1643) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (1643) CST816S: read_data(112): I2C read failed[1B][0m
[1B][0;31mE (1673) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (1673) CST816S: read_data(112): I2C read failed[1B][0m
[1B][0;31mE (1703) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (1703) CST816S: read_data(112): I2C read failed[1B][0m
[1B][0;31mE (1733) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (1733) CST816S: read_data(112): I2C read failed[1B][0m
[1B][0;31mE (1763) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (1763) CST816S: read_data(112): I2C read failed[1B][0m
[1B][0;31mE (1793) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (1793) CST816S: read_data(112): I2C read failed[1B][0m
[1B][0;31mE (1823) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
```

krupis, 2023-06-14:
**UPDATE**

So it turns out the touch funcionality actually works. I have added a debug printf statement inside touch callback:
```
    if (touchpad_pressed && touchpad_cnt > 0) {
        data->point.x = touchpad_x[0];
        data->point.y = touchpad_y[0];
        data->state = LV_INDEV_STATE_PRESSED;
        printf("data->point.x = %u \n",data->point.x);
        printf("data->point.y = %u \n",data->point.y);
    }
```

and when I click on the display, the callback executes and prints the coordinates.
However, I cant stop the 
```
[1B][0;31mE (1700) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (1700) CST816S: read_data(115): I2C read failed[1B][0m
```
logs from popping up.

```
[1B][0;32mI (105) esp_image: segment 1: paddr=000215a4 vaddr=3fc934[1B][0;32mI (183) esp_image: segment 4: paddr=0007f2bc vaddr=40380634 size=02d6ch ( 11628) load[1B][0m
[1B][0;32mI (192) boot: Loaded app from partition at offset 0x10000[1B][0m
[1B][0;32mI (192) boot: Disabling RNG early entropy source...[1B][0m
[1B][0;32mI (203) cpu_start: Pro cpu up.[1B][0m
[1B][0;32mI (203) cpu_start: Starting app cpu, entry point is 0x403753cc[1B][0m
[1B][0;32mI (0) cpu_start: App cpu up.[1B][0m
[1B][0;32mI (218) cpu_start: Pro cpu start user code[1B][0m
[1B][0;32mI (218) cpu_start: cpu freq: 160000000 Hz[1B][0m
[1B][0;32mI (218) cpu_start: Application information:[1B][0m
[1B][0;32mI (221) cpu_start: Project name:     i80_controller[1B][0m
[1B][0;32mI (227) cpu_start: App version:      21d1c86-dirty[1B][0m
[1B][0;32mI (232) cpu_start: Compile time:     Jun 14 2023 07:57:57[1B][0m
[1B][0;32mI (238) cpu_start: ELF file SHA256:  70ff9de2385d389f...[1B][0m
[1B][0;32mI (244) cpu_start: ESP-IDF:          v5.0.1-dirty[1B][0m
[1B][0;32mI (249) cpu_start: Min chip rev:     v0.0[1B][0m
[1B][0;32mI (254) cpu_start: Max chip rev:     v0.99 [1B][0m
[1B][0;32mI (259) cpu_start: Chip rev:         v0.1[1B][0m
[1B][0;32mI (264) heap_init: Initializing. RAM available for dynamic allocation:[1B][0m
[1B][0;32mI (271) heap_init: At 3FC967B8 len 00052F58 (331 KiB): D/IRAM[1B][0m
[1B][0;32mI (277) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM[1B][0m
[1B][0;32mI (284) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM[1B][0m
[1B][0;32mI (290) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM[1B][0m
[1B][0;32mI (297) spi_flash: detected chip: winbond[1B][0m
[1B][0;32mI (301) spi_flash: flash io: dio[1B][0m
[1B][0;33mW (305) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.[1B][0m
[1B][0;32mI (319) cpu_start: Starting scheduler on PRO CPU.[1B][0m
[1B][0;32mI (0) cpu_start: Starting scheduler on APP CPU.[1B][0m
This is esp32s3 chip with 2 CPU core(s), WiFi/BLE, silicon revision v0.1, 2MB external flash
Minimum free heap size: 379292 bytes
[1B][0;32mI (380) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (390) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (390) gpio: GPIO[38]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (400) LVGL_SETUP: Initialize Intel 8080 bus[1B][0m
[1B][0;32mI (410) LVGL_SETUP: Install LCD driver of st7789[1B][0m
[1B][0;32mI (410) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (540) LVGL_SETUP: Initializing I2C for display touch[1B][0m
[1B][0;32mI (550) LVGL_SETUP: esp_lcd_new_panel_io_i2c[1B][0m
[1B][0;32mI (550) LVGL_SETUP: esp_lcd_touch_new_i2c_cst816s[1B][0m
[1B][0;32mI (550) gpio: GPIO[16]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2 [1B][0m
[1B][0;32mI (560) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (970) CST816S: IC id: 181[1B][0m
[1B][0;32mI (970) LVGL_SETUP: Register display driver to LVGL[1B][0m
[1B][0;32mI (970) LVGL_SETUP: Install LVGL tick timer[1B][0m
[1B][0;31mE (1310) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (1310) CST816S: read_data(115): I2C read failed[1B][0m
[1B][0;31mE (1340) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (1340) CST816S: read_data(115): I2C read failed[1B][0m
[1B][0;31mE (1370) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[... 77 lines cut ...]
[1B][0;31mE (5450) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (5450) CST816S: read_data(115): I2C read failed[1B][0m
[1B][0;31mE (5480) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (5480) CST816S: read_data(115): I2C read failed[1B][0m
[1B][0;31mE (5510) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (5510) CST816S: read_data(115): I2C read failed[1B][0m
[1B][0;31mE (5540) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (5540) CST816S: read_data(115): I2C read failed[1B][0m
[1B][0;31mE (5570) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (5570) CST816S: read_data(115): I2C read failed[1B][0m
[1B][0;31mE (5600) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (5600) CST816S: read_data(115): I2C read failed[1B][0m
[... cut ...]
[1B][0;31mE (5690) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (5690) CST816S: read_data(115): I2C read failed[1B][0m
[1B][0;31mE (5720) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (5720) CST816S: read_data(115): I2C read failed[1B][0m
[1B][0;31mE (5750) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (5750) CST816S: read_data(115): I2C read failed[1B][0m
[1B][0;31mE (5780) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m

```

Lzw655, 2023-06-14:
@krupis Oh, I get the same error logs as below. But the touch function works well, I can control my screen correctly. Can't you?
```
[1B][0;31mE (1793) CST816S: read_data(112): I2C read failed[1B][0m
[1B][0;31mE (1823) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
```
I think the error reason maybe is caused by the work mode ([datasheet](https://www.buydisplay.com/download/ic/DS-CST816S_DS_V1.3.pdf) Page Four) of CST816. And its I2C can't work when not in dynamic mode.

And I find a [doc ](https://doc.riot-os.org/group__drivers__cst816s.html)that also enhances that.
[image]

So we should call `esp_lcd_touch_get_coordinates()` when get a signal through interrupt callback.

krupis, 2023-06-14:
> @krupis Oh, I get the same error logs as below. But the touch function works well, I can control my screen correctly. Can't you?
>
> ```
> [1B][0;31mE (1793) CST816S: read_data(112): I2C read [...]

Yeah I just figured out that the touch callback actually works but its constantly printing those i2c error messages. I have made an update at the same time you wrote your response :)

Would you be able to show how to properly handle it in this case (only calling the get_coordinates upon the interrupt)?

Lzw655, 2023-06-14:
Yeah, let me try.

Lzw655, 2023-06-14:
Good news, I finish it. You can refer to the following steps (code):

First, use a mutex for touch and create it before init touch:

```
static SemaphoreHandle_t touch_mux;                 // Touch mutex

touch_mux = xSemaphoreCreateBinary();
```

Then, create a callback which used to send mutex and install it:

```
static void touch_callback(esp_lcd_touch_handle_t tp)
{
    BaseType_t xHigherPriorityTaskWoken = pdFALSE;
    xSemaphoreGiveFromISR(touch_mux, &xHigherPriorityTaskWoken);

    if (xHigherPriorityTaskWoken) {
        portYIELD_FROM_ISR();
    }
}

const esp_lcd_touch_config_t tp_cfg = {
       ...
    .interrupt_callback = touch_callback,
};
```

Finally, take mutex before read_data:

```
static void bsp_touchpad_read(lv_indev_drv_t *indev_drv, lv_indev_data_t *data)
{
    esp_lcd_touch_handle_t tp = (esp_lcd_touch_handle_t) indev_drv->user_data;
    uint16_t touchpad_x[1] = {0};
    uint16_t touchpad_y[1] = {0};
    uint8_t touchpad_cnt = 0;

    assert(tp);

    /* Read data from touch controller into memory */
    if (xSemaphoreTake(touch_mux, 0) == pdTRUE) {
        esp_lcd_touch_read_data(tp);
    }

    /* Read data from touch controller */
    bool touchpad_pressed = esp_lcd_touch_get_coordinates(tp, touchpad_x, touchpad_y, NULL, &touchpad_cnt, 1);

    if (touchpad_pressed && touchpad_cnt > 0) {
        data->point.x = touchpad_x[0];
        data->point.y = touchpad_y[0];
        data->state = LV_INDEV_STATE_PRESSED;
    } else {
        data->state = LV_INDEV_STATE_RELEASED;
    }
}
```

krupis, 2023-06-14:
@Lzw655 
Thanks.

I am trying to implement functions from your code and put it in my code.

I cant fully understand where do you define functions such as:
```
BSP_NULL_CHECK
```

My project does not understand such functions. Perhaps I can replace them with something else?

Lzw655, 2023-06-14:
You can refer to this [file](https://github.com/espressif/esp-bsp/blob/master/esp-box/priv_include/bsp_err_check.h) or just simply replace them with `ESP_ERROR_CHECK()` or `assert()`.

krupis, 2023-06-14:
@Lzw655 
Thank you very much, what a legend you are!

I have got it to work. The updated code for the Lilygo T-3 Display is on my repository.

Just one quick question:

Whenever I want to draw a widget, do I need to call ```bsp_display_lock(0);```  and after I finished drawing ```bsp_display_unlock();```
Is that how it supposed to be?

Lzw655, 2023-06-14:
> Whenever I want to draw a widget, do I need to call `bsp_display_lock(0);` and after I finished drawing `bsp_display_unlock();` Is that how it supposed to be?

@krupis In any function that is not processed by the LVGL task (which runs `lv_timer_handler()`). And you shouldn't call them in LVGL's event or timer callback.

krupis, 2023-06-14:
@Lzw655 
Thanks, sounds good. Glad we got it working in the end. It was a tough one 👍 
I am sure this will be very helpful for other people as well! I know at least a couple of people who struggled with Lilygo CST816 touch on the esp-idf

Lzw655, 2023-06-14:
> @Lzw655 Thanks, sounds good. Glad we got it working in the end. It was a tough one 👍 I am sure this will be very helpful for other people as well! I know at least a couple of people who struggled with Lilygo CST816 touch on the esp-idf

Yeah, thank you for helping us to test it!

krupis, 2023-06-14:
Should I close this issue for the time being and reopen it if having any other issues with the CST816S touch?

Lzw655, 2023-06-14:
> Should I close this issue for the time being and reopen it if having any other issues with the CST816S touch?

Not necessary. It'll be closed when this [PR ](https://github.com/espressif/esp-bsp/pull/181)is merged.

closed 2023-06-15 by espzav

krupis, 2023-08-11:
@Lzw655 

Hello. I know this issue has been completed but I have been coming back to it and it haunts me. I hope you can clarify to me.

Its been a while but you have helped me solve the issue with the touch callback.

Current solution is as following: (pseudocode)

```

SemaphoreHandle_t touch_mux;                 // Touch mutex

esp_lcd_touch_config_t tp_cfg = {
            .x_max = 320,
            .y_max = 170,
            .rst_gpio_num = 21,
            .int_gpio_num = 16,
            .levels = {
                .reset = 0,
                .interrupt = 0,
            },
            .flags = {
                .swap_xy = 1,
                .mirror_x = 0,
                .mirror_y = 0,
            },
            .interrupt_callback = touch_callback,
        };

        static lv_indev_drv_t indev_drv;    // Input device driver (Touch)
        lv_indev_t *indev_touchpad;
        lv_indev_drv_init(&indev_drv);
        indev_drv.type = LV_INDEV_TYPE_POINTER;
        indev_drv.disp = disp;
        indev_drv.read_cb = bsp_touchpad_read;
        indev_drv.user_data = tp;
        indev_touchpad = lv_indev_drv_register(&indev_drv);
        BSP_NULL_CHECK(indev_touchpad, ESP_ERR_NO_MEM);

static void bsp_touchpad_read(lv_indev_drv_t * drv, lv_indev_data_t * data)
{
    uint16_t touchpad_x[1] = {0};
    uint16_t touchpad_y[1] = {0};
    uint8_t touchpad_cnt = 0;
    /* Read data from touch controller into memory */
    if (xSemaphoreTake(touch_mux, 0) == pdTRUE) {
        esp_lcd_touch_read_data(drv->user_data);
    }
    /* Get coordinates */
    bool touchpad_pressed = esp_lcd_touch_get_coordinates(drv->user_data, touchpad_x, touchpad_y, NULL, &touchpad_cnt, 1);
    if (touchpad_pressed && touchpad_cnt > 0) {
        data->point.x = touchpad_x[0];
        data->point.y = touchpad_y[0];
        data->state = LV_INDEV_STATE_PRESSED;
        printf("data->point.x = %u \n",data->point.x);
        printf("data->point.y = %u \n",data->point.y);

    } else {
        data->state = LV_INDEV_STATE_RELEASED;
    }
}

static void touch_callback(esp_lcd_touch_handle_t tp)
{
    BaseType_t xHigherPriorityTaskWoken = pdFALSE;
    xSemaphoreGiveFromISR(touch_mux, &xHigherPriorityTaskWoken);
    if (xHigherPriorityTaskWoken) {
        portYIELD_FROM_ISR();
    }
}
```

### Question1
As you can see from above, in ```lv_indev_drv_t ``` we create callback ```bsp_touchpad_read```.
I cannot wrap my head around what is the reason of using ```touch_callback``` in ```esp_lcd_touch_config_t ```. Why can't we not do anything in ```bsp_touchpad_read```?

### Question2

Lets say I have connected an I2C sensor to the same I2C bus that I2C touch and its collecting data in a seperate FreeRTOS task.
How can I ensure that my sensor does not collect data at the same time the touch is triggered. Do I need to wrap ```touch_mux``` in another semaphore called ```i2c_mux``` ?

Lzw655, 2023-08-11:
Hi @krupis , hope the answers can be helpful.

### Answer 1

LVGL will periodically call `touch_callback` in `esp_lcd_touch_config_t` to obtain the state and coordinates of a touch point. We need to update the `lv_indev_data_t * data` in this function.

### Answer 2

Yes, you need to perform synchronization, and using the same semaphore is the correct approach.

krupis, 2023-08-11:
@Lzw655 

Thanks for a quick response

**Regarding Question1:**
but ```indev_drv.read_cb = bsp_touchpad_read;``` Ensures that ```bsp_touchpad_read``` is called immediatelly when touch is detected. Why cannot update ```lv_indev_data_t * data``` directly in ```bsp_touchpad_read``` and why use ```touch_callback``` at all?

**Regarding Question2:**
I dont really understand how would that be possible with the current logic. The semaphore is taken in ```bsp_touchpad_read```. and only released after touch event. That means that no one can take this semaphore.

Could you provide some pseudo code how could I synchronize it using the same semaphore ```touch_mux```.

Why I2C sensor function that I use:
```
//will return values in uA
float ADS7828_Read_Current(enum ADS7828_channel_e channel,uint8_t device_address,uint8_t coeff){
        //need to implement semaphore here
	uint16_t read_adc = I2C_read_ads7828(channel,device_address);
	return read_adc ;
}

```

Lzw655, 2023-08-12:
@krupis 

1. As we talked before, the CST816S needs to wait for the interrupt signal to arrive before it can read the coordinates. This signal can be obtained through the `touch_callback()` function.

2. I'm sorry, I misunderstood your intention. The `touch_mux` can only be used to obtain the interrupt signal from the CST816S. You need to create another semaphore called `i2c_mux` to synchronize the operations of devices on the same I2C bus. Like below:

```c
SemaphoreHandle_t i2c_mux;               

static void bsp_touchpad_read(lv_indev_drv_t * drv, lv_indev_data_t * data)
{
    uint16_t touchpad_x[1] = {0};
    uint16_t touchpad_y[1] = {0};
    uint8_t touchpad_cnt = 0;
    /* Read data from touch controller into memory */
    if ((xSemaphoreTake(touch_mux, 0) == pdTRUE) && (xSemaphoreTake(i2c_mux, your_wait_time) == pdTRUE)) {
        esp_lcd_touch_read_data(drv->user_data);
        xSemaphoreGive(i2c_mux);
    }
   ...
}

float ADS7828_Read_Current(enum ADS7828_channel_e channel,uint8_t device_address,uint8_t coeff){
        //need to implement semaphore here
	uint16_t read_adc = 0;
        if (xSemaphoreTake(i2c_mux, your_wait_time) == pdTRUE ) {
                I2C_read_ads7828(channel,device_address);
                xSemaphoreGive(i2c_mux);
        }
	return read_adc ;
}
```

krupis, 2023-08-12:
@Lzw655 
I understood. Thank you very much !

krupis, 2023-09-15:
@Lzw655 Sorry to bring this back again after such a long time. My project was on hold for a while since I had some other urgent projects to develop.

I have tried your suggestion regarding synchronizing multiple I2C devices on the same I2C bus and it does not seem to work as expected.

In my Display component I have the following:
```

SemaphoreHandle_t i2c_mux;
SemaphoreHandle_t lvgl_mux;  // LVGL mutex
SemaphoreHandle_t touch_mux; // Touch mutex

static void bsp_touchpad_read(lv_indev_drv_t *drv, lv_indev_data_t *data)
{
    uint16_t touchpad_x[1] = {0};
    uint16_t touchpad_y[1] = {0};
    uint8_t touchpad_cnt = 0;
    if ((xSemaphoreTake(touch_mux, 0) == pdTRUE) && (xSemaphoreTake(i2c_mux, 100) == pdTRUE)) {
        esp_lcd_touch_read_data(drv->user_data);
        xSemaphoreGive(i2c_mux);
    }
    /* Get coordinates */
    bool touchpad_pressed = esp_lcd_touch_get_coordinates(drv->user_data, touchpad_x, touchpad_y, NULL, &touchpad_cnt, 1);
    if (touchpad_pressed && touchpad_cnt > 0)
    {
        data->point.x = touchpad_x[0];
        data->point.y = touchpad_y[0];
        data->state = LV_INDEV_STATE_PRESSED;
        printf("data->point.x = %u \n", data->point.x);
        printf("data->point.y = %u \n", data->point.y);
    }
    else
    {
        data->state = LV_INDEV_STATE_RELEASED;
    }
}

static void touch_callback(esp_lcd_touch_handle_t tp)
{
    BaseType_t xHigherPriorityTaskWoken = pdFALSE;
    xSemaphoreGiveFromISR(touch_mux, &xHigherPriorityTaskWoken);
    if (xHigherPriorityTaskWoken)
    {
        portYIELD_FROM_ISR();
    }
}

void lvgl_setup()
{

    static lv_disp_draw_buf_t disp_buf; // contains internal graphic buffer(s) called draw buffer(s)
    static lv_disp_drv_t disp_drv;      // contains callback functions

    gpio_config_t pwr_gpio_config =
        {
            .mode = GPIO_MODE_OUTPUT,
            .pin_bit_mask = 1ULL << EXAMPLE_PIN_NUM_POWER};
    ESP_ERROR_CHECK(gpio_config(&pwr_gpio_config));
    gpio_set_level(EXAMPLE_PIN_NUM_POWER, EXAMPLE_LCD_BK_LIGHT_ON_LEVEL);

    gpio_config_t input_conf =
        {
            .mode = GPIO_MODE_INPUT,
            .pull_up_en = GPIO_PULLUP_ENABLE,
            .pin_bit_mask = 1ULL << PIN_LCD_RD};
    ESP_ERROR_CHECK(gpio_config(&input_conf));

    gpio_config_t bk_gpio_config =
        {
            .mode = GPIO_MODE_OUTPUT,
            .pin_bit_mask = 1ULL << EXAMPLE_PIN_NUM_BK_LIGHT};
[... 257 lines cut ...]

I think the issue lies somewhere within touch_mux. Instead of taking a temperature reading every 100ms, it only captures a temperature/humidity reading every time I touch on the display. See the logs below:

```
2023/09/15 08:27:21: data->point.x = 217 
data->point.y = 35 
data->point.x = 217 
data->point.y = 35 
temperature = 26.43 
humidity = 45.33 
2023/09/15 08:27:23: data->point.x = 155 
data->point.y = 114 
data->point.x = 155 
data->point.y = 114 
temperature = 26.42 
humidity = 44.88 
2023/09/15 08:27:27: data->point.x = 166 
data->point.y = 97 
data->point.x = 166 
data->point.y = 97 
temperature = 26.40 
humidity = 43.70 
```

Perhaps you have any ideas why that might happen and how to overcome this to ensure proper synchronization between multiple I2C devices? I really appreciate your help!

Lzw655, 2023-09-18:
@krupis Yeah, your guess is right. The `touch_mux` shouldn't be taken in `SHT40_task()`. And your `SHT40_task()` should like below:
```c
//SHT40_task is a free rtos task that is being created in my main.c.
void SHT40_task(void *argument)
{
    for (;;)
    {
        if (xSemaphoreTake(i2c_mux, 100) == pdTRUE)
        {
            Measure_temp_humidity(SHT40_TEMP_HUMID_REG_HIGH_PREC);
            xSemaphoreGive(i2c_mux);
        }
        vTaskDelay(100 / portTICK_PERIOD_MS);
    }
}
```

tao-a-man, 2024-01-19:
hii @Lzw655 I have same issue like picture bellow, Help me plss , I config like 
`#define LCD_TOUCH_SDA 18
#define LCD_TOUCH_SCL 17
#define LCD_TOUCH_RST 21
#define LCD_TOUCH_INT 16
#define LCD_TOUCH_I2C_CLK_SPEED_HZ 400000
#define LCD_TOUCH_I2C_NUM 0`....
`ESP_ERROR_CHECK(i2c_param_config(LCD_TOUCH_I2C_NUM, &i2c_conf));
    ESP_ERROR_CHECK(i2c_driver_install(LCD_TOUCH_I2C_NUM, i2c_conf.mode, 0, 0, 0));

    esp_lcd_panel_io_i2c_config_t io_config = ESP_LCD_TOUCH_IO_I2C_CST816S_CONFIG();
    ESP_ERROR_CHECK(esp_lcd_new_panel_io_i2c((esp_lcd_i2c_bus_handle_t)LCD_TOUCH_I2C_NUM, &io_config, &io_handle));`

[image]

davidthings, 2024-09-18:
I think this is because the CST816 goes to sleep after a while, after which it no longer responds to I2C transactions.

Modifying `read_data( )` as follows fixes this problem by declining to read from the sensor unless IRQ is low.

``` c
static esp_err_t read_data(esp_lcd_touch_handle_t tp)
{
    typedef struct {
        uint8_t num;
        uint8_t x_h : 4;
        uint8_t : 4;
        uint8_t x_l;
        uint8_t y_h : 4;
        uint8_t : 4;
        uint8_t y_l;
    } data_t;

    data_t point;

    // Only do this read if the interrupt line is low
    if ( ( tp->config.int_gpio_num != GPIO_NUM_NC ) && gpio_get_level( tp->config.int_gpio_num ) ) {
        return ESP_OK;
    }

    ESP_RETURN_ON_ERROR(i2c_read_bytes(tp, DATA_START_REG, (uint8_t *)&point, sizeof(data_t)), TAG, "I2C read failed");

    portENTER_CRITICAL(&tp->data.lock);
    point.num = (point.num > POINT_NUM_MAX ? POINT_NUM_MAX : point.num);
    tp->data.points = point.num;
    /* Fill all coordinates */
    for (int i = 0; i < point.num; i++) {
        tp->data.coords[i].x = point.x_h << 8 | point.x_l;
        tp->data.coords[i].y = point.y_h << 8 | point.y_l;
    }
    portEXIT_CRITICAL(&tp->data.lock);

    return ESP_OK;
}
```

FrankBJensen, 2024-09-22:
Hi there.

I have a problem, that seems somewhat similar to this. I am using esp_lvgl_port and the esp_led_touch_cs816s driver for that. I have installed both as components, and it kind of work.

Problem is the wakeup and sleep of the CST816.

This function fails `esp_lcd_touch_new_i2c_cst816s(tp_io_handle, &tp_cfg, &touch_handle);` due to no communication from adr. 0x15, due to the CST816 sleeping.

So I need to wake up the CST816 in order to be able to register the interupt callback. But I can only communicate with the CST816 in a timewindow following the interupt. Its like the hen and the egg.......

I can not wakeup the CST816 by toggling the reset.

Right now my only workaround is to continously tap the display during boot, to keep it alive, until I get a CST816 init OK. Then all works. This way the esp_lcd_touch_new_i2c_cst816s do not fail, and callback is registred correctly. But that is not a usable solution.

Any suggestions?

davidthings, 2024-09-22:
Did you try checking the interrupt level before running the I2C operation.  If no interrupt, don’t poll the chip?  This is working for me.

FrankBJensen, 2024-09-22:
Can you communicate with the chip, before the first interupt? My problem is, that LVGL wants to initialize it, before any interupts, meaning the chip is sleeping.

I call this function, in order to register it, but it fails, cause it tries to write to the chip, when its sleeping.

`static esp_err_t app_touch_init(void)
{
    // Initialize touch HW

    touch_mux = xSemaphoreCreateBinary();

    esp_lcd_panel_io_i2c_config_t tp_io_config = ESP_LCD_TOUCH_IO_I2C_CST816S_CONFIG();

    esp_lcd_touch_config_t tp_cfg = {
        .x_max = EXAMPLE_LCD_H_RES,
        .y_max = EXAMPLE_LCD_V_RES,
        .rst_gpio_num = TOUCH_GPIO_RST, // GPIO_NUM_39
        .int_gpio_num = TOUCH_GPIO_INT,
        .levels = {
            .reset = 0,
            .interrupt = 0,
        },
        .flags = {
            .swap_xy = 0,
            .mirror_x = 0,
            .mirror_y = 0,
        }
    };

    esp_lcd_panel_io_handle_t tp_io_handle = NULL;

    ESP_RETURN_ON_ERROR(esp_lcd_new_panel_io_i2c((esp_lcd_i2c_bus_handle_t)EXAMPLE_TOUCH_I2C_NUM, &tp_io_config, &tp_io_handle), TAG, "");
    return esp_lcd_touch_new_i2c_cst816s(tp_io_handle, &tp_cfg, &touch_handle);
}`

FrankBJensen, 2024-09-22:
My solution for this problem. Catch the first interupt, and init the touch there, including disable autosleep. Only catch is, that first touch after reboot is ignored.

`void IRAM_ATTR gpio_isr_handler(void* arg) {
    // Handle the interrupt, e.g., toggle a flag or send data to a queue
    int gpio_num = (int) arg;
    ESP_EARLY_LOGI(TAG, "Interrupt on GPIO %d", gpio_num);
    gpio_isr_handler_remove(gpio_num);
    xQueueSendFromISR(gpio_evt_queue, &gpio_num, NULL);
}`

And handle this interupt in my main loop, to add the touch handler to LVGL.

`if (xQueueReceive(gpio_evt_queue, &gpio_num, portMAX_DELAY)) {
    ESP_LOGI(TAG, "Handling GPIO %d initialization in task", gpio_num);
    app_touch_init();
    ESP_LOGI(TAG, "Initialization done for GPIO %d", gpio_num);
    esp_err_t result = i2c_master_write_byte1(0x15, 0xFE, 0xFF); // Disable autosleep
}`