espressif/esp-bsp #182: Support for ST7701S display driver (IDFGH-10449)

krupis, 2023-06-19:
### Is your feature request related to a problem?

Nowadays, it has become increasingly popular to include a touch display for various projects. This allows consumer/user to control and monitor the device much more easily.

I think it would be very beneficial to support as many of the popular displays as possible to give developers freedom to choose display for their particular application.

Currently, I do not think esp-idf support any of the LCD round displays.

### Describe the solution you'd like.

We have recently gone through some issues to get the 1.9" touch display from LilyGo to work properly.

I think it would also be very cool to get the 2.1" Round display from LilyGo to work on esp-idf:
https://www.lilygo.cc/products/t-rgb?variant=42503165673653

This display is based on ST7701S TFT LCD display and CST820 touch chip.

### Describe alternatives you've considered.

Using different displays such as ST7789/ILI9341 but these driver chips does not seem to support round displays.

### Additional context.

Please consider this. I think it will be very useful for esp community :)

krupis, 2023-06-19:
> > Currently, I do not think esp-idf support any of the LCD round displays.
> 
> We do have round display support, e.g. https://components.espressif.com/components/espressif/esp_lcd_gc9a01
> 
> We're maintaining LCD display and touch driver in our [bsp repo](https://github.com/espressif/esp-bsp/tree/master/components/lcd). You can ask for feature requests there in the future.

Thanks for a quick response. I will look up gc9a01 for now :)

krupis, 2023-07-03:
@espzav 
Hello. Can you clarify to me if there are any plans to create a driver for the ST7701S display in the near future?

I really believe this could take ESP32 projects to the next level as there 2.1Inch and 2.8Inch display ST7701S display versions. Also, huge advantage of those displays is that they support to 24 bit color depth!

Yes, they require a lot of pins but you can make all the required connections with the ESP32-S3 and with the upcomming ESP32-P4 it might open even more doors!

espzav, 2023-07-04:
Hello @krupis,
I am so sorry. I want to add a driver for every LCD, which customers want, but I cannot do it. The main decision of the company is, support mainly LCDs, which are on the Espressif's boards. Other LCD controllers can be supported by the community. I will prepare some contribution guide, how to create a new LCD controller.

igrr, 2023-07-13:
To clarify our approach to adding support for new LCD drivers:
- In this repository we will add LCD drivers for controllers used on official ESP development boards.
- If someone needs a display controller for their commercial project, and they contact Espressif through business support channels, we can also accommodate such requests.
- At a lower priority, and depending on time available, we might add support also for third-party boards which are very common. (For example: m5stack.)
- We may also add support for common display controllers like ILI9341 if we get many customer requests for this.
- We do intend to prepare a guide for community contributors on how to create a new BSP or a new LCD driver, and publish it in the [component registry](https://components.espressif.com/) for others to use.

We can keep this issue open for feedback, and if more users express interest in ST7701S, we will consider it when planning our work.

Lzw655, 2023-09-05:
@igrr I have finished the ST7701S driver due to some customers' requirements. Besides, I also have some other LCD & touch drivers and I'm going to push them into [esp-iot-solution](https://github.com/espressif/esp-iot-solution/tree/master/components/display/lcd), which also can be found in the component registry.

espzav, 2023-09-05:
@Lzw655 Why do you want to put it into esp-iot-solution? All other LCD drivers are in BSP repository.

Lzw655, 2023-09-05:
> To clarify our approach to adding support for new LCD drivers:
>
> * In this repository we will add LCD drivers for controllers used on official ESP development boards.
> * If someone needs a display [...]

I agree with this explanation. And I often develop some special screens for customers. Uploading the driver to esp-iot-solution would be more convenient. If it's an LCD driver needed for development boards, I'll upload it to esp-bsp. Moreover, all these drivers can eventually be found in the component registry.

krupis, 2023-09-05:
@Lzw655 
Thanks for that. Can you just clarify to me one thing please.
Does ESP32 support RGB888? According to this, it does not:
[image]

Lzw655, 2023-09-05:
@krupis Actually, the ESP32-S3 supports both 16-bit (data line width is 16) RGB565 and 8-bit (data line width is 8) RGB888. But in the usual sense, RGB888 actually refers to 24-bit, which is not supported.

VojtechBartoska, 2024-03-13:
@espzav Probably we can close this ticket or do you plan to add support for this component?

closed 2024-03-13 by espzav