espressif/esp-bsp #189: please add a full example source file

dannybackx, 2023-07-13:
I'm following the code samples in README.md and there's so many things I am looking for.
Could you just please provide a full source file from which you've pasted the stuff in README.md to avoid the (seemingly) endless search ?

Thanks you !

tore-espressif, 2023-07-13:
Hello @dannybackx , the code snippets are usually coming from our BSP examples https://github.com/espressif/esp-bsp/tree/master/examples . 
Is that what you are looking for?

dannybackx, 2023-07-14:
Strange, replying doesn't work. Anyway, here's what I said yesterday : 
It might be but that code looks well structured so it isn't simple any more.

The README.md appears to provide something simple to follow but it doesn't; there are too many omissions and (I suspect, I'm trying to get a simple sample together) errors.

There are no obvious definitions or types of some variables in the sample code in README.md, e.g. for tp. I am using esp-idf v5.1 so I tend to skip the part that says "only in 5.0 and older", to find later that it defines and assigns values to e.g. io_handle. To put it politely I don't think the "only in 5.0" part is entirely accurate.

So : no.

    Danny

I'm currently at 

I (00:00:02.134) LVGL: Starting LVGL task

assert failed: lvgl_port_add_disp esp_lvgl_port.c:187 (disp_cfg->io_handle != NULL)

Backtrace: 0x403758de:0x3fca1f80 0x4037b059:0x3fca1fa0 0x403807ed:0x3fca1fc0 0x42009657:0x3fca20e0 0x42008afd:0x3fca2120 0x4204d94a:0x3fca21d0 0x4037d47a:0x3fca21f0
0x403758de: panic_abort at /home/danny/src/github/esp32/esp-idf-v5.1/components/esp_system/panic.c:452

0x4037b059: esp_system_abort at /home/danny/src/github/esp32/esp-idf-v5.1/components/esp_system/port/esp_system_chip.c:84

igrr, 2023-07-14:
Since there are multiple README.md files in the repository (one for each component and example), could you please point to the one you are looking at? (I'm guessing it might be [this one](https://github.com/espressif/esp-bsp/blob/master/components/esp_lvgl_port/README.md), for esp_lvgl_port?)

dannybackx, 2023-07-14:
Apologies for that. Yes, you pointed to the one that I meant.

igrr, 2023-07-14:
Got it, agree that the existing readme is confusing in the part which has to do with "v5.0 or earlier". Looks like the code block there should have been split into two.

Since esp_lvgl_port can be used on its own outside of BSP components, adding an example inside that component makes sense! Have renamed the issue accordingly.

PR #197 (2023-07-28): esp_lvgl_port: Add simple example and improved readme.
# ESP-BSP Pull Request checklist

> Note: For new BSPs create a PR with this [link](https://github.com/espressif/esp-bsp/compare/main...my-branch?quick_pull=1&template=pr_template_bsp.md).

- [x] Version of modified component bumped
- [x] CI passing

# Change description
Added simple example for esp_lvgl_port. Tested on IDF4.4 and master.
Closes #189
closed 2023-07-28 by espzav