espressif/esp-bsp #191: esp_lvgl_port appears to lack declaration of lv_tick_inc()

dannybackx, 2023-07-15:
In an esp32 project with these dependencies
```
dependencies:
  lvgl/lvgl: "*"
  espressif/esp_lcd_touch: "*"
  espressif/esp_lvgl_port: "*"
  espressif/esp_lcd_touch_gt911: "==1.0.0"

```
there is always a compile error because in managed_components/espressif__esp_lvgl_port/esp_lvgl_port.c around line 790, lv_tick_inc() isn't declared. So I added a single line in my copy as such : 
```
static void lvgl_port_tick_increment(void *arg)
{
    void lv_tick_inc(uint32_t);

    /* Tell LVGL how many milliseconds have elapsed */
    lv_tick_inc(lvgl_port_timer_period_ms);
}

```
I can't find a setting to automate this so I'm guessing it's an oversight.
Example : https://sourceforge.net/p/esp32s3-4827s043/code/HEAD/tree/trunk/lvgl-port .

Thanks,
  Danny

igrr, 2023-07-17:
Hi Danny,
That's an interesting issue, as all the components and examples in this repository pass through CI build process, and it isn't failing, it seems.

Is it possible that you somehow have CONFIG_LV_TICK_CUSTOM enabled in your project's sdkconfig file? Looking at LVGL source code, lv_tick_inc function won't be declared if LV_TICK_CUSTOM is set: https://github.com/lvgl/lvgl/blob/2cdd41345afb5e0dc44a354a2c6141ceaee6f353/src/tick/lv_tick.h#L38-L44

dannybackx, 2023-07-17:
Nice wording. "interesting" as in probably invalid. I need to figure out what's wrong but with a test I just did I proved your point.
Apologies.

closed 2023-07-17 by dannybackx