espressif/esp-bsp #193: esp_lcd_touch_tt21100: crash on touch with CONFIG_LOG_MAXIMUM_LEVEL_DEBUG=y

stintel, 2023-07-18:
We're experiencing a crash on touch when we build with CONFIG_LOG_MAXIMUM_LEVEL_DEBUG=y.

```
  espressif/esp_lcd_touch_tt21100:
    component_hash: 27d0ec1226e2ce81ad6a170b27530dbcf959cd1fb9abe1ba207509a8a2f2ed22
    source:
      service_url: https://api.components.espressif.com/
      type: service
    version: 1.0.7~1
```

```
abort() was called at PC 0x4037b4bf on core 0
0x4037b4bf: lock_acquire_generic at /opt/esp/idf/components/newlib/locks.c:139

Backtrace: 0x4037a1be:0x3fceb830 0x40384095:0x3fceb850 0x4038c1b2:0x3fceb870 0x4037b4bf:0x3fceb8e0 0x4037b687:0x3fceb910 0x42131645:0x3fceb940 0x42131d9d:0x3fceb960 0x42131e19:0x3fceb990 0x4212cc1f:0x3fceb9c0 0x4212cbf1:0x3fceba20 0x42124f07:0x3fceba40 0x4214b61d:0x3fceba60 0x4206f3c5:0x3fcebac0 0x4206fe96:0x3fcebb10 0x4206f7d5:0x3fcebb30 0x4207e296:0x3fcebb60 0x420722d4:0x3fcebb80 0x4207fa3f:0x3fcebbd0 0x4206fe04:0x3fcebc00
0x4037a1be: panic_abort at /opt/esp/idf/components/esp_system/panic.c:408

0x40384095: esp_system_abort at /opt/esp/idf/components/esp_system/esp_system.c:137

0x4038c1b2: abort at /opt/esp/idf/components/newlib/abort.c:46

0x4037b4bf: lock_acquire_generic at /opt/esp/idf/components/newlib/locks.c:139

0x4037b687: _lock_acquire_recursive at /opt/esp/idf/components/newlib/locks.c:167 (discriminator 3)
 (inlined by) __retarget_lock_acquire_recursive at /opt/esp/idf/components/newlib/locks.c:323 (discriminator 3)

0x42131645: __env_lock at /builds/idf/crosstool-NG/.build/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdlib/envlock.c:43

0x42131d9d: _findenv_r at /builds/idf/crosstool-NG/.build/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdlib/getenv_r.c:85

0x42131e19: _getenv_r at /builds/idf/crosstool-NG/.build/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdlib/getenv_r.c:126

0x4212cc1f: _tzset_unlocked_r at /builds/idf/crosstool-NG/.build/xtensa-esp32s3-elf/src/newlib/newlib/libc/time/tzset_r.c:28

0x4212cbf1: _tzset_unlocked at /builds/idf/crosstool-NG/.build/xtensa-esp32s3-elf/src/newlib/newlib/libc/time/tzset.c:115

0x42124f07: localtime_r at /builds/idf/crosstool-NG/.build/xtensa-esp32s3-elf/src/newlib/newlib/libc/time/lcltime_r.c:34 (discriminator 8)

0x4214b61d: esp_log_system_timestamp at /opt/esp/idf/components/log/log_freertos.c:84

0x4206f3c5: esp_lcd_touch_tt21100_read_data at /willow/managed_components/espressif__esp_lcd_touch_tt21100/esp_lcd_touch_tt21100.c:215

0x4206fe96: esp_lcd_touch_read_data at /willow/managed_components/espressif__esp_lcd_touch/esp_lcd_touch.c:36

0x4206f7d5: lvgl_port_touchpad_read at /willow/managed_components/espressif__esp_lvgl_port/esp_lvgl_port.c:658

0x4207e296: _lv_indev_read at /willow/managed_components/lvgl__lvgl/src/hal/lv_hal_indev.c:186

0x420722d4: lv_indev_read_timer_cb at /willow/managed_components/lvgl__lvgl/src/core/lv_indev.c:83

0x4207fa3f: lv_timer_exec at /willow/managed_components/lvgl__lvgl/src/misc/lv_timer.c:313
 (inlined by) lv_timer_handler at /willow/managed_components/lvgl__lvgl/src/misc/lv_timer.c:109

0x4206fe04: lvgl_port_task at /willow/managed_components/espressif__esp_lvgl_port/esp_lvgl_port.c:535
```

pborcin, 2024-03-13:
Hi @stintel, could you please provide source code and hardware you are using? Tried this with esp32s3-box-3 and display examples and didn't experience any crash.

pborcin, 2024-04-15:
Hi @stintel, can you please provide more information about which board you're using and share a relevant piece of code? I've tested this with esp32_s3_korvo_2 and ran display examples without encountering any issues.
Thanks!

pborcin, 2024-05-15:
Closing due to inactivity. Feel free to reopen if you have any updates.

closed 2024-05-15 by espzav