espressif/esp-bsp #200: [DS18B20] Feature Request: Trigger temperature conversion on all devices

CJCombrink, 2023-07-29:
When working with more than one DS18B20 temperature sensors, three in my case, it would be nice to call `ds18b20_trigger_temperature_conversion()` on all three and then only delay for the required amount before reading the temperatures from all three. 

The current implementation forces me to wait `x ms` for each sensor where ideally I could only wait once.

I would imagine an function defined as follow:
```
esp_err_t ds18b20_trigger_temperature_conversions(const ds18b20_device_handle_t *ds18b20, uint8_t ds18b20_size );
```

espzav, 2023-11-07:
@suda-morris FYI

VojtechBartoska, 2024-03-13:
@suda-morris Can you please share your 2 cents regarding this? Thanks

PR #595 (2025-09-04): feat(ds18b20): support to trigger multi sensors conversion (BSP-693)
Closes #200
closed 2025-09-04 by suda-morris