espressif/esp-bsp #208: BSP for ESP32-S3-LCD-EV-Board-SUB2 480x480

ahmed8812, 2023-08-17:
Hi,

Is there a BSP for the said ESP32-S3-LCD-EV-Board-SUB2 480x480 (GC9503CV Display and FT5x06 Touch)?
We are trying to use Squareline, but the current available template only exist for ESP32-S3-LCD-EV-Board-SUB3 800 x 480.

Thanks

espzav, 2023-08-17:
Hello @ahmed8812,
the BSP for ESP32-S3-LCD-EV-Board contains selection for both LCD screens. You can select it in `menuconfig` (Component config -> Board Support Package -> LCD -> Select Sub board -> ...) or you can put into `sdkconfig.defaults` option `CONFIG_BSP_LCD_SUB_BOARD_480_480=y`.

In new version of the BSP, there will be automatic selection for LCD. We are working on it.

ahmed8812, 2023-08-17:
Hi @espzav,

I have done the settings. It compile and run successfully no error.  But the display just dark.
When the setting CONFIG_BSP_LCD_SUB_BOARD_480_480=y it should selects GC9503CV Display and FT5x06 Touch already.
I am doing anything wrong on the Squareline perhaps?

Thanks

espzav, 2023-08-17:
Hi @ahmed8812,
When you compiled the code before, than you put the configuration option into `sdkconfig.defaults`, it wouldn't be accepted after next compilation. You should remove `sdkconfig` file and try compilation again. 

Do you see any error on console after flash and reset the board?

ahmed8812, 2023-08-17:
@espzav Hi,

Yup that is exactly I did :)

No errors what so ever, except it just give warning, that this board cannot adjust lcd brightness

Thanks

ahmed8812, 2023-08-17:
@espzav Hi,

This is the log:

I (0) cpu_start: App cpu up.
I (764) esp_psram: SPI SRAM memory test OK
I (773) cpu_start: Pro cpu start user code
I (773) cpu_start: cpu freq: 240000000 Hz
I (773) cpu_start: Application information:
I (776) cpu_start: Project name:     display_lvgl_demos
I (782) cpu_start: App version:      1
I (786) cpu_start: Compile time:     Aug 17 2023 17:11:11
I (792) cpu_start: ELF file SHA256:  2b93674a25869747...
I (798) cpu_start: ESP-IDF:          v5.1
I (803) cpu_start: Min chip rev:     v0.0
I (808) cpu_start: Max chip rev:     v0.99 
I (812) cpu_start: Chip rev:         v0.2
I (817) heap_init: Initializing. RAM available for dynamic allocation:
I (824) heap_init: At 3FC9B758 len 0004DFB8 (311 KiB): DRAM
I (831) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (837) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (843) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (850) esp_psram: Adding pool of 7360K of PSRAM memory to heap allocator
I (857) spi_flash: detected chip: gd
I (861) spi_flash: flash io: qio
W (865) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (879) sleep: Configure to isolate all GPIO pins in sleep state
I (885) sleep: Enable automatic switching of GPIO sleep configuration
I (892) app_start: Starting scheduler on CPU0
I (897) app_start: Starting scheduler on CPU1
I (897) main_task: Started on CPU0
I (907) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (915) main_task: Calling app_main()
I (2506) SUB-BOARD_480x480: Initialize RGB panel
I (2522) S3-LCD-EV-BOARD: Starting LVGL task
I (2544) app_main: Display LVGL demo
I (2673) main_task: Returned from app_main()

I use exactly the examples from https://github.com/espressif/esp-bsp/tree/master/examples/display_lvgl_demos
And set CONFIG_BSP_LCD_SUB_BOARD_480_480=y

Lzw655, 2023-08-18:
Hi @ahmed8812, I have generated a Squareline project and tested well in my board with SUB2-480x480. Did you check if your sub-board is good? Or did you modify some code that may cause it?

ahmed8812, 2023-08-18:
Hi

I haven't modify a single line of code

Can you provide your application binary, bootloader and partition table
file?

I would like to try esp tool to flash to my board. I have try other samples
code but the screen still black. No error when monitoring. I use esp idf
v5.1

Thanks

Lzw655, 2023-08-18:
You can try to use [ESP LaunchPad](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://raw.githubusercontent.com/espressif/esp-dev-kits/master/launch.toml) to flash the factory bin, as below:

[image]

ahmed8812, 2023-08-21:
Hi

Thanks for replying. I did successfully run the sample '86 Demo' with ESP Launchpad.

But what puzzles me is that I try to compile the '86 Demo' too from this: 
https://github.com/espressif/esp-dev-kits/tree/master/esp32-s3-lcd-ev-board/examples/86box_demo
But the display does not run. I use release/v5.1

Here are the logs from both ESP Launchpad and the dev-kit hub which ran the same project '86 Demo'
I have 2 units of ESP32-S3-LCD-EV-Board 480x480. But same results no display appears

[Log File ESP-IDF V5.1 86 Demo.txt](https://github.com/espressif/esp-bsp/files/12391085/Log.File.ESP-IDF.V5.1.86.Demo.txt)
[Log File ESP Launch Pad 86 Demo.txt](https://github.com/espressif/esp-bsp/files/12391086/Log.File.ESP.Launch.Pad.86.Demo.txt)

I hope this problem can be sorted out
Thanks

Lzw655, 2023-08-21:
Hi @ahmed8812, I know the reason now. The board I used is an old version and the newest version should set the level of GPIO3 (ESP_PANEL_LCD_RGB_IO_VSYNC) to high before initialization as following code:

```c
    gpio_config_t io_conf = {};
    io_conf.pin_bit_mask = BIT64(BSP_LCD_VSYNC);
    io_conf.mode = GPIO_MODE_OUTPUT;
    io_conf.pull_up_en = true;
    gpio_config(&io_conf);
    gpio_set_level(BSP_LCD_VSYNC, 1);
```

And the sample '86 Demo' in the repo also miss these code. This is our oversight, and I will fix this issue in the next version of the BSP.

ahmed8812, 2023-08-21:
Hi,

May I know where can I modify this?
To see how it goes

Thanks

Lzw655, 2023-08-21:
You can modify the *main.c* as following code:

```c
...
#include "driver/gpio.h"
...

void app_main(void)
{
    /* Initialize I2C (for touch and audio) */
    bsp_i2c_init();

    gpio_config_t io_conf = {};
    io_conf.pin_bit_mask = BIT64(BSP_LCD_VSYNC);
    io_conf.mode = GPIO_MODE_OUTPUT;
    io_conf.pull_up_en = true;
    gpio_config(&io_conf);
    gpio_set_level(BSP_LCD_VSYNC, 1);

    /* Initialize display and LVGL */
    ...
}

```

ahmed8812, 2023-08-21:
Hi

Thanks it works now
My squareline project can also display out now

Thanks
Hope Espressif can release the updated bsp as soon as possible

Lzw655, 2023-08-21:
Sure, thank you for reporting this problem!

Lzw655, 2023-09-05:
@ahmed8812 The BSP has been updated. Please try the newest version and check if it works.

ahmed8812, 2023-09-05:
@Lzw655 Hi,

I have tried both:
https://github.com/espressif/esp-bsp
https://github.com/espressif/esp-dev-kits

These works fine now. However I would like to suggest for the BSP 'esp32_s3_lcd_ev_board' to add some note on to select display size depending on user hardware will be helpful (e.g. menuconfig (Component config -> Board Support Package -> LCD -> Select Sub board -> ...) since the same firmware works on both display sizes

Thanks

Lzw655, 2023-09-05:
@ahmed8812 It's a good suggestion. I will add the note in the README of BSP's next release. Thank you!

closed 2024-01-23 by tore-espressif