espressif/esp-bsp #230: examples position for esp-bsp

FBEZ, 2023-09-26:
Hi everyone,

I'm "porting" the bsp to the esp-eye (the older one) and have a quick question. 
So far I'm using an application inside the example folder of the component to test it (esp-bsp/esp_eye/examples), as it's usually done (I guess at least). These bsp s don't have an example folder inside but a common one at the same level esp-bsp/examples.

So if I'd submit a PR, should I also add an example in the esp-bsp/examples or can I keep the esp-bsp/esp_eye/examples? Because the esp-eye doesn't have a display, hence the display+camera example wouldn't fit.

Thanks!

igrr, 2023-09-26:
In the case of esp-bsp, indeed we keep examples outside of component directories, since the examples are kind of designed to be common between the BSPs.

To make the examples available via `idf.py create-project-from-example`, we had an idea to copy relevant examples into the component directory right before uploading the component to the registry, but it's not done yet.

As for the esp-eye example: what do you expect the example will do, other than grabbing the frames from the camera?

FBEZ, 2023-09-29:
Ok, everything's clear, thank you.
It may be the same as the [PR](https://github.com/espressif/idf-extra-components/pull/240), i.e. scanning the image for a qrcode using quirc and reporting the data on the terminal. If scanned correctly, it could flash one of the leds, or something like that.
Otherwise something as simple as taking the frame and writing on the terminal the data of the sensor and picture size.

espzav, 2025-04-02:
I am closing this issue, it was solved inside mentioned PR and examples are included in each BSP now. Feel free to reopen this issue.

closed 2025-04-02 by espzav