espressif/esp-bsp #233: Freeze while trying to read id using ESP32C6 and releasev5.1

MohammedBENHADINE, 2023-10-10:
Hello,

Before opening this issue, I've tried two base codes to get CST816S working : [esp-watch from esp-bsp](https://github.com/Lzw655/esp-bsp/tree/62ff5c68de1e56eabfe40ca22a6e4ca25fb21631/examples/esp-watch) and [T display S3 ](https://github.com/krupis/T-Display-S3-esp-idf) both codes doesn't seem to work with my ESP32C6 and LCD TFT setup.
The LCD works fine and displays whatever you instructs it to show with LVGL.
But once I configure the CST816S driver the MCU stucks at doing its first hadnshake with the driver, at id reading.
I've made a minimal code (without this driver) to test the hardware wiring and i'm able to notice INT interrupts when I touch the screen and I'm also able to get the ID with this command : 
`static void id_register_read(uint8_t addr)
{
    uint8_t data[2];

    i2c_master_write_read_device(0, 0x15, &addr, 1, data, 1, 1000 / portTICK_PERIOD_MS);
    ESP_LOGI(TAG, "WHO_AM_I = %X", data[0]);

}`

Following up the issue with the library I ended up seeing the freeze/blocking at this function 
`esp_err_t panel_io_i2c_rx_buffer(esp_lcd_panel_io_t *io, int lcd_cmd, void *buffer, size_t buffer_size)` 
in esp_lcd/src/esp_lcd_panel_io_i2c.c file.

I've this minimal code to debug this issue : 
```
#include <stdio.h>
#include <inttypes.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "driver/i2c.h"
#include "esp_system.h"
#include "esp_err.h"
#include "esp_check.h"
#include "esp_log.h"
#include "esp_lcd_touch_cst816s.h"
#include "freertos/semphr.h"

static const char *TAG = "i2CTest";
#define CHIP_ID_REG         (0xA7)

static QueueHandle_t gpio_evt_queue = NULL;
static esp_lcd_touch_handle_t tp;   // LCD touch handle

static SemaphoreHandle_t touch_mux;

static void IRAM_ATTR touch_callback(esp_lcd_touch_handle_t tp)
{
    uint32_t gpio_num = 8;
    xQueueSendFromISR(gpio_evt_queue, &gpio_num, NULL);

    BaseType_t xHigherPriorityTaskWoken = pdFALSE;
    xSemaphoreGiveFromISR(touch_mux, &xHigherPriorityTaskWoken);

    if (xHigherPriorityTaskWoken) {
        portYIELD_FROM_ISR();
    }
}

void i2c_init(void)
{
    const i2c_config_t i2c_conf = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = 19,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_io_num = 20,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .master.clk_speed = 400000
    };
    i2c_param_config(0, &i2c_conf);
    i2c_driver_install(0, i2c_conf.mode, 0, 0, 0);
}

static void reset(void)
{
        gpio_set_level(21, 0);
        vTaskDelay(pdMS_TO_TICKS(200));
        gpio_set_level(21, 1);
        vTaskDelay(pdMS_TO_TICKS(200));
}

static void id_register_read(uint8_t addr)
{
    uint8_t data[2];

    i2c_master_write_read_device(0, 0x15, &addr, 1, data, 1, 1000 / portTICK_PERIOD_MS);
    ESP_LOGI(TAG, "WHO_AM_I = %X", data[0]);

}

void app_main(void)
{
    printf("Hello world!\n");
    gpio_evt_queue = xQueueCreate(10, sizeof(uint32_t));
    touch_mux = xSemaphoreCreateBinary();

    i2c_init();

    esp_lcd_panel_io_i2c_config_t io_config = ESP_LCD_TOUCH_IO_I2C_CST816S_CONFIG();

    esp_lcd_touch_config_t tp_cfg = {
        .x_max = 240,
        .y_max = 280,
        .rst_gpio_num = 21,
        .int_gpio_num = 2,
        .levels = {
            .reset = 0,
            .interrupt = 0,
        },
        .flags = {
            .swap_xy = 0,
            .mirror_x = 0,
            .mirror_y = 0,
        },
        .interrupt_callback = touch_callback,
    };
    esp_lcd_panel_io_handle_t io_handle = NULL;
    esp_lcd_new_panel_io_i2c((esp_lcd_i2c_bus_handle_t)0 , &io_config, &io_handle);
    esp_lcd_touch_new_i2c_cst816s(io_handle, &tp_cfg, &tp);
    uint32_t io_num;
    for(;;) {
        if (xSemaphoreTake(touch_mux, 0) == pdTRUE) {
        esp_lcd_touch_read_data(tp); // read only when ISR was triggled
        uint16_t touch_x[1];
        uint16_t touch_y[1];
        uint16_t touch_strength[1];
        uint8_t touch_cnt = 0;
        bool touchpad_pressed = esp_lcd_touch_get_coordinates(tp, touch_x, touch_y, touch_strength, &touch_cnt, 1);
        ESP_LOGI(TAG, "touch pressed = %d , x = %u , y=%u", touchpad_pressed, touch_x[0], touch_y[0]);

     }
        if(xQueueReceive(gpio_evt_queue, &io_num, portMAX_DELAY)) {
            printf("GPIO[%"PRIu32"] intr, val: %d\n", io_num, gpio_get_level(io_num));
        }
    }
}
```

Please note that I've added a queue to see if the ISR generates the interrupts but I don't see them coming. The esp32c6 is lost in the id reading step in esp_lcd_touch_new_i2c_cst816s function.
Could you please provide me with some guidelines to debug this issue or resolve it.

Regards,
Mohammed

Lzw655, 2023-10-11:
Hi @MohammedBENHADINE, I tested the newest [v1.0.3 driver](https://components.espressif.com/components/espressif/esp_lcd_touch_cst816s) in esp32c6 and it worked well.

The drivers in [esp-watch from esp-bsp](https://github.com/Lzw655/esp-bsp/tree/62ff5c68de1e56eabfe40ca22a6e4ca25fb21631/examples/esp-watch) were created a long time ago and are not well-maintained. Please try again with the latest drivers mentioned above.

MohammedBENHADINE, 2023-10-11:
Hi @Lzw655 ,
Thanks for the quick reply.
I'm getting the same behavior with the newest version of the cst816s driver.
Now I wonder if the LCD has a different IC. The supplier gave me [this source code](https://drive.google.com/file/d/1sujTWDI8Nbjxv358Zwta7AOV8811Jyy6/view?usp=sharing) but it's not usable from my ESP32C6.
The IC reference is CST816D , is that different from the CST816S ?
Some pictures of the display :
[image]
[image]
[image]

Lzw655, 2023-10-12:
According to the provided code, the CST816D has the same I2C address and register address as CST816S. So the [v1.0.3 driver](https://components.espressif.com/components/espressif/esp_lcd_touch_cst816s) should work with CST816D.

Can you share the whole project for me to test?

MohammedBENHADINE, 2023-12-02:
Hey @Lzw655 ,
Sorry for the delay.
Here is my test repo : https://github.com/MohammedBENHADINE/cst816d_test

Regards

SinnerK0N, 2024-04-15:
any update on this?

MohammedBENHADINE, 2024-04-15:
After many tests I figured out that the issue was hardware based.
The IC has some manufacturing issue and after using another one the library worked fine.

closed 2024-04-15 by MohammedBENHADINE