espressif/esp-bsp #234: CST816S touch and ST7789 display (X axis inverted)

krupis, 2023-10-13:
I am creating this issue to solve : 
https://github.com/krupis/T-Display-S3-esp-idf/issues/1

An issue has been discovered on T-Display-S3 development board from Lilygo:
https://www.lilygo.cc/products/t-display-s3

The issue is that the X axis for touch seems to be inverted which does not allow to use sliders and other display elements that require proper X axis.

The example project can be found:
https://github.com/krupis/T-Display-S3-esp-idf

The example contains a simple slider. Please see the video attached below:
https://www.youtube.com/shorts/KaQ7J90QQLY

I have tried to play around with ```esp_lcd_panel_mirror``` but could not achieve the proper results. Please can someone suggest how to ensure the X axis is not inverted for touch?

Lzw655, 2023-10-13:
Hi @krupis, the CST816S driver in your project is an old version, please update to [v1.3.0](https://components.espressif.com/components/espressif/esp_lcd_touch_cst816s) and use with ISR feature ([example](https://github.com/espressif/esp-bsp/tree/master/components/lcd_touch/esp_lcd_touch_cst816s#example-use)).

`esp_lcd_panel_mirror()` is the API for LCD not for touch. If you want to mirror X axis, there are two ways:
1. Set `.flags.mirror_x ` of `esp_lcd_touch_config_t ` to 1.
2. Call `esp_lcd_touch_set_mirror_x()`.

krupis, 2023-10-13:
@Lzw655 
Thanks for a rapid response.
Can I first fix the issue on the old version and then update it later to avoid more confusion, or the fix requires new version?

I have tried all possible combinations of
```
        .flags.swap_xy
        .flags.mirror_x
        .flags.mirror_y

```

and none of them work sadly.

**TEST1**
```
    esp_lcd_touch_config_t tp_cfg = {
        .x_max = 320,
        .y_max = 170,
        .rst_gpio_num = 21,
        .int_gpio_num = 16,
        .levels = {
            .reset = 0,
            .interrupt = 0,
        },
        .flags = {
            .swap_xy = 1,
            .mirror_x = 0,
            .mirror_y = 0,
        },
        .interrupt_callback = touch_callback,
    };
```

This is the initial configuration. The X axis is inverted as shown in the youtube video attached to initial post. The touch coordinates go as below:
[image]

**TEST2**
Your suggested method
```
    esp_lcd_touch_config_t tp_cfg = {
        .x_max = 320,
        .y_max = 170,
        .rst_gpio_num = 21,
        .int_gpio_num = 16,
        .levels = {
            .reset = 0,
            .interrupt = 0,
        },
        .flags = {
            .swap_xy = 1,
            .mirror_x = 1,
            .mirror_y = 0,
        },
        .interrupt_callback = touch_callback,
    };
```
The touch coordinates go as following:
[image]

As you can see its really strange. The y coordinates no longer goes from 0 to 170 but from 320 to 150. Could you help me understand this? Is there anything else you would like to me to test?

Lzw655, 2023-10-13:
Please try to swap the `x_max` and `y_max`. These two parameters should be the actual resolution of touch.

krupis, 2023-10-13:
You are right. I have managed to get my display and touch working properly with the following parameters:
```
    /*
        esp_lcd_touch_config_t tp_cfg = {
        .x_max = 170,
        .y_max = 320,
        .rst_gpio_num = 21,
        .int_gpio_num = 16,
        .levels = {
            .reset = 0,
            .interrupt = 0,
        },
        .flags = {
            .swap_xy = 1,
            .mirror_x = 0,
            .mirror_y = 1,
        },
        .interrupt_callback = touch_callback,
    };*/
```

but it is not fully clear to me why do I need to use ```.swap_xy = 1,``` and why do I need to do ```.swap_xy = 1,``` and why I need to change ```y_max``` and ```x_max``` to get it to work properly.

Lzw655, 2023-10-13:
We will add documents to introduce these API usages in the future. Here, let me briefly explain this:
1.  `x_max` and `y_max` should always be the actual touch resolution even if you need to swap or mirror it later. Because they are the limits of the hardware. According to your picture, its actual axis should be like this:
[image]
2. So your `x_max` and `y_max` should be  320x170 and just set `mirror_x ` to 1. The structure is as below: 
```
/*
        esp_lcd_touch_config_t tp_cfg = {
        .x_max = 170,
        .y_max = 320,
        .rst_gpio_num = 21,
        .int_gpio_num = 16,
        .levels = {
            .reset = 0,
            .interrupt = 0,
        },
        .flags = {
            .swap_xy = 0,
            .mirror_x = 1,
            .mirror_y = 0,
        },
        .interrupt_callback = touch_callback,
    };*/
```

Lzw655, 2023-10-13:
The actual touch resolution is usually as same as the LCD. So you can simply check them by the following information:
[image]

krupis, 2023-10-13:
@Lzw655 
But thats not how it is. If I have:
```
        esp_lcd_touch_config_t tp_cfg = {
        .x_max = 170,
        .y_max = 320,
        .rst_gpio_num = 21,
        .int_gpio_num = 16,
        .levels = {
            .reset = 0,
            .interrupt = 0,
        },
        .flags = {
            .swap_xy = 1,
            .mirror_x = 0,
            .mirror_y = 1,
        },
        .interrupt_callback = touch_callback,
    };
```

The coordinates are as following:
[image]

With the above configuration, the slider works perfectly fine.

If I use``` mirror_x = 1``` (as you have suggested):

```
        esp_lcd_touch_config_t tp_cfg = {
        .x_max = 170,
        .y_max = 320,
        .rst_gpio_num = 21,
        .int_gpio_num = 16,
        .levels = {
            .reset = 0,
            .interrupt = 0,
        },
        .flags = {
            .swap_xy = 1,
            .mirror_x = 1,
            .mirror_y = 0,
        },
        .interrupt_callback = touch_callback,
    };
```
The coordinates will be:
[image]

In this configuration, the X axis is inverted and Y also does not work properly

As you previously shown in the Lilygo picture, the resolution is  170x320, but since I use the display in horizontal position, I want my longer axis (horizontal axis), to be longer if that makes sense.

Lzw655, 2023-10-13:
Oh, sorry for that mistake. The `mirror_y` should be 1 instead of `mirror_x`. Because the mirror operation is earlier than the swap operation, you can check the source code here: https://github.com/espressif/esp-bsp/blob/162d138b82fb052f33b9b49263974b465b41a6c8/components/lcd_touch/esp_lcd_touch/esp_lcd_touch.c#L88-L103

That means we should mirror the actual axis.

krupis, 2023-10-13:
@Lzw655 
Thanks for all your efforts explaining this :) I get the idea now

Lzw655, 2023-10-13:
Anytime!

closed 2023-10-13 by krupis