espressif/esp-bsp #242: Compilation error with ESP IDF 5.1.1

HamitAkay, 2023-10-28:
I used the previous version with ESP IDF 5.1.1 and had no problem. After updating to 1.1.0, I get these error messages:

> esp_lcd_st7796.c:77:29: error: 'esp_lcd_panel_dev_config_t' has no member named 'rgb_ele_order'
> esp_lcd_st7796.c:78:10: error: 'LCD_RGB_ELEMENT_ORDER_RGB' undeclared (first use in this function)

I checked esp_lcd_panel_dev_config_t in ESP IDF and in fact it doesn't have such a member.

Lzw655, 2023-10-28:
Hi @HamitAkay, the v1.1.0 ST7796 driver uses a new parameter `rgb_ele_order` of `esp_lcd_panel_dev_config_t`, which was just updated in ESP-IDF v5.0 and v5.1 a few weeks ago. So please update your ESP-IDF to the up-to-date commit.

valkuc, 2023-10-28:
> Hi @HamitAkay, the v1.1.0 ST7796 driver uses a new parameter `rgb_ele_order` of `esp_lcd_panel_dev_config_t`, which was just updated in ESP-IDF v5.0 and v5.1 a few weeks ago. So please update your ESP-IDF to the up-to-date commit.

@Lzw655 ESP IDF 5.0.4 does not have `rgb_ele_order` in `esp_lcd_panel_dev_config_t`. That's why I raised https://github.com/espressif/esp-bsp/issues/240

valkuc, 2023-10-28:
@HamitAkay Actually ESP IDF 5.1 has `LCD_RGB_ELEMENT_ORDER_RGB` constant. See  [   LCD_RGB_ELEMENT_ORDER_RGB, /*!< RGB element order: RGB */](https://github.com/espressif/esp-idf/blob/8c6114d0d2f904fbc7c6ef82f828d38308550487/components/hal/include/hal/lcd_types.h#L27). You probably misconfigured your project.

Lzw655, 2023-10-30:
> @Lzw655 ESP IDF 5.0.4 does not have `rgb_ele_order` in `esp_lcd_panel_dev_config_t`. That's why I raised #240

Hi @valkuc, ESP-IDF release/v5.0 (v5.0.4) does have [rgb_ele_order](https://github.com/espressif/esp-idf/blob/f333925342c47f54c8fbe020f7607cd992040b23/components/esp_lcd/include/esp_lcd_panel_vendor.h#L24).

igrr, 2023-10-30:
@Lzw655 the link you posted is to the release/v5.0 branch, not to v5.0.4 tag. Here's the link to the same file of v5.0.4, it doesn't contain this feature: https://github.com/espressif/esp-idf/blob/v5.0.4/components/esp_lcd/include/esp_lcd_panel_vendor.h

> the v1.1.0 ST7796 driver uses a new parameter rgb_ele_order of esp_lcd_panel_dev_config_t, which was just updated in ESP-IDF v5.0 and v5.1 a few weeks ago. So please update your ESP-IDF to the up-to-date commit

Please note that each component declares the supported IDF version in the manifest file. If the supported version is specified as, say, >=4.4, then the component should work with both v5.0.3 and v5.0.4. So either the code shouldn't depend on a feature introduced in a bugfix release, or should have appropriate ifdefs to handle both versions.

Lzw655, 2023-10-30:
@igrr Got it. I will fix this by using a compatible name `rgb_endian`.

valkuc, 2023-10-30:
> @igrr Got it. I will fix this by using a compatible name `rgb_endian`.

@Lzw655 could you please also fix compilation on TAG version of 5.0.4 (https://github.com/espressif/esp-bsp/issues/240)

HamitAkay, 2023-10-30:
I downloaded IDF 5.1.1 almost a week ago and it worked fine until I updated st7796 to 1.1.0. Then as @Lzw655 mentioned I pulled the new changes from the ESP IDF repo and it works fine now. However, I agree with @igrr. The component says it works with IDF >=4.4 but even though I have 5.1.1 it still throws the error.

Lzw655, 2023-10-30:
> > @igrr Got it. I will fix this by using a compatible name `rgb_endian`.
> 
> @Lzw655 could you please also fix compilation on TAG version of 5.0.4 (#240)

Off course.

Lzw655, 2023-10-30:
> I downloaded IDF 5.1.1 almost a week ago and it worked fine until I updated st7796 to 1.1.0. Then as @Lzw655 mentioned I pulled the new changes from the ESP IDF repo and it works fine now. However, [...]

Sorry for that. I will fix this right now.

tore-espressif, 2023-10-30:
> Sorry for that. I will fix this right now.

Meanwhile, the affected versions were yanked in component manager, so version solving will not pick them (unless the user explicitly requests them)

PR #244 (2023-11-01): lcd: fix incompatible feature
# ESP-BSP Pull Request checklist

> Note: For new BSPs create a PR with this [link](https://github.com/espressif/esp-bsp/compare/main...my-branch?quick_pull=1&template=pr_template_bsp.md).

- [x] Version of modified component bumped
- [x] CI passing

# Change description
* replace `rgb_ele_order` with `rgb_endian`
* add more idf version check
* closes #240 
* closes #242