espressif/esp-bsp #251: WT32-SC01 Plus missing library

KianLuis297, 2023-11-16:
I recently bought a WT32-SC01 Plus and I am trying to program a gui using Squareline Studio. 
Unfortunately the library for the WT32-SC01 Plus seems to have been deleted from the espressif github.
In Squareline Studio you can choose Espressif WT32-SC01 Plus as a preset but when trying to export any project, an error message says that the "wt32_sc01_plus.zip file cannot be found". I think the root of the problem is that the wt32-sc01 plus is not listed on this github anymore and therefore cannot be downloaded and used for the build / compiling.

Are the files still available or is there any otherway to program the WT32-SC01 Plus using Squareline Studio?
I tried to build my own .zip file containing the needed header files but I can't get it to work. The screen won't show any output using any other library from the espressif presets.

Thanks!

espzav, 2023-11-20:
Hello @KianLuis297,
I am sorry, we never supported this board. It must be some mistake. I don't see this board in SquareLine Studio.

tore-espressif, 2024-02-27:
@KianLuis297 The WT32-SC01 board was never listed under Espressif's boards in SquareLineStudio. If you want to use it, you will have to write your own hardware initialization code. This typically includes
1. Init of ESP peripherals (SPI...)
2. Init of the display driver
3. Init of LVGL (init + draw buffers allocation)

There are many tutorials and manuals online

I'll close this issue for now, as we don't plan to add WT32-SC01 board to esp-bsp anytime soon

closed 2024-02-27 by tore-espressif