espressif/esp-bsp #261: 3 errors found in : examples  /display_audio_photo

Bobobel, 2023-12-16:
I am **deeply depressed** about the information and documentation that I find with Espressif examples!
It is a common style to clearly give advice for newcommers regarding setup, prerequisites and other demands.
I do not find any word on which platform e.g. the mentioned example will run.
In the end I found **3 errors** when I tried that example on my ESP32S3 BOX-3:
1) compilation standard for ESP-BOX, not BOX-3!
	Fix: delete all other sdkconfig files
2) Partition is default table
	Fix: changed to custom in menuconfig
3) SPIFFS flash size is set to 2MB but 4MB needed
	Fix: changed accordingly
Especially with the last error: that example could never have worked on any platform!
In ReadMe no word about changes to be made in menuconfig.

**Please provide better** "## How to use the example" and especially a "## What to setup in menuconfig" !

tore-espressif, 2023-12-21:
Hello @Bobobel ,
I'm sorry to hear about the troubles.

ad 1: You're right we didn't finish the documentation yet, so it is difficult to set correct Development Kit for the example. This is on our TODO list with high priority.

ad 2 and 3: We provide the default configuration (16MB flash + custom partition table) for this example in [sdkconfig.bsp.esp-box-3](https://github.com/espressif/esp-bsp/blob/master/examples/display_audio_photo/sdkconfig.bsp.esp-box-3), but for some reason the configuration was not used in your project. I'll have a look

Additionally, especially for esp-box development kits, we have more examples here https://github.com/espressif/esp-box/tree/master/examples . You will have to select your esp-box-3 in menuconfig: `idf.py menuconfig` - > `HMI Board Config`

Bobobel, 2023-12-22:
Thanks for the reply and the good work you are doing.
When I had to mange software it was "discipline and documentation first".
Now I looked for examples for BOX-Lite and changed it to BOX-3. Nice that the routines are compatible. You can give a hint for users, because the genereic examples for BOX-3 are yet very limited.