espressif/esp-bsp #265: esp_lvgl_port这个组件如何不添加官方的屏幕初始化使用

Szeroy, 2023-12-29:
Add screen这个例子里面使用了esp_lcd_new_panel_io_spi和esp_lcd_new_panel_st7789然后调用 disp_handle = lvgl_port_add_disp(&disp_cfg);来进行初始化lvgl的显示，但是我现在不希望用官方的屏幕初始化，因为我自己写好了屏幕的驱动，就想放进lvgl_port_flush_callback这个函数里面刷新屏幕，请问这个应该怎么操作，还是说我只能自己在重新移植一遍lvgl？

Lzw655, 2023-12-31:
Hi @Szeroy, 在直接使用 `esp_lvgl_port` 组件的情况下，是需要采用乐鑫官方的 LCD 驱动，如果你的屏幕是 SPI/RGB 接口，可以参考[文档](https://docs.espressif.com/projects/esp-iot-solution/zh_CN/latest/display/lcd/lcd_development_guide.html#id13)进行移植。在使用自定义驱动（非官方接口）的情况下，可以参考[例程](https://docs.espressif.com/projects/esp-iot-solution/zh_CN/latest/display/lcd/lcd_development_guide.html#id2)移植 LVGL。

Lzw655, 2023-12-31:
这个问题很有可能是没有调用 `lv_disp_flush_ready(disp_drv);` 函数来通知 LVGL 可以进行继续运行，否则 LVGL 会如你 LOG 显示循环死等导致任务看门狗超时，请确保在每次刷屏完成后（如 SPI 的 `post_cb()`）调用该函数，如[示例](https://github.com/espressif/esp-idf/blob/5d8fb281e7b4ae3f3827c91aa6dc7ef969580b5a/examples/peripherals/lcd/spi_lcd_touch/main/spi_lcd_touch_example_main.c#L80)所示。

Szeroy, 2024-01-01:
> 这个问题很有可能是没有调用 函数来通知 LVGL 可以进行继续运行，否则 LVGL 会如你 LOG 显示循环死等导致任务看门狗超时，请确保在每次刷屏完成后（如 SPI 的 ）调用该函数，如[示例](https://github.com/espressif/esp-idf/blob/5d8fb281e7b4ae3f3827c91aa6dc7ef969580b5a/examples/peripherals/lcd/spi_lcd_touch/main/spi_lcd_touch_example_main.c#L80)所示。`lv_disp_flush_ready(disp_drv);``post_cb()`

我现在把lvgl_port_flush_ready这个函数加进去是不会死机了，不过我现在换了个例程，是用idf里面自带的lvgl例子，不过两个例子差不多，现在还有一个问题，我想显示在单色屏幕上面，我把他配置成单色然后把缓冲区打印出来全是1，如果换成8bit色彩是一值的，这个是什么原因？
```
static void example_lvgl_flush_cb(lv_disp_drv_t *drv, const lv_area_t *area, lv_color_t *color_map)
{
    esp_lcd_panel_handle_t panel_handle = (esp_lcd_panel_handle_t)drv->user_data;
    int offsetx1 = area->x1;
    int offsetx2 = area->x2;
    int offsety1 = area->y1;
    int offsety2 = area->y2;

    printf("offsetx1 = %d,offsetx2 = %d,offsety1 = %d,offsety2 = %d\n", offsetx1, offsetx2, offsety1, offsety2);
    for (size_t i = 0; i < offsety2 - offsety1; i++)
    {
        for (size_t j = 0; j < offsetx2 - offsetx1; j++)
        {
            printf("%x", ((char *)color_map)[i * (offsetx2 - offsetx1) + j]);
        }
        printf("\n");
    }
    printf("\n");
    lvgl_port_flush_ready(disp);
}
```
这个是设置成1bit打印的
```
offsetx1 = 120,offsetx2 = 199,offsety1 = 153,offsety2 = 199
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
1111111111111111111111111111111111111111111111111111111111111111111111111111111
[... 121 lines cut ...]
    ESP_LOGI(TAG, "Install LVGL tick timer");
    // Tick interface for LVGL (using esp_timer to generate 2ms periodic event)
    const esp_timer_create_args_t lvgl_tick_timer_args = {
        .callback = &example_increase_lvgl_tick,
        .name = "lvgl_tick"};
    esp_timer_handle_t lvgl_tick_timer = NULL;
    ESP_ERROR_CHECK(esp_timer_create(&lvgl_tick_timer_args, &lvgl_tick_timer));
    ESP_ERROR_CHECK(esp_timer_start_periodic(lvgl_tick_timer, EXAMPLE_LVGL_TICK_PERIOD_MS * 1000));

    // lv_port_fs_init();

    xSemaphoreTake(lvgl, portMAX_DELAY);
    // ui_init();
    xSemaphoreGive(lvgl);
    while (1)
    {
        // raise the task priority of LVGL and/or reduce the handler period can improve the performance
        vTaskDelay(pdMS_TO_TICKS(10));
        // The task running lv_timer_handler should have lower priority than that running `lv_tick_inc`
        xSemaphoreTake(lvgl, portMAX_DELAY);
        lv_timer_handler();
        xSemaphoreGive(lvgl);
    }
}
```

espressif2022, 2024-01-02:
单色模式，有没有注册 disp_drv 的 set_px_cb 回调？

Szeroy, 2024-01-08:
> 单色模式，有没有注册 `disp_drv` 的 `set_px_cb` 回调？
> `disp_ctx->disp_drv.set_px_cb = lvgl_port_pix_monochrome_callback;`

是这个问题，谢谢啦

closed 2024-01-08 by Szeroy