espressif/esp-bsp #270: SSD1306 for 128 x 32 and some reflections

nicklasb, 2024-01-06:
Hi, 

I have some suggestion and questions on the SSD1306 library:

1. I don't feel that it says so anywhere, but the SSD1306 seems to only work for 128 x 64 OLEDs. 
It would be nice if the documentation said so. 
2. The 0x40 is actually the "Set Display Start Line (40h~7Fh)" setting. Not the "Set VCOM Deselect Level"
3. It would be quite easy to just add a setting to make it work with 128 *32 displays. 
Here is a diff of the changes that is needed: https://github.com/espressif/esp-bsp/compare/master...nicklasb:esp-bsp:master .
The duty cycle and the VCOM changes are typical, the some odd one is the "Set Display Start Line", ant that fixes the alignment. I am not sure if that works on all 32:s but the 0.91" I've got seems to be at least common.  

For me, the nice thing about this library is that it is minimal, and that it lets me handle the I2C so that I can avoid having duplicate initiations on the port. So for my money, adding a ssd1306_create_32() variant would be the easiest solution. That would be the minimal change to the library. Obviously one could take that further and do more changes, but I am thinking good enough and don't break stuff.

Should I make that PR (that also fixes the 0x40 comment)? Would that be in the spirit of this repo?

espzav, 2024-01-08:
Hi @nicklasb,
thank you for your suggestions. You can use component from ESP-IDF for SSD1306. It was updated for change screen size.
Here: https://github.com/espressif/esp-idf/blob/master/components/esp_lcd/include/esp_lcd_panel_ssd1306.h

nicklasb, 2024-01-08:
Hi, thanks, hadn't seen that!

@espzav, It would be good if the page of this component indicated that there was an official implementation in ESP-IDF, as this was where I ended up when looking for an ssh1306-driver for ESP-IDF.  (and maybe if this also applies to other components in this repo).

closed 2024-01-08 by nicklasb

espzav, 2024-01-09:
Hi @nicklasb,
I overlooked that we have it twice. I will solve it soon. For official information about LCD, we have this table: https://github.com/espressif/esp-bsp/blob/master/LCD.md The link to this table is in official documentation and the link is in BSP readme too.
I am so sorry for the confusing information.