espressif/esp-bsp #292: ESP32S3-EYE BSp lvgl build error

MichelBerg, 2024-02-07:
I am using ESP-IDF /release/v5.0 and i am facing some issues using the BSP for the ESP32S3-EYE.

I have created a new project using `idf.py create-project`.
After that I added the ESP32S3-EYE component using `idf.py add-dependency "espressif/esp32_s3_eye^3.0.3~1"`
When I try to build the project using idf.py set-target esp32s3 and idf.py build I get the following error:

```
...

In file included from /home/bergm/Desktop/ESP32S3-EYE_Modules/default_esp32s3_bsp/managed_components/lvgl__lvgl/src/lv_init.h:17,

                 from /home/bergm/Desktop/ESP32S3-EYE_Modules/default_esp32s3_bsp/managed_components/lvgl__lvgl/lvgl.h:24,

                 from /home/bergm/Desktop/ESP32S3-EYE_Modules/default_esp32s3_bsp/managed_components/espressif__esp32_s3_eye/include/bsp/esp32_s3_eye.h:17,

                 from /home/bergm/Desktop/ESP32S3-EYE_Modules/default_esp32s3_bsp/managed_components/espressif__esp32_s3_eye/esp32_s3_eye.c:24:

/home/bergm/Desktop/ESP32S3-EYE_Modules/default_esp32s3_bsp/managed_components/lvgl__lvgl/src/lv_conf_internal.h:64:17: note: '#pragma message: Possible failure to include lv_conf.h, please read the comment in this file if you get errors'

   64 |         #pragma message("Possible failure to include lv_conf.h, please read the comment in this file if you get errors")

      |                 ^~~~~~~

In file included from /home/bergm/Desktop/ESP32S3-EYE_Modules/default_esp32s3_bsp/managed_components/espressif__esp32_s3_eye/esp32_s3_eye.c:24:

/home/bergm/Desktop/ESP32S3-EYE_Modules/default_esp32s3_bsp/managed_components/espressif__esp32_s3_eye/include/bsp/esp32_s3_eye.h:386:42: error: unknown type name 'lv_disp_rot_t'; did you mean 'lv_disp_t'?

  386 | void bsp_display_rotate(lv_disp_t *disp, lv_disp_rot_t rotation);

      |                                          ^~~~~~~~~~~~~

      |                                          lv_disp_t

/home/bergm/Desktop/ESP32S3-EYE_Modules/default_esp32s3_bsp/managed_components/espressif__esp32_s3_eye/esp32_s3_eye.c:378:42: error: unknown type name 'lv_disp_rot_t'; did you mean 'lv_disp_t'?

  378 | void bsp_display_rotate(lv_disp_t *disp, lv_disp_rot_t rotation)

      |                                          ^~~~~~~~~~~~~

      |                                          lv_disp_t

ninja: build stopped: subcommand failed.

ninja failed with exit code 1, output of the command is in the /home/bergm/Desktop/ESP32S3-EYE_Modules/default_esp32s3_bsp/build/log/idf_py_stderr_output_117850 and /home/bergm/Desktop/ESP32S3-EYE_Modules/default_esp32s3_bsp/build/log/idf_py_stdout_output_117850

```

espzav, 2024-02-08:
Hi @MichelBerg, I am so sorry about this issue. We had issue in `esp_lvgl_port` component and we yanked it. It seems, that it still downloading and compiling this bad version. We are fixing the issue.
Please, add this line to yours `main/idf_component.yml`:
```
espressif/esp_lvgl_port: "1.4.0"
```
And remove the `build` folder and `dependencies.lock` file before compile the project again.

MichelBerg, 2024-02-08:
No Problem, thanks for the fast response. For now, this issue with the lvgl version is fixed.

I wrote test functions for some BSP wrappers that I want to use for my project. But I think there are still some problems within the ESP32-S3-EYE camera default configuration.

https://github.com/espressif/esp-bsp/blob/899b6d066c004b36327da8fa5062f751db6ddbba/bsp/esp32_s3_eye/include/bsp/esp32_s3_eye.h#L193-L220

```
 .pin_sccb_sda = GPIO_NUM_NC,      \ 
 .pin_sccb_scl = GPIO_NUM_NC,      \ 
```

Should be replaced with:
https://github.com/espressif/esp-bsp/blob/899b6d066c004b36327da8fa5062f751db6ddbba/bsp/esp32_s3_eye/include/bsp/esp32_s3_eye.h#L46-L48

And the bsp is also missing the` # include "esp_cam.h"`.
I know that I can include this to my main file. But its is weird, because the bsp file does include lvgl, iot_button etc.

And also this is confusing:
https://github.com/espressif/esp-bsp/blob/899b6d066c004b36327da8fa5062f751db6ddbba/bsp/esp32_s3_eye/include/bsp/esp32_s3_eye.h#L170-L192

As it says _@attention I2C must be enabled by bsp_i2c_init(), before camera is initialized_

If i build my code like this:

```
bsp_i2c_init(); // causes the following error
const camera_config_t camera_config = BSP_CAMERA_DEFAULT_CONFIG; //fixed as mentioned before
esp_err_t err = esp_camera_init(&camera_config);
```

```
I (1626) s3 ll_cam: DMA Channel=4

I (1626) cam_hal: cam init ok

I (1626) sccb: pin_sda 4 pin_scl 5

I (1626) sccb: sccb_i2c_port=1

E (1626) i2c: i2c driver install error

E (1626) camera: sccb init err

E (1636) camera: Camera probe failed with error 0xffffffff(ESP_FAIL)
```

If i dont init the i2c with the bsp it works perfectly fine.
To verify that i2c is getting initialized anyways, I looked into the esp_cam library. Where i2c automaticaly gets initialized as soon we call esp_camera_init();

https://github.com/espressif/esp32-camera/blob/f0bb42917cddcfba2c32c2e2fb2875b4fea11b7a/driver/esp_camera.c#L282

But I still wonder why it is failing at all.

espzav, 2024-02-08:
We have example here: https://github.com/espressif/esp-bsp/blob/master/examples/display_camera/main/display_camera.c which is working properly with our BSP. 

Pins `pin_sccb_sda`, `pin_sccb_scl` are not filled, because the I2C can be used for more purposes in BSP and it is initialized inside BSP instead of inside camera driver.

MichelBerg, 2024-02-08:
Ah ok with this example it makes sense. I was just confused, because I thought the BSP is ready to use for my case.
Thanks for the support.

tore-espressif, 2024-02-20:
@MichelBerg is there any outstanding issue, or can we close this?

MichelBerg, 2024-02-20:
We can close this. Thank you!

closed 2024-02-20 by MichelBerg