espressif/esp-bsp #297: Use esp_lcd_ili9341 Appears to be a red and yellow reversal

yel-best, 2024-02-21:
**Use esp_lcd_ili9341 last version library
Use ESP-IDF 5.0.2 version  library
Use LVGL 8.3.11 version  library**

**The screen used is ILI9341 IPS screen
The touch chip is FT6336**

After I configure all the parameters and initialize LVGL, the colors seem to be inverted when I light up the screen. Red turns to yellow, and yellow turns to red. The effect picture is as follows, but the blue is not inverted. I don’t know what it is. Why, can you give me some help?

[image]

Introduction to the screen I use: 
http://www.lcdwiki.com/3.2inch_IPS_SPI_Module_ILI9341

thx

yel-best, 2024-02-21:
> @yel-best Please try to enable or disable option: `CONFIG_LV_COLOR_16_SWAP=y` in `menuconfig`

Thanks, I have updated the description of the question, can you help me take a look?

yel-best, 2024-02-21:
> @yel-best Please try to enable or disable option: `CONFIG_LV_COLOR_16_SWAP=y` in `menuconfig`

`CONFIG_LV_COLOR_16_SWAP `is even weirder when disabled, so I enabled `CONFIG_LV_COLOR_16_SWAP=y `by default

[image]

yel-best, 2024-02-21:
Updates

Since ILI9341 IPS LCD is used

I adjusted it to `disable CONFIG_LV_COLOR_16_SWAP`

Add code:
`esp_lcd_panel_invert_color(lcd_panel, true);`

**Because we see that the inverted color address is 0x21, it should be sufficient.**

```
static esp_err_t panel_ili9341_invert_color(esp_lcd_panel_t *panel, bool invert_color_data)
{
     ili9341_panel_t *ili9341 = __containerof(panel, ili9341_panel_t, base);
     esp_lcd_panel_io_handle_t io = ili9341->io;
     int command = 0;
     if (invert_color_data) {
         command = LCD_CMD_INVON;
     } else {
         command = LCD_CMD_INVOFF;
     }
     ESP_RETURN_ON_ERROR(esp_lcd_panel_io_tx_param(io, command, NULL, 0), TAG, "send command failed");
     return ESP_OK;
}
```

[image]

After selecting the result, the color should be normal according to the fuzzy judgment, but the interface will be blurry. How to solve this problem of fuzzy interface? what is the reason?

espzav, 2024-02-21:
This is still not right color settings. It seems, that you should change vendor initialization data: https://github.com/espressif/esp-bsp/blob/05d471ac4a65ea84c6d8bf1c83250b58f1a53beb/components/lcd/esp_lcd_ili9341/include/esp_lcd_ili9341.h#L37 these data can be different for each vendor. 

You can try this change: https://github.com/espressif/esp-bsp/commit/b35a400e47cf8a0034cf769c32cbf12fbafdc90a

Lzw655, 2024-02-21:
> > @yel-best Please try to enable or disable option: `CONFIG_LV_COLOR_16_SWAP=y` in `menuconfig`
> 
> `CONFIG_LV_COLOR_16_SWAP `is even weirder when disabled, so I enabled `CONFIG_LV_COLOR_16_SWAP=y `by default
> 
> ![d435759a5edeee06c9763108c23e291](https://private-user-images.githubusercontent.com/82205638/306534683-9d576f72-35a9-44e0-9029-b074e595028e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg1MDc0MzgsIm5iZiI6MTcwODUwNzEzOCwicGF0aCI6Ii84MjIwNTYzOC8zMDY1MzQ2ODMtOWQ1NzZmNzItMzVhOS00NGUwLTkwMjktYjA3NGU1OTUwMjhlLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjIxVDA5MTg1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgyMzAzNWM2ODc1N2YxYzFkMGZjOGE3ZDMzN2U0YjAxOGJiMzdjN2JkODZkNGIwNzMwOTA1OWFhOTI4YzE1ZWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.6bOrPkzUJQXK50F-9hEVHPflidnnijfB1xW2grYhh3g)

Hi @yel-best, in this situation (without using `invert_color()`), I think you can try to change the RGB elements' order as the following code:

```c
    const esp_lcd_panel_dev_config_t panel_config = {
        .reset_gpio_num = EXAMPLE_PIN_NUM_LCD_RST,      // Set to -1 if not use
#if ESP_IDF_VERSION < ESP_IDF_VERSION_VAL(5, 0, 0)      // Implemented by LCD command `36h`
        .color_space = ESP_LCD_COLOR_SPACE_BGR,
#else
        .rgb_endian = LCD_RGB_ENDIAN_BGR,
#endif
        .bits_per_pixel = 16,                           // Implemented by LCD command `3Ah` (16/18)
        // .vendor_config = &vendor_config,            // Uncomment this line if use custom initialization commands
    };
```

yel-best, 2024-02-21:
> Hi @yel-best, in this situation (without using `invert_color()`), I think you can try to change the RGB elements' order as the following code:
>
> ```c
>     const esp_lcd_panel_dev_config_t panel_config [...]

thanks , ok

However, in this case, after using inversion, the colors will appear to be correct, but the text content on the screen will be blurred.

[image]

Lzw655, 2024-02-21:
> This is still not right color settings. It seems, that you should change vendor initialization data:
> 
> https://github.com/espressif/esp-bsp/blob/05d471ac4a65ea84c6d8bf1c83250b58f1a53beb/components/lcd/esp_lcd_ili9341/include/esp_lcd_ili9341.h#L37
> 
> these data can be different for each vendor.
> You can try this change: [b35a400](https://github.com/espressif/esp-bsp/commit/b35a400e47cf8a0034cf769c32cbf12fbafdc90a)

@yel-best Ok, looks like this step should be necessary.

yel-best, 2024-02-22:
> @yel-best Ok, looks like this step should be necessary.

yes，you are right. The LCD driver I am currently using should be` ILI9341v,` not the ordinary` ILI9341` protocol, so I need to make some modifications. Directly using the cmd in the package will not display properly.

I modified the configuration and used `CONFIG_LV_COLOR_16_SWAP=y`, and finally I got the desired result

Check out my latest results

[image]

**Thank you all for your help, thank you very much  👏 **

closed 2024-02-22 by yel-best

kisvegabor, 2024-02-22:
FYI, in LVGL v9 `LV_COLOR_16_SWAP` is removed and you can call `lv_draw_sw_rgb565_swap` in the flush_cb.

yel-best, 2024-02-27:
> FYI, in LVGL v9 `LV_COLOR_16_SWAP` is removed and you can call `lv_draw_sw_rgb565_swap` in the flush_cb.

oh..thank you for your reminder, I will pay attention to this option when upgrading lvgl 9.