espressif/esp-bsp #302: esp_lcd_touch GT911 Initializes but the touches is not read

decaldaswcl, 2024-02-29:
I'm using an ILI9488 display  with a GT911 touch, using the libs atanisoft/esp_lcd_ili9488 for the display and espressif/esp_lcd_touch_gt911 for the touch. 

The display is working normally, as it can read the ID according to the monitor track below, however the touch is initialized but touching the screen does not work

> I (800) gpio: GPIO[26]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2
> I (800) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
> I (820) GT911: TouchPad_ID:0x39,0x31,0x31
> I (820) GT911: TouchPad_Config_Version:0

Analyzing the GT911 datasheet, I verified that the interrupt pin should have 3.3v (High) after the chip is initialized, but oscillates at 2.4v. Reading the datasheet, this behavior indicates that the chip entered spleep mode, but this worried me because I never used the esp_lcd_touch_gt911_enter_sleep function.

According to the image below in the datasheet, to exit this mode, a high level is required on the INT or RESET pin.

[image]

I did the test by manually applying the high level to the INT pin and the reading of the coordinates worked and the voltage level returned to 3.3v.
Doing the same using the esp_lcd_touch_exit_sleep() function didn't work, now I can't get it to work correctly.

I'm trying to find a way to get around this in esp_lcd_touch_gt911.c, but if anyone can help me I'd be grateful.

decaldaswcl, 2024-02-29:
Looking again at esp_lcd_touch_gt911.c, I verified that the initialization is not correct in accordance with the datasheet.

**Current GT11 init script**
> esp_lcd_touch_gt911.c -> esp_lcd_touch_new_i2c_gt911()

INT pin is set as INPUT in line 91.
RESET pin is set as OUTPUT in line 107.
Is RESET is set OUTPUT how level HIGH in line 117
Read status of GT11 in line 121

The problem occurs when the INT pin is set to INPUT. Because as shown in the image below, INT must initially be set as OUTPUT with a high level so that the GT911 can choose the I2C address. and after that the address is chosen by defining the RESET pin as HIGH to 0x5D and LOW to 0x14, however there is no written script for this.

[image]

Only after choosing the address should the INT pin be set to LOW and defined as output so that the GT911 understands that it can start reading the touch.

**Temporary solution with low intervention** (worked for me)

In
> esp_lcd_touch_gt911.c -> esp_lcd_touch_new_i2c_gt911()

cut if in line 91 that defines the INT with INPUT and paste it before the touch_gt911_read_cfg() function.

In 
> main code

call the esp_lcd_touch_exit_sleep() function after the esp_lcd_touch_new_i2c_gt911() function.

This may be useful until the next corrected release.

For a new release I also see that it would be necessary to include a new function that defines the correct initialization routine for the GT911 address because the way it was done only address 0x14 can be used.

espressif2022, 2024-03-01:
@decaldaswcl ，you are right. The reset and interrupt pins are used during power-on initialization to set the required I2C address. However, this component did not perform this operation.
We chose to poll addresses A or B instead, it's not very good, thanks for the suggestion.

VojtechBartoska, 2024-03-13:
@espressif2022 Will you work on this?

espressif2022, 2024-03-21:
> @espressif2022 Will you work on this?

Sorry, I missed this ISSUE. I'll think about how to add an initialization interface.

espzav, 2024-08-06:
This feature is available in latest version of the component. Closing.

closed 2024-08-06 by espzav