espressif/esp-bsp #309: LVGL v9 example for ESP32_S3_LCD_EV_BOARD

kisvegabor, 2024-03-12:
Hi,

I'm preparing an LVGL demo ESP32_S3_LCD_EV_BOARD for Embedded World. I wonder if there is a ready to use LVGL v9 based example project for this board.

Just to share my excitement, the demo contains some Lottie animations too, and I'm very curious how they run the  this board. 😁

kisvegabor, 2024-03-13:
I've just discovered the `lvgl9` branch, but is still contains the v8 LVGL API in [esp32_s3_lcd_ev_board.h](https://github.com/espressif/esp-bsp/blob/master/bsp/esp32_s3_lcd_ev_board/include/bsp/esp32_s3_lcd_ev_board.h#L382).

espzav, 2024-03-14:
Hi @kisvegabor, we have prepared all boards to LVGL9 except ESP32_S3_LCD_EV_BOARD. This board is not supported in `ESP LVGL port` for now and we want to add support it soon. I am so sorry.

kisvegabor, 2024-03-14:
Oh, what an unlucky coincidence. :slightly_frowning_face: 
Can you estimate when will it be updated?

espzav, 2024-03-14:
> Can you estimate when will it be updated?

I would like add it into plan to next quarter (Q2). We still don't support LVGL9 in default, we are waiting for some changes for platforms, if we enable LVGL9 now, we will break all examples. We want to handle lot of things inside LVGL port instead of in example `sdkconfig.defaults`.

kisvegabor, 2024-03-14:
A big distributor company (I'm not sure if I can mention the name here) asked me to prepare a demo with that board for Embedded World hence Q2 is too late for this case. 

Anyway, I think we only need to change the LVGL API calls, so we can update the project locally.

sukesh-ak, 2024-03-14:
> ESP32_S3_LCD_EV_BOARD

@kisvegabor Which controller does it use?

I just completed a sample using the latest `esp_lvgl_port 2` with `lvgl 9.x`
https://github.com/sukesh-ak/IDF5-ESP_LCD-LVGL9

kisvegabor, 2024-03-14:
> Which controller does it use?

How can I check it? I don't have the board with me now. Maybe I can just read the part number from an IC? :slightly_smiling_face:

sukesh-ak, 2024-03-15:
> > Which controller does it use?
> 
> How can I check it? I don't have the board with me now. Maybe I can just read the part number from an IC? 🙂

The Espressif page for the board talks about different displays. So, I am guessing you can interchange displays for use.
https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp32s3/esp32-s3-lcd-ev-board/user_guide.html
[image]

So you need to check with your customer/vendor.

espzav, 2024-03-15:
> Which controller does it use?

This board has RGB connection between chip and LCD (there is no controller for data). This is the reason, why it is not inside ESP LVGL port yet. It is more complicated than classic LCD controller.

kisvegabor, 2024-03-16:
I have the 800x480 board. We will take look next week.

espressif2022, 2024-03-18:
@espzav @kisvegabor , I have uploaded ESP32_S3_LCD_EV_BOARD's support for the latest esp_lvgl_port V9 in this [branch.](https://github.com/espressif2022/esp-bsp/tree/feature/add_RGB_support) 
It is still being updated. You can use the display_lvgl_demos routine to test.

kisvegabor, 2024-03-18:
Thank you @espressif2022!

I've tried it, but got this error:
```
CMake Error at /home/kisvegabor/esp/esp-idf/tools/cmake/build.cmake:268 (message):
  Failed to resolve component 'esp_lvgl_port'.
Call Stack (most recent call first):
  /home/kisvegabor/esp/esp-idf/tools/cmake/build.cmake:304 (__build_resolve_and_add_req)
  /home/kisvegabor/esp/esp-idf/tools/cmake/build.cmake:311 (__build_expand_requirements)
  /home/kisvegabor/esp/esp-idf/tools/cmake/build.cmake:599 (__build_expand_requirements)
  /home/kisvegabor/esp/esp-idf/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:12 (project)

-- Configuring incomplete, errors occurred!
```

Am I missing something?

espressif2022, 2024-03-19:
> Thank you @espressif2022!
>
> I've tried it, but got this error:
>
> ```
> CMake Error at /home/kisvegabor/esp/esp-idf/tools/cmake/build.cmake:268 (message):
>   Failed to resolve component 'esp_lvgl_port'. [...]

Hello, did you compile it in the example directory of esp-bsp?
The override_path here will give priority to the local path. Is the path wrong? I tested normally under the `examples/display_lvgl_demos` path.
By modifying `idf_component.yml` under `esp_lvgl_port`, both V8 and V9 can be compiled.

kisvegabor, 2024-03-19:
I cloned your repo again to have a clean start and now it compiles butuses v8.3.11. 

I've changed [this line](https://github.com/espressif2022/esp-bsp/blob/feature/add_RGB_support/components/esp_lvgl_port/idf_component.yml#L7) to `version: "9"`, cleaned the project, but it still uses LVGL v8.

espressif2022, 2024-03-20:
> I cloned your repo again to have a clean start and now it compiles butuses v8.3.11.
> 
> I've changed [this line](https://github.com/espressif2022/esp-bsp/blob/feature/add_RGB_support/components/esp_lvgl_port/idf_component.yml#L7) to `version: "9"`, cleaned the project, but it still uses LVGL v8.

I'm sorry, some new commits was submitted last night. The version dependency is determined by the board now, [here](https://github.com/espressif2022/esp-bsp/blob/33ff68e445dfec2c311e4517713a442c372c09ad/bsp/esp32_s3_lcd_ev_board/idf_component.yml#L26).
If the pull is still wrong, you can clean the local `dependencies.lock` file and then `fullclean` it, It should work.

@espzav ,hello, I have a time difference here, so I can't reply in time. Can you help me confirm it?? 

[image]

kisvegabor, 2024-03-20:
I've enabled LVGL v9 in `bsp/esp32_s3_lcd_ev_board/idf_component.yml`, cleaned, built, but still LVGL v8 is used.
[image]

Are there any extra steps needed to switch version?

espressif2022, 2024-03-20:
Has the `dependencies.lock` file been deleted? If the local version is v8 and the dependency here is ^9, an error should be reported.
Also try deleting the `managed_components` and `build` folders？

kisvegabor, 2024-03-20:
Ah, shame on me. I forgot the delete the `dependencies.lock`... :facepalm: 

Now compiles with v9. Thank you! 

Today evening I'll try using my LVGL fork and compile it with some Lottie animations.

kisvegabor, 2024-03-21:
I'm getting closer. Compiled and flushed, but the initialization fails on an assert:
```c
[08:48:19:491] <0x1b>[0;32mI (871) LVGL: Starting LVGL task<0x1b>[0m␍␊
[08:48:19:499] <0x1b>[0;32mI (874) bsp_probe: Detect sub_board3 with 800x480 LCD (ST7262), Touch (GT1151)<0x1b>[0m␍␊
[08:48:19:504] <0x1b>[0;32mI (882) bsp_sub_board: Initialize RGB panel<0x1b>[0m␍␊
[08:48:19:537] ␍␊
[08:48:19:537] assert failed: lvgl_port_add_disp esp_lvgl_port_disp.c:77 (disp_cfg->io_handle != NULL)␍␊
```

So the `io_handle` is not set in case of the 800x480 board. Which is really the case as the `io_handle` is [not set for this board type](https://github.com/espressif2022/esp-bsp/blob/33ff68e445dfec2c311e4517713a442c372c09ad/bsp/esp32_s3_lcd_ev_board/src/bsp_sub_board.c#L165-L208) but it is [set for the other one](https://github.com/espressif2022/esp-bsp/blob/33ff68e445dfec2c311e4517713a442c372c09ad/bsp/esp32_s3_lcd_ev_board/src/bsp_sub_board.c#L159-L162).

espressif2022, 2024-03-21:
> I'm getting closer. Compiled and flushed, but the initialization fails on an assert:
>
> ```c
> [08:48:19:491] <0x1b>[0;32mI (871) LVGL: Starting LVGL task<0x1b>[0m␍␊
> [08:48:19:499] <0x1b>[0;32mI (874) [...]

I test it with `board2` (800*480), I'll check the `board3` (800*480) now.

kisvegabor, 2024-03-21:
I've just commented out the assert and now it works.

There is [a compiler bug](https://github.com/jcmvbkbc/gcc-xtensa/issues/14#issuecomment-1994490029) which prevents from building Lottie too, however the new demo is already working.

kisvegabor, 2024-03-21:
I'm about to add a 800x480 (16 bit) wallpaper, but it overflows the app partion. What is the recommended way to workaround it (if possible)? I know that I can create a CSV, but I'm not sure if this the best way.

espzav, 2024-03-21:
The recommended way is use own partition CSV. For example like here: https://github.com/espzav/Simple-LVGL-Player/blob/master/partitions.csv 
and then set this `CONFIG_PARTITION_TABLE_CUSTOM=y` in menuconfig.

kisvegabor, 2024-03-21:
Thank you, it worked! 
I just needed to lower the size of the app section to fit into 2MB:
```
# Name,   Type, SubType, Offset,  Size
nvs,      data, nvs,     0x9000,  0x4000
phy_init, data, phy,     0xd000,  0x1000
factory,  app,  factory, 0x10000, 0x1F0000,
```

espzav, 2024-03-21:
If you have bigger flash, you can change it in menuconfig to  `CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y` or 16MB. 
You can put into CSV value like this: `factory,  app,  factory, 0x10000, 2M,` instead of hexa.

tore-espressif, 2024-06-25:
Our examples support LVGL v9 by default now!

closed 2024-06-25 by tore-espressif