espressif/esp-bsp #315: lvgl 9.x and esp_lvgl_port with ST7796 8bit Parallel

sukesh-ak, 2024-03-26:
I have built a sample using lvgl 9.x + esp_lvgl_port with ST7796 SPI and got it working.

Now I am trying to get ST7796 8bit Parallel working. 
Could not find a suitable sample to borrow code. And the code snippet mentioned in [README.md](https://github.com/espressif/esp-bsp/blob/master/components/lcd/esp_lcd_st7796/README.md) of `esp_bsp` is incomplete.

Device being tested on is `WT32-SC01 Plus` which is `ESP32-S3`.

Need help since the display is showing only colored dots for now. Sample is here on `ST7796P` branch.
https://github.com/sukesh-ak/IDF5-ESP_LCD-LVGL9/tree/ST7796P

sukesh-ak, 2024-04-02:
Hi @Lzw655 

> 1. `HMI_PIN_NUM_CS` pin is set to `-1`.

CS PIN is already -1 in the working samples.

Here I am trying to make it work using `esp_lcd` and `esp_lvgl_port`.
Screenshot talks about custom initialization commands. Not sure if they are required?

This was done using `LovyanGFX` drivers
https://github.com/sukesh-ak/ESP32-TUX/blob/master/main/devices/conf_WT32SCO1-Plus.h

sukesh-ak, 2024-04-02:
@espzav Can you take a look and help?
I don't want to switch back to `LovyanGFX`. I prefer to use `esp_lvgl_port` & `esp-bsp` going forward.

espzav, 2024-04-02:
HI @sukesh-ak,
I apologize, I haven't got HW for this screen, but I see in upper comment, you should change vendor config. You can do it here: https://github.com/sukesh-ak/IDF5-ESP_LCD-LVGL9/blob/1532084ed91ca553b749b49b173dbdce1fffd4ad/main/main.c#L286 same as in comment https://github.com/espressif/esp-bsp/issues/315#issuecomment-2021783381.
It should work with `esp_lvgl_port`. I don't see the `esp_bsp` in your code.

sukesh-ak, 2024-04-02:
> Hi @sukesh-ak, I apologize, I haven't got HW for this screen, but I see in upper comment, you should change vendor config. You can do it here: https://github.com/sukesh-ak/IDF5-ESP_LCD-LVGL9/blob/1532084ed91ca553b749b49b173dbdce1fffd4ad/main/main.c#L286 same as in comment [#315 (comment)](https://github.com/espressif/esp-bsp/issues/315#issuecomment-2021783381). It should work with `esp_lvgl_port`. I don't see the `esp_bsp` in your code.

I was looking at this example for reference since its for the same controller `ST7796 Parallel`.

sukesh-ak, 2024-04-03:
@espzav / @Lzw655 
Thank you for the help. I got it working without the vendor specific additions.

Also I have built it as a `esp_bsp` and now available here.
https://github.com/sukesh-ak/BSP-IDF5-ESP_LCD-LVGL9

Hope to add support for other popular display devices.

espzav, 2024-04-04:
@sukesh-ak Great! Could we close the issue now?

sukesh-ak, 2024-04-04:
> @sukesh-ak Great! Could we close the issue now?

Yes closing now.

closed 2024-04-04 by sukesh-ak