espressif/esp-bsp #316: PRIV_REQUIRES onewire_bus

morgana2313, 2024-03-26:
Using ds18b20  gives an error:

```
components/espressif__ds18b20/src/ds18b20.c:10:10: fatal error: onewire_bus.h: No such file or directory
   10 | #include "onewire_bus.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
[...]
BUG: onewire_bus.h found in component espressif__onewire_bus which is already in the requirements list of espressif__ds18b20
```

Fix: components/ds18b20/CMakeLists.txt

```diff
idf_component_register(SRCS "src/ds18b20.c"
                       INCLUDE_DIRS "include"
+                        PRIV_REQUIRES onewire_bus
)
```

suda-morris, 2024-04-07:
Hi @morgana2313 May I know how you use the onewire_bus and the ds18b20 components? Are you using them with [component manager](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-component-manager.html)? In the manifest file, we have declared the [dependency] (https://github.com/espressif/esp-bsp/blob/master/components/ds18b20/idf_component.yml#L5) on the onewire_bus.

tore-espressif, 2024-06-10:
Closing due to lack of feedback. Feel free to reopen

closed 2024-06-10 by tore-espressif

morgana2313, 2024-06-10:
Dank voor uw bericht.

Ik ben afwezig tot 16 juni 2024 en lees sporadisch mijn mail.

Voor dringende zaken kunt u contact opnemen met mijn collega's via 070 891 9 891 of via ***@***.***