espressif/esp-bsp #318: Using local lv_conf.h with custom esp_bsp as a component

sukesh-ak, 2024-04-09:
Sample project
https://github.com/sukesh-ak/BSP-IDF5-ESP_LCD-LVGL9

Usually for `lvgl` to work with local custom `lv_conf.h`, I use the following in the `CMakeLists.txt` which is in the root.
This always worked but now with `bsp` in the components folder, this does not work anymore.

```bash
#LVGL custom config file setup
idf_build_set_property(COMPILE_OPTIONS "-DLV_LVGL_H_INCLUDE_SIMPLE=1" APPEND)
idf_build_set_property(COMPILE_OPTIONS "-I../main" APPEND)
```

### Not working (default sdkconfig)
```bash
#
# LVGL configuration
#
CONFIG_LV_CONF_SKIP=y
# CONFIG_LV_CONF_MINIMAL is not set
```

### Working
> [!NOTE]  
> If we uncheck `Check this to not use custom lv_conf.h` under `Component config > LVGL configuration` with `idf.py menuconfig`.

```bash
#
# LVGL configuration
#
# CONFIG_LV_CONF_SKIP is not set
# CONFIG_LV_CONF_MINIMAL is not set
```

So the question is, how to make the above work without the user going into `idf.py menuconfig`, just to uncheck the `Check this to not use custom lv_conf.h` ?

espzav, 2024-04-10:
Hello @sukesh-ak ,
when you are using LVGL with Espressif chips, there is only `menuconfig` solution for LVGL configuration. For default settings is recommended to use `sdkconfig.defaults`. 

It seems, that you have same question as we from Espressif. We are solving this here: https://github.com/lvgl/lvgl/issues/6010 
We want to set some default settings from component and not need for it `sdkconfig.defaults` in examples/demos.

espzav, 2024-04-10:
@kisvegabor it seems, that it is same issue as here https://github.com/lvgl/lvgl/issues/6010

kisvegabor, 2024-04-16:
Thanks for tagging me. I have a few items on my todo list, but after that we can prioritize https://github.com/lvgl/lvgl/issues/6010