espressif/esp-bsp #323: Can the ds18b20_device_t struct definition be moved from ds18b20.c to a header file ?

michal-olszanowski, 2024-04-28:
### Related area

Component espressif/ds18b20

### Hardware specification

esp32c3

### Is your feature request related to a problem?

I'd like to use the ds18b20_device_hande_t type definition (the addr member) in my code, however, the struct is defined in .c and not .h:

ds18b20_device_handle_t sensor = ds18b20_sensors[i];
onewire_device_address_t address = sensor->addr;

ESP_LOGI( log_tag(), "Sensor %08" PRIx32 "%08" PRIx32 " (%s) reports %.3f°C",
    (uint32_t)(address >> 32),
    (uint32_t)address,
    "DS18B20",
    measured_temps[ i]);

The compiler is reporting:

C:/Users/m/Dev/esp/pump-driver/components/SensorService/SensorService.cpp:112:54: error: invalid use of incomplete type 'struct ds18b20_device_t'
  112 |             onewire_device_address_t address = sensor->addr;
      |                                                      ^~
In file included from C:/Users/m/Dev/esp/pump-driver/components/SensorService/SensorService.h:14,
                 from C:/Users/m/Dev/esp/pump-driver/components/SensorService/SensorService.cpp:1:
C:/Users/m/Dev/esp/pump-driver/managed_components/espressif__ds18b20/include/ds18b20.h:19:16: note: forward declaration of 'struct ds18b20_device_t'
   19 | typedef struct ds18b20_device_t *ds18b20_device_handle_t;
      |                ^~~~~~~~~~~~~~~~

### Describe the solution you'd like

Add to ds18b20_types.h:

typedef struct ds18b20_device_t {
    onewire_bus_handle_t bus;
    onewire_device_address_t addr;
    uint8_t th_user1;
    uint8_t tl_user2;
    ds18b20_resolution_t resolution;
} ds18b20_device_t;

### Describe alternatives you've considered

I modifed the header locally in my project but would like to relay fully on the official ds18b20 component release. I do not know   other workarounds, I'd appreciate any suggestions as an alternative. 

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

suda-morris, 2024-05-08:
@michal-olszanowski What information do you want to be exposed from the `ds18b20_device_t`? The address? You can get the device address from `onewire_device_t::address`

michal-olszanowski, 2024-05-19:
Yes, I'm looking for the device address. 
Correct me if I'm wrong, but one_wire_device_t::address is only available when we are scanning the onewire bus using the onewire_device_iter_handle_t. At this point we are adding a onewire_device_t to ds18b20_device_handle_t devices[] and to make use of the device address we would need to add the device to an array of onewire_device_t devices[], which seems unnecessary, given the address is also available in ds18b20_device_t. I need an address later when we read temperatures for the devices in ds18b20_device_handle_t[].

closed 2025-04-08 by suda-morris

Kainarx, 2025-07-31:
Hi @arc12 

Currently, `ds18b20_device_handle_t` is an opaque pointer, shielding internal fields such as `addr`. Exposing the full struct in the header ties users to its layout, making future internal changes breaking and harder to maintain. Users might depend on fields that we may want to modify or remove later.