espressif/esp-bsp #326: espressif_esp_lvgl_port fails to retrieve the lvgl version if the lvgl__lvgl dependency is moved to the components directory.

i400s, 2024-05-02:
### Board

ESP32-C6-DevKitC

### Hardware Description

Adafruit 2090

### IDE Name

VSCode

### Operating System

Debian

### Description

Due to needing to make a change to the lvgl version 9.1.0 lv_scale.c file in the managed_components/lvgl__lvgl/ directory I moved the whole directory to components/lvgl__lvgl/ and made the change there. When I tried to recompile the project the compile failed.
```
/home/wilsonj/Documents/foldered/vscode/esp-idf/i400s__/lcd-2090/examples/managed_components/espressif__esp_lvgl_port/src/lvgl8/esp_lvgl_port_disp.c:39:5: error: unknown type name 'lv_disp_drv_t'
```

I noticed that the error was being generated for the lvgl8 directory even though I was using lvgl 9.1.0.

Further investigation showed that the LVGL version reported in the build was blank.
``` 
-- Adding linker script /mnt/md8/_development/esp/esp-idf/components/soc/esp32c6/ld/esp32c6.peripherals.ld
-- LVGL version:
-- ESP_LCD_ILI9341: 2.0.0
```

I wondered if this might have been due to the espressif_esp_lvgl_port being in a different directory to the lvgl__lvgl component so I moved the espressif_esp_port component to the components/ directory. That failed to resolve the issue.

The issue is that the CmakeLists.txt in the esp_lvgl_port fails to get the lvgl__lvgl COMPONENT_VERSION property correctly.

Manually adding in a set(lvgl_ver "9.1.0") to the CMakeLists.txt file allows the build process to work correctly.
```
#Get LVGL version
idf_component_get_property(lvgl_ver lvgl__lvgl COMPONENT_VERSION)
if(lvgl_ver EQUAL "")
    idf_component_get_property(lvgl_ver lvgl COMPONENT_VERSION)
endif()

set(lvgl_ver "9.1.0")
```

Interestingly the following change to the CmakeLists.txt file didn't work
```

#Get LVGL version
idf_component_get_property(lvgl_ver lvgl__lvgl COMPONENT_VERSION)
if(lvgl_ver EQUAL "")
    idf_component_get_property(lvgl_ver lvgl COMPONENT_VERSION)
endif()
if(lvgl_ver EQUAL "")
    set(lvgl_ver "9.1.0")
endif()
```
So I'm assuming the idf_component_get_property is getting something, just not the correct value.

### Sketch

```cpp
No sketch as this is a build/location issue.
```

### I have checked existing issues, README.md and ESP32 Forum

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

kumekay, 2024-05-31:
@i400s Could you please try the development branch of the IDF component manager, where this issue is addressed? https://github.com/espressif/idf-component-manager/commit/9d0f228482431cfd113358e9197596043c297522

Instructions https://github.com/espressif/idf-component-manager?tab=readme-ov-file#installing-a-development-version-of-the-component-manager

i400s, 2024-05-31:
Hi there @kumekay. Can confirm that after deleting the build directory, managed_components directory, and dependencies.lock file and then having my patched lvgl__lvgl within the components directory that once the espressif__esp_lvgl_port is auto-downloaded to the managed_components directory that the build is successful with the development branch of the component-manager.

I did have to modify the ```.espressif/espidf.constraints.v5.3.txt`` file by omitting the requirement for the component manager due to this error: ```Requirement 'idf-component-manager~=1.2' was not met. Installed version: 2.0.0.dev0```

But apart from that it works. Thank you.

Will you close this issue, or should I? (I'm still very new to this).

kumekay, 2024-05-31:
@i400s Thank you for checking! 

So far we only updated the constraints for the `master` branch of ESP-IDF  (v5.4-dev currently) to allow the component manager 2.0. We will update it later for other branches a bit later. 

Usually the issues are closed automatically with the commit message, but since this issue not in the [idf-component-manager](https://github.com/espressif/idf-component-manager) repo, where the actual fix is, it wasn't mentioned and closed, so feel free to close it.

Thank you once again for reporting

i400s, 2024-05-31:
You're welcome. And thanks for the excellent work, and a great ecosystem.

closed 2024-05-31 by i400s

tank104, 2024-06-14:
Thanks so much for this thread - saved me a load of time!

tank104, 2024-06-14:
Actually I still have an issue where lvgl__lvgl is getting downloaded to managed_components.  I have removed it from idf_component.yml - but guessing its getting added as a dependency of esp_lgvl_port.  

Weird I am sure I was just getting build errors yesterday with the new component only..

tank104, 2024-06-14:
Ahhh ok I was doing a git clone today of lvgl instead of moving the folder from managed_components.  That was the difference.

How do weget fully latest branch of that dependency without it downloading again to managed_components.  Somehow it detects the git as different, but I can't figure out how looking at the diff of both folders.  Any idea @i400s ?

i400s, 2024-06-15:
Hi @tank104,

There seems to be another "bug" with the dependency resolution but I can't quite figure out what causes it or how it decides what is "this already exists in components which I'll use, so I don't need to download it to managed_components". It might be that our issues are related (or our understandings incorrect).

I've not really had time to look into it as other things came up, but I'm hoping to look in to it over the next week so I can raise a bug report for the maintainers. I also need to make sure I'm using the correct/latest version of component manager (just in case its something that has already been resolved!) and that I've not done something silly in the way I created my own components or the idf_components.yml file (both distinct possibilities).

Your best bet is to raise a bug report on the issue on the [https://github.com/espressif/idf-component-manager/issues](url) repository detailing:

What you move from where (and why).
What exists in the directories before the build starts.
What ends up in the directories after it pulls its own dependencies.
What you think should end up in those directories (and which should be used).
And what it downloads and why you think that is wrong.
And I'd also include your dependencies.lock file in the report (as that I believe is used to decide if a dependency needs to be [re-]downloaded or not).

Its also worth deleting the "managed_components" directory, the "dependencies.lock" and "build" as those are all built/created as part of the build process to "clean it out" so to speak. (Obviously make sure you have backups!)

Sorry I couldn't help more as I'm still very much in the learning stage of using espressif development.

tank104, 2024-06-16:
@i400s by trial an error it seems to be idf_component.yml that decides whether it gets lvgl again to managed_components.  If I pull latest from git for lvgl, then swap out that file, it doesn't re-download lvgl to managed_components and uses components folder.

I think we are at the same level on the library though - don't fully understand it yet.  learning :)

kumekay, 2024-06-17:
@tank104 If you manually clone lvgl to `components` directory of your project, then IDF component manager won't download it to the `managed_components`.
Or did you add it as a `git` dependency in `idf_component.yml`? Could you please share your `idf_component.yml`?

tank104, 2024-06-30:
Sorry for slow reply @kumekay was away.

I got cloned it to components, and then when building it would also download again to managed_components.

idf_component.yml looks like:
```

## IDF Component Manager Manifest File
dependencies:
  espressif/esp_lcd_touch_cst816s: "^1.0.3"
  espressif/esp_lcd_gc9a01: "^2.0.0"
  espressif/esp_lvgl_port: "^2.2.2"
  ## Required IDF version
  idf:
    version: ">=4.1.0"
  # # Put list of dependencies here
  # # For components maintained by Espressif:
  # component: "~1.0.0"
  # # For 3rd party components:
  # username/component: ">=1.0.0,<2.0.0"
  # username2/component2:
  #   version: "~1.0.0"
  #   # For transient dependencies `public` flag can be set.
  #   # `public` flag doesn't have an effect dependencies of the `main` component.
  #   # All dependencies of `main` are public by default.
  #   public: true

```

i400s, 2024-07-01:
Hi @tank104.

After cloning lvgl to `components/lvgl__lvgl` (double underscore) you will probably need to modify the `idf_components.yml` in that directory to include the line `version: 9.1.1` (the version is dependent on the manifest requirement from the `idf_components.yml` file, most likely the one in the lvgl__port, that is ultimately responsible for pulling in the `lvgl__lvgl` component).

I used 9.1.1 as that is .1 greater than the current version on the component directory (and my requirement is for the latest master due to a change that was made in lvgl). If the espressif component, on the component directory, is updated to something higher then that will then obviously need to be updated to something bigger again depending on how long the project takes. Note: the lvgl__port requirement is currently lvgl__lvgl >8, <10.

tank104, 2024-07-02:
Thanks - that was it!
I guess it just doesn't have a version (latest master).  So if I had cloned a specific version it would have been ok.