espressif/esp-bsp #335: Is there a library for the PCA9554 io_expander?

karunt, 2024-05-31:
Found a library for the TCA9554 io_expander, but is there also one for the PCA9554 expander? Using it on an Adafruit Qualia board with ESP32-S3 to connect a 40 pin LCD display. The SPI interface for the LCD display is connected to the PCA9554 expander. But I'm unable to find a library for the PCA9554 expander. So my question is if I can use the TCA9554 library instead. If yes, any modifications I need to make? From what I understand, and my understanding is basic at best, the only difference between the two io_expanders is the minimum supply voltatge setting (1.65 for the TCA9554 and 2.3 for the PCA9554). If that has to be changed somewhere in the TCA9554 library to be able to use it for the PCA9554 expander, where should that be changed?

Lzw655, 2024-06-01:
Hi @karunt, 

I have reviewed the datasheets for both the PCA9554 and TCA9554. They share the same I2C address and command bytes. So, I think you can use the TCA9554 library directly to control the PCA9544. And please notice the hardware decides the actual I2C address.

* The I2C address and command types of TCA9554:
[image]
[image]

* The I2C address and command types of PCA9554:
[image]
[image]

karunt, 2024-06-01:
@Lzw655 Thanks. What about the difference in supply voltage range between the 2 expanders? Does that have to be modified somewhere in the TCA9554 library so it doesn't create any issues for the PCA9554 expander?

Lzw655, 2024-06-01:
Sorry, I don't know much about hardware, but I believe there is no need to modify the software side, as both chips do not have any voltage-setting registers.

tore-espressif, 2024-08-26:
@karunt Is there anything else? Or can we close this issue?

karunt, 2024-10-09:
The TCA9554 component uses the driver/i2c component. ESP is deprecating this component and is recommending moving to the driver/i2c_master component. Is the TCA9554 component being upgraded to reference the new driver/i2c_master component?

karunt, 2024-10-09:
You can close this issue