espressif/esp-bsp #337: How to work with touch interrupts? Need an example

ddprog, 2024-06-10:
### Board

ESP32-8048S043C (ESP32-S3-WROOM1-N16R8V)

### Hardware Description

RGB 565 Display 800*480
Touch GT911

### IDE Name

VS-Code with ESP-IDF Plugin

### Operating System

Windows 10

### Description

An example is required to understand how to work with touchscreen interrupts. I want to stop polling the touch in LVGL and resume it when interrupted. But I don’t quite understand how to do this. I give an example of code on how I initialize the touchscreen and turn off its polling timer. It's not clear whether the callback function is suitable for my case...

### Sketch

```cpp
static lv_indev_drv_t indev_drv;    // Input device driver (Touch)
    lv_indev_drv_init(&indev_drv);
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.disp = disp;
    indev_drv.read_cb = example_lvgl_touch_cb;
    indev_drv.user_data = tp;

    lv_indev_drv_register(&indev_drv);
    // Stop polling indev
    lv_timer_del( indev_drv.read_timer ); // TODO
    // Callback from indev
    esp_lcd_touch_register_interrupt_callback(&indev_drv, callback_indev); //TODO
```

### I have checked existing issues, README.md and ESP32 Forum

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

espzav, 2024-06-11:
Hello @ddprog,
we haven't got example for this feature, but it is implemented in last version of LVGL port component: https://components.espressif.com/components/espressif/esp_lvgl_port/
I can recommend to use this component.

If you don't want to use it, here are some recommendations
You can see inside the component. The important think is set this mode: https://github.com/espressif/esp-bsp/blob/8d403e718188a25811e869fdafe613b36dd09007/components/esp_lvgl_port/src/lvgl9/esp_lvgl_port_touch.c#L63
And then call read in own task here: https://github.com/espressif/esp-bsp/blob/8d403e718188a25811e869fdafe613b36dd09007/components/esp_lvgl_port/src/lvgl9/esp_lvgl_port.c#L230

espzav, 2024-06-26:
Hello @ddprog,
is it ok for you? Can we close the issue?

ddprog, 2024-06-26:
Ok, thank you. Close the issue.

closed 2024-06-26 by espzav