espressif/esp-bsp #345: Can't get component esp_lcd_gc9503 to work (IDFGH-13231)

dannybackx, 2024-07-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using all the example / advice on https://components.espressif.com/components/espressif/esp_lcd_gc9503/versions/3.0.1?language=en but my display stays blank.

The hardware (ZX3D95CD01S board available from multiple sources on AliExpress) works just fine if you compile stuff with the board support package. However, that is based on old esp-idf versions and this bothers me for other (dependency) reasons.

Board reference https://www.aliexpress.com/item/1005005886057393.html?spm=a2g0o.order_list.order_list_main.23.3afe1802V0TN7o

So I'm trying to get the board to display something with the generic components, and have tried a lot of different scenarios already. The latest and cleanest is a merge between the included sample in the esp_lvgl_port component and the sample code on the esp_lcd_gc9503 page , so replacing the st7789 dependency with gc9503 code.

See https://sourceforge.net/p/esp32-s3-86-box-project-base/code/HEAD/tree/trunk/touchscreen 

This should cleanly build with a variety of recent esp-idf distributions, currently using yesterday's 5.3-rc1.

Dependency list is just stuff from the components repo.

The screen lights up a little bit (the backlight is active, I mean) but other than that remains dark.
How can I debug further ?

```
I (27) boot: ESP-IDF v5.3-rc1 2nd stage bootloader
I (27) boot: compile time Jul 11 2024 10:05:52
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=196bch (104124) map
I (118) esp_image: segment 1: paddr=000296e4 vaddr=3fc93300 size=03418h ( 13336) load
I (121) esp_image: segment 2: paddr=0002cb04 vaddr=40374000 size=03514h ( 13588) load
I (127) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=4eb08h (322312) map
I (190) esp_image: segment 4: paddr=0007eb30 vaddr=40377514 size=0bdd0h ( 48592) load
I (207) boot: Loaded app from partition at offset 0x10000
I (208) boot: Disabling RNG early entropy source...
I (219) octal_psram: vendor id    : 0x0d (AP)
I (220) octal_psram: dev id       : 0x02 (generation 3)
I (220) octal_psram: density      : 0x03 (64 Mbit)
I (225) octal_psram: good-die     : 0x01 (Pass)
I (230) octal_psram: Latency      : 0x01 (Fixed)
I (235) octal_psram: VCC          : 0x01 (3V)
I (240) octal_psram: SRF          : 0x01 (Fast Refresh)
I (246) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (252) octal_psram: BurstLen     : 0x01 (32 Byte)
I (257) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (263) octal_psram: DriveStrength: 0x00 (1/1)
I (269) MSPI Timing: PSRAM timing tuning index: 5
I (274) esp_psram: Found 8MB PSRAM device
I (279) esp_psram: Speed: 80MHz
I (296) mmu_psram: Read only data copied and mapped to SPIRAM
I (330) mmu_psram: Instructions copied and mapped to SPIRAM
I (330) cpu_start: Multicore app
I (721) esp_psram: SPI SRAM memory test OK
I (729) cpu_start: Pro cpu start user code
I (729) cpu_start: cpu freq: 240000000 Hz
I (729) app_init: Application information:
I (732) app_init: Project name:     touchscreen
I (737) app_init: App version:      1
I (742) app_init: Compile time:     Jul 11 2024 11:34:23
I (748) app_init: ELF file SHA256:  34dbbf3b3...
I (753) app_init: ESP-IDF:          v5.3-rc1
I (758) efuse_init: Min chip rev:     v0.0
I (763) efuse_init: Max chip rev:     v0.99 
I (768) efuse_init: Chip rev:         v0.2
I (772) heap_init: Initializing. RAM available for dynamic allocation:
I (780) heap_init: At 3FC9F218 len 0004A4F8 (297 KiB): RAM
I (786) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (792) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (798) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (804) esp_psram: Adding pool of 7744K of PSRAM memory to heap allocator
I (812) spi_flash: detected chip: generic
I (816) spi_flash: flash io: dio
W (820) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (834) sleep: Configure to isolate all GPIO pins in sleep state
I (840) sleep: Enable automatic switching of GPIO sleep configuration
I (00:00:00.118) main_task: Started on CPU0
I (858) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (00:00:00.132) main_task: Calling app_main()
I (00:00:00.137) gc9503.c: app_lcd_init
I (00:00:00.141) gc9503.c: Install 3-wire SPI panel IO
I (00:00:00.147) gpio: GPIO[38]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (00:00:00.157) gpio: GPIO[45]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (00:00:00.167) gpio: GPIO[48]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (00:00:00.177) lcd_panel.io.3wire_spi: Panel IO create success, version: 1.0.1
I (00:00:00.185) gc9503.c: Install GC9503 panel driver
I (00:00:00.190) gpio: GPIO[41]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
D (00:00:00.227) lcd_panel.rgb: new rgb panel(0) @0x3fcabd04, num_fbs=2, fb_size=460800, bb0 @0x0, bb1 @0x0, bb_size=0
D (00:00:00.228) lcd_panel.rgb: fb[0] @0x3c070900
D (00:00:00.232) lcd_panel.rgb: fb[1] @0x3c0e1120
I (00:00:00.236) gc9503: LCD panel create success, version: 3.0.1
D (00:00:00.532) lcd_panel.rgb: rgb panel(0) start, pclk=15000000Hz
I (00:00:00.532) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (00:00:00.537) gc9503.c: backlight: backlight set to 1
I (00:00:00.543) gc9503.c: app_lcd_init done
I (00:00:00.548) gc9503.c: app_touch_init done
I (00:00:00.553) EXAMPLE: LVGL version 9.1.0
I (00:00:00.558) LVGL: Starting LVGL task
E (00:00:00.564) lcd_panel.io.3wire_spi: Register event callbacks is not supported

```

espzav, 2024-08-06:
Hi @dannybackx,
I have looked to your code fast and it seems, there is problem in function `lvgl_port_add_disp`. You are using RGB LCD display and you should use function `lvgl_port_add_disp_rgb` like here: https://github.com/espressif/esp-bsp/blob/5d33575499110ccf0f9ff1eb58e4b17cf9b4da2b/bsp/esp32_s3_lcd_ev_board/src/esp32_s3_lcd_ev_board.c#L383C12-L383C34

dannybackx, 2024-08-19:
Thanks for pointing to that.
It doesn't change the result I'm currently getting : the backlight is the only visible thing, the display doesn't work. Touch does work.

I suspect the SPI code to configure the display isn't working.

The BSP does things like this :
```
void qmsd_rgb_spi_init() {
  spi_bus_config_t buscfg = {
    .sclk_io_num = LCD_SPI_CLK,
    .mosi_io_num = LCD_SPI_DATA0,
    .miso_io_num = -1,
    .quadwp_io_num = -1,
    .quadhd_io_num = -1,
    .max_transfer_sz = 10 * 1024,
  };
  ESP_ERROR_CHECK(spi_bus_initialize(SPI2_HOST, &buscfg, SPI_DMA_CH_AUTO));

  spi_device_interface_config_t devcfg = {
    .clock_speed_hz = SPI_MASTER_FREQ_10M,   //Clock out at 10 MHz
    .mode = 0,                               //SPI mode 0
    .spics_io_num = LCD_SPI_CS,              //CS pin
    .queue_size = 7,                         //We want to be able to queue 7 transactions at a time
  };

  ESP_ERROR_CHECK(spi_bus_add_device(SPI2_HOST, &devcfg, &g_screen_spi));
  rgb_driver_init();
```
while my code starts like this :
```
  /*
   * See https://docs.espressif.com/projects/esp-iot-solution/en/latest/display/lcd/rgb_lcd.html
   * This says use 3-wire via io_additions .
   * On the https://components.espressif.com/components/espressif/esp_lcd_panel_io_additions page
   * there's an init example. But a more complete one is on
   * https://components.espressif.com/components/espressif/esp_lcd_gc9503/versions/3.0.1 
   *
   * Additional sample https://github.com/espressif/esp-bsp/blob/master/components/lcd/esp_lcd_gc9503/README.md
   */
  ESP_LOGI(tag, "Install 3-wire SPI panel IO");
  spi_line_config_t line_config = {
    .cs_io_type         = IO_TYPE_GPIO,                 // Set to `IO_TYPE_EXPANDER` if using IO expander
    .cs_gpio_num        = LCD_SPI_CS,
    .scl_io_type        = IO_TYPE_GPIO,
    .scl_gpio_num       = LCD_SPI_CLK,
    .sda_io_type        = IO_TYPE_GPIO,
    .sda_gpio_num       = LCD_SPI_DATA0,
    .io_expander        = NULL,                        // Set to device handle if using IO expander
  };
```

ideas ?

dannybackx, 2024-09-11:
This is getting real frustrating.

The BSP from the board vendor is based on old esp-idf versions and they claim you guys move too fast. But it used esp-idf's components/esp_driver_spi/src/gpspi/spi_master.c .

The esp-bsp stuff for gc9503 doesn't seem to work, and it uses own SPI functions in managed_components/espressif__esp_lcd_panel_io_additions/esp_lcd_panel_io_3wire_spi.c .

I'm not sure what to do here. Except maybe give up on esp-idf's managed components.

tore-espressif, 2024-09-12:
Hi @dannybackx ,
Unfortunately I could not find schematic of your board, without this information it is very difficult to check that everything is setup correctly.
The GC9503 uses RGB interface for color data. This interface is usually a lot more difficult to configure compared to SPI.
From my experience with similar displays, most of the issues come from incorrect RGB interface configuration and timing. Each display requires different timings, you can find this information in the display's datasheet and the configure the driver accordingly here https://github.com/espressif/esp-idf/blob/master/components/esp_lcd/rgb/include/esp_lcd_panel_rgb.h#L128

If this doesn't help, you can check your schematic whether you need to do HW reset of the display and whether SPI is sending the data as you intend

Lzw655, 2024-09-14:
Hi @dannybackx,
Since the link is inaccessible, I can't view your code. However, after reviewing the product manual, I found that LCD_SCLK and LCD_MOSI share the same GPIOs as RGB_D0 and RGB_D1 on the LCD. 

[image]

Therefore, please ensure that when using `gc9503_vendor_config_t`, you set `.flags.auto_del_panel_io = 1 && .flags.mirror_by_cmd = 0`.

```c
    gc9503_vendor_config_t vendor_config = {
        .rgb_config = &rgb_config,
        // .init_cmds = lcd_init_cmds,      // Uncomment these line if use custom initialization commands
        // .init_cmds_size = sizeof(lcd_init_cmds) / sizeof(gc9503_lcd_init_cmd_t),
        .flags = {
            .mirror_by_cmd = 0,             // Only work when `auto_del_panel_io` is set to 0
            .auto_del_panel_io = 1,         /**
                                             * Set to 1 if panel IO is no longer needed after LCD initialization.
                                             * If the panel IO pins are sharing other pins of the RGB interface to save GPIOs,
                                             * Please set it to 1 to release the pins.
                                             */
        },
    };
```

dannybackx, 2024-09-14:
Responding to the "inaccessible" remark, do you refer to the subversion url in my message ?
If yes, I don't understand. I just checked if I could use https://sourceforge.net/p/esp32-s3-86-box-project-base/code/HEAD/tree/trunk/touchscreen/main/gc9503.c with an anonymous browser session, and with curl. Both worked.

dannybackx, 2024-09-28:
For clarity - I was originally working on a more complicated app, so I cut that down to the minimum and showed that in this thread. Tried your suggestions in the more complicated app but didn't show that in the minimal source. I did that now, still not getting anywhere.

So the minimal source at https://sourceforge.net/p/esp32-s3-86-box-project-base/code/HEAD/tree/trunk/touchscreen/ has been updated, I'm not further yet.

```
I (00:00:00.118) main_task: Started on CPU0
I (1164) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (00:00:00.132) main_task: Calling app_main()
I (00:00:00.137) gc9503.c: app_lcd_init
I (00:00:00.141) gc9503.c: Install 3-wire SPI panel IO
I (00:00:00.147) gpio: GPIO[38]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (00:00:00.157) gpio: GPIO[45]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (00:00:00.167) gpio: GPIO[48]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (00:00:00.177) lcd_panel.io.3wire_spi: Panel IO create success, version: 1.0.1
I (00:00:00.185) gc9503.c: Install GC9503 panel driver
I (00:00:00.190) gpio: GPIO[41]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
W (00:00:00.483) lcd_panel.io.3wire_spi: Delete but keep CS line inactive
I (00:00:00.484) gpio: GPIO[45]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (00:00:00.490) gpio: GPIO[48]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (00:00:00.550) gc9503: LCD panel create success, version: 3.0.1
I (00:00:00.674) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (00:00:00.674) gc9503.c: backlight: backlight set to 1
I (00:00:00.678) gc9503.c: app_lcd_init done
I (00:00:00.683) ft5x06: app_touch_init
I (00:00:00.689) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2 
I (00:00:00.699) ft5x06: app_touch_init done
I (00:00:00.703) EXAMPLE: LVGL version 9.2.0
I (00:00:00.708) LVGL: Starting LVGL task
I (00:00:00.781) EXAMPLE: app_lvgl_init: disp 0x3fc99bb8
I (00:00:00.874) main_task: Returned from app_main()
```

Lzw655, 2024-10-08:
Sorry for the late reply. Thank you very much for organizing such a minimal code example. After reviewing it, I found two issues:
1. The pins used for the LCD in the code seem to be mismatched with the schematic in the product link you provided (https://github.com/espressif/esp-bsp/issues/345#issuecomment-2345418351).
3. The `vendor_config` in the code does not set `init_cmds`. Different LCD screens from various manufacturers may require different initialization configuration parameters. The default parameters used in the GC9503 driver are compatible with the official development board, so you need to set your screen's parameters in `init_cmds`. Please refer to the [examples](https://github.com/espressif/esp-bsp/tree/master/components/lcd/esp_lcd_gc9503#example-use) in the README for guidance.

dannybackx, 2024-10-08:
Ok, so there are several versions of this board, even from the same vendor/website, and I managed to point you to the wrong spec sheet.
Look on https://en.wireless-tag.com/product-item-29.html and you'll see a more or less identical board except some of the pins are different. Also on the TR board (this link) some of the esp32 GPIOs are directly exported on the connector I use to program the board, not so on the AR board.

The TR board data sheet has a very similar setup as your screenshot shows, but with different pin numbers - the one in my include file.
[image]

Furthermore, the include file I use is derived from a source directory which I got from the merchant site at the time. I believe the lot is also at https://github.com/smartpanle/QMSD-ESP32-BSP/tree/master .

More importantly (because it works), in the same repo as the cut down example you studied, there's a directory "minimal" which is less minimal than what you saw, but it's based on the original BSP, with more or less the same include file. See https://sourceforge.net/p/esp32-s3-86-box-project-base/code/HEAD/tree/trunk/minimal/ .

I see they currently support esp-idf 5.[23] but that took them a long time.

So I believe the pin numbers you saw are not the issue.

Regarding the init_cmds. I basically have several versions of such a table. There is a default one in the espressif__esp_lcd_gc9503 component that gets loaded if you don't specify anything. So I tried not specifying anything.
I also tried copying the one from the BSP, same result.

Bothersome is that almost all of the codes in those tables are missing in the gc9503 documentation, so I have no clue what they do. But I shouldn't complain to you about that.

dannybackx, 2024-10-08:
[image]
Screenshot of that "minimal"

Lzw655, 2024-10-10:
Yes, I understand. Did you try commenting out `ESP_ERROR_CHECK(esp_lcd_panel_reset(lcd_panel));` in the source at https://sourceforge.net/p/esp32-s3-86-box-project-base/code/HEAD/tree/trunk/touchscreen/ after replacing the init_cmds table?

dannybackx, 2024-10-13:
Whow, first time I am actually seeing something.
Committed the current code.

Thanks !!

I suppose the image being off by a number of pixels can be fixed with stuff in the GC9503_480_480_PANEL_TIMING macro ?

[image]

Lzw655, 2024-10-15:
Yes, I think so.

dannybackx, 2024-10-15:
Porting the changes to my real application was ridiculously easy.
Now looking for why the colours are off ;-)

[image]

espzav, 2025-03-13:
Hi @dannybackx, was it solved your issue? Can we close it?

dannybackx, 2025-03-18:
You'll see my last comment dates from October. I gave up on using these components a bit after that, I was just wasting my time on them instead of doing productive work.

Let me know if anything changed since that, I could test again if you want me to I suppose.

So : as far as I know, no.

dannybackx, 2025-07-06:
I tried all the things that were suggested, and more.
See https://sourceforge.net/p/esp32-s3-86-box-project-base/code/HEAD/tree/trunk/try/ for two examples, with pictures of the visuals they produce.
This is on the same hardware (see the dirty spots on the screen) so it's definitely software that causes it.

Including the picture with the wrong colours here for reference.

[image]

espzav, 2025-07-07:
Hi @dannybackx, I quickly look at your code, I see that you made copy of part LVGL port. This is not a good solution. 
Did you try to swap colors, as I recommend before? 
This line in LVGL port: https://github.com/espressif/esp-bsp/blob/872016e913efa7dd9194b30c28bdc8c434232dad/components/esp_lvgl_port/src/lvgl9/esp_lvgl_port_disp.c#L673C9-L673C31

dannybackx, 2025-07-07:
Hi,
Yes I did try that (in another example), produced even stranger colours.

I am a little puzzled by your "copy of LVGL port" remark. The 04 directory is a trimmed down version of managed_components/espressif__esp_lcd_st7701/examples/rgb_avoid_tearing  : 
```
victus: {33} pwd
/home/danny/src/esp-idf/ZX3D95CD01S/alarm/managed_components/espressif__esp_lcd_st7701/examples/rgb_avoid_tearing
victus: {34} ls -l main
total 56
-rw-rw-r-- 1 danny backx   218 Nov 13  2024 CMakeLists.txt
-rw-rw-r-- 1 danny backx 12013 Nov 13  2024 example_rgb_avoid_tearing.c
-rw-rw-r-- 1 danny backx   373 Nov 13  2024 idf_component.yml
-rw-rw-r-- 1 danny backx  4809 Nov 13  2024 Kconfig.projbuild
-rw-rw-r-- 1 danny backx 20055 Nov 13  2024 lvgl_port.c
-rw-rw-r-- 1 danny backx  5033 Nov 13  2024 lvgl_port.h
victus: {35} 
```

dannybackx, 2025-07-10:
See latest commit, I changed the 04 implementation to use the esp_lvgl_port component.
This doesn't bring improvement, visuals still wrong.

espzav, 2025-07-11:
> Yes I did try that (in another example), produced even stranger colours.

Ok, then colors can be bad because of the initial settings of the screen here: https://github.com/espressif/esp-bsp/blob/490942fb65ad8563836aa944255f358edfd0b180/components/lcd/esp_lcd_gc9503/esp_lcd_gc9503.c#L172

The settings can be different for the screens.

dannybackx, 2025-07-11:
Just copying the list from a BSP example into a esp_lcd_gc9503 source doesn't work - the display stays blank.

So how do I go about that ?

dannybackx, 2025-07-11:
That's what I was already trying, see the commit I made seconds ago.

dannybackx, 2025-07-11:
Aha, I've added two lines that relate to documented commands (most of these gc9503 related commands are not documented, or not anywhere I could find) at the end of my table, and this changes things for the better.

Let's see if I can make this work in my application as well as on the LVGL demo...

Viry426, 2025-07-14:
Baja la versión  del submodulo ESP_LCD_GC9503 a la versión 1.0.0,  eso me funcionó a mi

Saludos