espressif/esp-bsp #358: esp_lcd_touch: Add a track_id for esp_lcd_touch_get_coordinates

lijunru-hub, 2024-08-05:
### Related area

esp_lcd_touch

### Hardware specification

ALL Chip

### Is your feature request related to a problem?

In multi-touch systems, the track ID is a crucial piece of information. Some chips, such as the GT911, inherently support track ID information. We should ensure that the driver exposes this information.

### Describe the solution you'd like

like **bool esp_lcd_touch_get_coordinates(esp_lcd_touch_handle_t tp, uint16_t *x, uint16_t *y, uint16_t *strength, uint8_t *point_num, uint8_t max_point_num)**

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lijunru-hub, 2024-08-05:
cc@espzav

lijunru-hub, 2024-08-05:
In GT911's datasheet, it can be supported easily.
[image]

lijunru-hub, 2024-08-13:
FT5x06
[image]

tore-espressif, 2024-08-13:
@lijunru-hub sounds like a nice feature. We can discuss during next BSP meeting :) 
cc @espzav