espressif/esp-bsp #365: espressif/esp_lvgl_port keeps forcing LVGL v9.1 to be installed with ESP-IDF v5.3 on Windows

joulupukki, 2024-08-08:
### Board

ESP32-S3

### Hardware Description

Using a Heltec ESP32 WiFi Kit v3, but the build isn't making it far enough to be able to build successfully on Windows 10.

### IDE Name

VSCode

### Operating System

Windows 10

### Description

I'm trying to build an ESP-IDF v5.3 project. I'm using VS Code and have it working properly on Windows 11, and also my Mac. But, if I try to have a colleague build it using his Windows 10 machine (he's tried multiple) he gets this strange problem where the component manager will load in LVGL v9.1.

I believe it's happening because of espressif/esp_lvgl_port. If he temporarily comments out the espressif/esp_lgvl_port from our idf_component.yml file, the build system will properly pull in LVGL v8.4.0. But, as soon as we introduce esp_lvgl_port, it installs LVGL v9.1.

### Sketch

```cpp
Here's our idf_component.yml file:

dependencies:
  espressif/esp_lvgl_port: "^2.3.0"
  lvgl/lvgl: "^8.4.0"
  ## Required IDF version
  idf:
    version: ">=4.1.0"
```

### I have checked existing issues, README.md and ESP32 Forum

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

rpschultz13, 2024-08-08:
If I try:
  lvgl/lvgl:
    version: "^8"
  espressif/esp_lvgl_port: "^1"

This outputs: 
  espressif/esp_lvgl_port: 1.4.0
  lvgl/lvgl: 8.4.0

And it builds.

Would be nice to get lvgl/lvgl: 8.4.0 and espressif/esp_lvgl_port: 2.x

tore-espressif, 2024-08-09:
Hi guys,
this seems to be a bug in the component version solving. This should work just right
```
dependencies:
  espressif/esp_lvgl_port: "^2.3.0"
  lvgl/lvgl: "^8.4.0"
  ## Required IDF version
  idf:
    version: ">=4.1.0"
```

Could you try updating the idf-component-manager `pip install idf-component-manager --upgrade` and try again?

rpschultz13, 2024-08-09:
updating the component manager appeared to work. But it still forced v9.1 when I set Target.

tore-espressif, 2024-08-26:
Could you please provide your `idf_component.yml` of your `main` component? And also provide a full log of `idf.py reconfigure`?

tore-espressif, 2024-08-26:
I checked out your project and LVGL got correctly resolved to v8.4.0.

I tried with IDF component manager v1.5.3 and also latest v2.0.3.

Please let me know if there is anything else, or we can close this issue

joulupukki, 2024-08-26:
@tore-espressif what operating system are you using? It also works for me, but on Windows 10 it has not been working for @rpschultz13.

tore-espressif, 2024-08-26:
> @tore-espressif what operating system are you using? It also works for me on MacOS, but on Windows 10 it has not been working for @rpschultz13.

I'm on Windows 10. (64bit, version 22H2 build 19045.4780)

rpschultz13, 2024-08-26:
[idf_py_stdout_output_23624.txt](https://github.com/user-attachments/files/16749849/idf_py_stdout_output_23624.txt)

Is this what you're looking for? If not, please specify.
Thanks!

hfudev, 2024-08-29:
Hi @joulupukki, thanks for reporting the issue. I could reproduce the issue with idf-component-manager 1.5.3 with esp-idf 5.3.

Could you try to unset the environment variable `IDF_COMPONENT_STORAGE_URL` and run `idf.py build` again? This environment variable shall be set by our offline installer.

rpschultz13, 2024-08-29:
@hfudev , thanks for chiming in. I am willing, but lack the knowledge to do what you ask. Can you provide more detailed instructions.
Thanks!

hfudev, 2024-08-30:
no worries. @rpschultz13 here's a guide on how to set it. https://superuser.com/a/1528432/1924620.

To unset an environment variable, simply select it and click the "Delete" button.

Also, remember to restart your terminal or application (e.g., vscode) you're using after making changes to the environment variables -- otherwise the changes won't take effect.

rpschultz13, 2024-08-31:
Hey that worked. Thanks for the help!

joulupukki, 2024-08-31:
Glad that worked @rpschultz13 ! @hfudev is this something that needs to be fixed/adjusted in the IDF Component Manager or maybe one of the ESP-IDF-related VS Code extensions?

hfudev, 2024-09-03:
We’ve identified the cause and are actively working on a solution. A new release of the offline installer will be available soon, which will address this issue.

hfudev, 2024-09-26:
I'll close the issue. The offline installer issue has been fixed. Thanks for reporting :)

closed 2024-09-26 by espzav