espressif/esp-bsp #374: M5Stack CoreS3 and W5500 Error during network initialization

bgiovanni, 2024-08-16:
### Board

M5Stack CoreS3

### Hardware Description

M5Stack CoreS3 with W5500 PoE Base

### IDE Name

VSCode with PlatFormIO 6.1.16b1

### Operating System

Linux

### Description

W5500 share the the same SPI bus with the ILI9342C display controller and fails to initialize with the following log messages

```log
V bus_lock: dev 1 acquired.
V (2633) spi_master: polling trans
V (2633) spi_master: polling trans done
V (2633) bus_lock: dev 1 released.
V bus_lock: dev 1 acquired.
V (2653) spi_master: polling trans
V (2653) spi_master: polling trans done
V (2653) bus_lock: dev 1 released.
E (2663) w5500.mac: W5500 version mismatched, expected 0x04, got 0x00
E (2663) w5500.mac: emac_w5500_init(826): verify chip ID failed
I (2663) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (2673) esp_eth: esp_eth_driver_install(228): init mac failed
ESP_ERROR_CHECK failed: esp_err_t 0x10a (ESP_ERR_INVALID_VERSION) at 0x42002e37
  #0  0x42002e37 in eth_start at src/main.c:113 (discriminator 1)

file: "src/main.c" line 113
func: eth_start
expression: esp_eth_driver_install(&config, &s_eth_handle)

```

Error occurs while allocation the driver  ```s_mac = esp_eth_mac_new_w5500(&w5500_config, &mac_config);```.   
When the program starts without ```bsp_display_start();``` the network interface is working fine.  

It could be a SPI bus sharing conflict but I can't find any solution to fix the problem.

Link to the devices used.  
[M5Stack Core-S3](https://shop.m5stack.com/products/m5stack-cores3-esp32s3-lotdevelopment-kit)   
[Network Interface](https://shop.m5stack.com/products/lan-module-w5500-with-poe-v12)

### Sketch

```cpp
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_eth.h"
#include "esp_mac.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_netif.h"
#include "esp_log.h"

#include "driver/gpio.h"
#include "driver/spi_master.h"
#include "nvs_flash.h"

#include "bsp/esp-bsp.h"

#define TAG "M5CoreS3Test"
const char *appver = "0.0.5";

//Ethernet
static esp_eth_handle_t s_eth_handle = NULL;
static esp_eth_mac_t *s_mac = NULL;
static esp_eth_phy_t *s_phy = NULL;

static void eth_on_got_ip(void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data) {
    ip_event_got_ip_t *event = (ip_event_got_ip_t *)event_data;
    ESP_LOGI(TAG, "Got IPv4 event: Interface \"%s\" address: " IPSTR, esp_netif_get_desc(event->esp_netif), IP2STR(&event->ip_info.ip));
}

static void on_eth_event(void *esp_netif, esp_event_base_t event_base, int32_t event_id, void *event_data) {
    uint8_t mac_addr[6] = { 0 };
    esp_eth_handle_t _eth_handle = *(esp_eth_handle_t*) event_data;

    switch (event_id) {
    case ETHERNET_EVENT_CONNECTED:
        esp_eth_ioctl(_eth_handle, ETH_CMD_G_MAC_ADDR, mac_addr);
        ESP_LOGI(TAG, "Ethernet Link Up");
        ESP_LOGI(TAG, "Ethernet HW Addr %02x:%02x:%02x:%02x:%02x:%02x",
                 mac_addr[0], mac_addr[1], mac_addr[2], mac_addr[3], mac_addr[4], mac_addr[5]);
        break;
    case ETHERNET_EVENT_DISCONNECTED:
        ESP_LOGI(TAG, "Ethernet Link Down");
        break;
    case ETHERNET_EVENT_START:
        ESP_LOGI(TAG, "Ethernet Started");
        break;
    case ETHERNET_EVENT_STOP:
        ESP_LOGI(TAG, "Ethernet Stopped");
        break;
    default:
        ESP_LOGI(TAG, "Ethernet Unhandled event (%ld)", event_id);
        break;
    }
}

static void eth_start(void) {
    ESP_LOGI(TAG, "Starting Ethernet");

    // Initialize TCP/IP network interface (should be called only once in application)
    ESP_ERROR_CHECK(esp_netif_init());
    // Create default event loop that running in background
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    esp_netif_inherent_config_t esp_netif_config = ESP_NETIF_INHERENT_DEFAULT_ETH();
    esp_netif_config.if_desc = "eth0";
    esp_netif_config.if_key = "ETH_SPI_0";
    esp_netif_config.route_prio = 64;
    esp_netif_config_t netif_config = {
        .base = &esp_netif_config,
        .stack = ESP_NETIF_NETSTACK_DEFAULT_ETH
    };
    esp_netif_t *netif = esp_netif_new(&netif_config);
    assert(netif);

    eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();
    mac_config.rx_task_stack_size = 3072;
    eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
    phy_config.phy_addr = 1;
    phy_config.reset_gpio_num = GPIO_NUM_7;

    spi_device_interface_config_t spi_devcfg = {
        .command_bits = 0,
        .address_bits = 0,
        .mode = 0,
        .clock_speed_hz = 20 * 1000 * 1000,
        .spics_io_num = GPIO_NUM_9,
        .queue_size = 20,
        .pre_cb = NULL,
        .post_cb = NULL
    };

    ESP_LOGI(TAG, "Initializing w5500 driver");
    eth_w5500_config_t w5500_config = ETH_W5500_DEFAULT_CONFIG(BSP_LCD_SPI_NUM, &spi_devcfg);
    w5500_config.int_gpio_num = GPIO_NUM_14;
    s_mac = esp_eth_mac_new_w5500(&w5500_config, &mac_config);
    s_phy = esp_eth_phy_new_w5500(&phy_config);

    vTaskDelay(pdMS_TO_TICKS(10));

    ESP_LOGI(TAG, "Installing w5500 driver");
    esp_eth_config_t config = ETH_DEFAULT_CONFIG(s_mac, s_phy);
    ESP_ERROR_CHECK(esp_eth_driver_install(&config, &s_eth_handle));

    uint8_t eth_mac[6] = {0};
    ESP_ERROR_CHECK(esp_read_mac(eth_mac, ESP_MAC_ETH));
    ESP_ERROR_CHECK(esp_eth_ioctl(s_eth_handle, ETH_CMD_S_MAC_ADDR, eth_mac));

    // attach Ethernet driver to TCP/IP stack
    ESP_ERROR_CHECK(esp_netif_attach(netif, esp_eth_new_netif_glue(s_eth_handle)));

    ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ETHERNET_EVENT_CONNECTED, &on_eth_event, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &eth_on_got_ip, NULL));

    esp_eth_start(s_eth_handle);
}

void init_spi() {
    spi_bus_config_t buscfg = {
        .mosi_io_num = GPIO_NUM_37,
        .miso_io_num = GPIO_NUM_35,
        .sclk_io_num = GPIO_NUM_36,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
    };

    // Initialize the SPI bus
    esp_err_t ret = spi_bus_initialize(BSP_LCD_SPI_NUM, &buscfg, SPI_DMA_CH_AUTO);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "Failed to initialize bus: %s", esp_err_to_name(ret));
        return;
    }

    ESP_LOGI(TAG, "SPI bus initialized successfully");
}

void app_main() {
    gpio_install_isr_service(0);

	esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    bsp_display_start();    
    //init_spi();

    eth_start();

    int reboot = 2000;
    while (1) {
        vTaskDelay(pdMS_TO_TICKS(10));
        
        if (reboot == 0) {
            esp_restart();
        } else {
            reboot--;
        }
    }
}
```

### I have checked existing issues, README.md and ESP32 Forum

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

diplfranzhoepfinger, 2024-08-21:
i need this too

diplfranzhoepfinger, 2024-08-21:
I experience the following: 

using the Demo of 

[https://github.com/espressif/esp-bsp/tree/master/examples/display](https://github.com/espressif/esp-bsp/tree/master/examples/display)

this runs when ONLY the USB is plugged. 

as soon as i plug in the Power from the W5500 the Demo stops and the Screen turns dark.

diplfranzhoepfinger, 2024-08-21:
there is one way to keep the Demo Running: 

1.  plug in Base Power
2.  plug in USB Power.

then it runs. 

I think to make this fully stable, AXP2101 Chip must be used to comunicate with i2c with the ESP32.

bgiovanni, 2024-08-21:
@diplfranzhoepfinger it's the correct way to work.
Check the  [Power Management Section](https://docs.m5stack.com/en/core/CoreS3). 

The bsp driver does not fully control the AXP2101 and set BUS_OUT and USB_OTG to 0. 
As you can see, in the power schema, the USB Power it's not sent to the bus.  
We need to wait the fully support of the AXP2101 to be able to change BUS_OUT or USB_OTG.

diplfranzhoepfinger, 2024-08-21:
i tested: 

i get different Errors: 

code:
https://gist.github.com/diplfranzhoepfinger/66b0cfa120cbc7f03e7be73b096e081c

log: 
https://gist.github.com/diplfranzhoepfinger/5605114bb738d2bd23052ab21f150981

diplfranzhoepfinger, 2024-08-21:
I am one step forward:

diplfranzhoepfinger, 2024-08-21:
[https://gist.github.com/diplfranzhoepfinger/ec0a67db28f43dab1a36b41bcffd2138](https://gist.github.com/diplfranzhoepfinger/ec0a67db28f43dab1a36b41bcffd2138)

diplfranzhoepfinger, 2025-01-31:
@igrr any Progress here ? 
it would be really cool to use Ethernet plus LCD ... 

lot of possible usecases. 
and a nice Hardware from housing / Certification point of view.