espressif/esp-bsp #377: W (1852) i2s_common: GPIO 33 is not usable, maybe conflict with othe

diplfranzhoepfinger, 2024-08-21:
### Board

M5Stack CoreS3

### Hardware Description

M5Stack CoreS3

### IDE Name

Espressif-IDE

### Operating System

Linux

### Description

https://github.com/espressif/esp-bsp/tree/master/examples/display_audio_photo

Log: 

```
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1918
load:0x403c8700,len:0x4
load:0x403c8704,len:0xe5c
load:0x403cb700,len:0x303c
entry 0x403c893c
I (27) boot: ESP-IDF v5.3-dirty 2nd stage bootloader
I (27) boot: compile time Aug 21 2024 20:53:14
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : QIO
I (49) boot.esp32s3: SPI Flash Size : 16MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot:  3 storage          Unknown data     01 82 00110000 002f0000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=3c070020 size=17e30h ( 97840) map
I (128) esp_image: segment 1: paddr=00027e58 vaddr=3fc97500 size=02e10h ( 11792) load
I (130) esp_image: segment 2: paddr=0002ac70 vaddr=40374000 size=053a8h ( 21416) load
I (137) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=626ach (403116) map
I (203) esp_image: segment 4: paddr=000926d4 vaddr=403793a8 size=0e0ach ( 57516) load
I (222) boot: Loaded app from partition at offset 0x10000
I (222) boot: Disabling RNG early entropy source...
I (234) esp_psram: Found 8MB PSRAM device
I (234) esp_psram: Speed: 80MHz
I (234) cpu_start: Multicore app
I (1153) esp_psram: SPI SRAM memory test OK
I (1162) cpu_start: Pro cpu start user code
I (1162) cpu_start: cpu freq: 240000000 Hz
I (1162) app_init: Application information:
I (1165) app_init: Project name:     display_audio_photo
I (1171) app_init: App version:      v0.15-112-g0630307-dirty
I (1177) app_init: Compile time:     Aug 21 2024 21:08:16
I (1183) app_init: ELF file SHA256:  5cd2b3773...
I (1189) app_init: ESP-IDF:          v5.3-dirty
I (1194) efuse_init: Min chip rev:     v0.0
I (1199) efuse_init: Max chip rev:     v0.99 
I (1204) efuse_init: Chip rev:         v0.2
I (1209) heap_init: Initializing. RAM available for dynamic allocation:
I (1216) heap_init: At 3FC9B0C8 len 0004E648 (313 KiB): RAM
I (1222) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (1228) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (1235) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (1241) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (1249) spi_flash: detected chip: generic
I (1253) spi_flash: flash io: qio
W (1257) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (1268) sleep: Configure to isolate all GPIO pins in sleep state
I (1275) sleep: Enable automatic switching of GPIO sleep configuration
I (1282) main_task: Started on CPU0
I (1292) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1292) main_task: Calling app_main()
I (1462) M5Stack: Partition size: total: 2293769, used: 1807706
I (1512) LVGL: Starting LVGL task
I (1512) gpio: GPIO[35]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1512) ili9341: LCD panel create success, version: 1.2.0
I (1672) M5Stack: Setting LCD backlight: 50%
W (1852) i2s_common: GPIO 33 is not usable, maybe conflict with othe
```

### Sketch

```cpp
-
```

### Other Steps to Reproduce

-

### I have checked existing issues, README.md and ESP32 Forum

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

diplfranzhoepfinger, 2024-08-21:
if i change the bsp\_espbox\_disp\_example.c in a Way that i change from: 

```c
    /* Initialize audio */
    app_audio_init();
```

to 

```c
    /* Initialize audio */
    //app_audio_init();
```

then the Example runs but no Sound.

tore-espressif, 2024-08-26:
The GPIO reservation feature is a rather new one in ESP-IDF so this can be red herring. @L-KAYA Could you please check why this warning is displayed when GPIOs 33 and 34 are exclusively used by I2S in this example? Also the error message is very confusing...

@diplfranzhoepfinger Audio playback works OK on my board without any changes in the code.

The recording path is distorted, I'll have a look soon

tore-espressif, 2024-09-12:
This warning `i2s_common: GPIO 33 is not usable, maybe conflict with othe` is mistakenly printed, it will be fixed soon.

I still need to check the recording path distortion

davidthings, 2024-09-16:
I have been working with another ESP32-S3 board and have just been wrangling my own GPIO33 issue. 

The problem is that with large PSRAM's (8MB), the S3 uses a wide data path.  GPIO33 is one of the lines needed (D4 I think).  So either you use PSRAM and avoid the alternate use, or you disable PSRAM and use the alternative function.  One or the other.

tore-espressif, 2024-09-17:
> I have been working with another ESP32-S3 board and have just been wrangling my own GPIO33 issue.
>
> The problem is that with large PSRAM's (8MB), the S3 uses a wide data path. GPIO33 is one of the [...]

Correct, for Octal PSRAM, GPIO33 is reserved for D4 signal to PSRAM.

M5Stack CoreS3 has Quad PSRAM, so the pin is usable

davidthings, 2024-09-17:
Excellent!  Thanks.

espzav, 2025-09-10:
It seems, that this issue is solved. Closing.
Feel free to reopen it.

closed 2025-09-10 by espzav

tore-espressif, 2025-11-26:
> I still need to check the recording path distortion

Here is an update about the reported noise in M5Stack-CoreS3 board from our audio team:

The ES7210 noise floor is within specs and is good-enough for voice applications. For environmental noise suppression, the NS module from the [esp-sr](https://github.com/espressif/esp-sr) repo can be used.

Here are results of our measurements:

[image]