espressif/esp-bsp #378: esp32_p4_function_ev_board Display Type Support？

Y1hsiaochunnn, 2024-08-23:
### Related area

BSP supports display types

### Hardware specification

Support esp32_p4_function_ev_board v1,4 or V1.4 or newer if available

### Is your feature request related to a problem?

I have tried to use esp32_p4_function_ev_board to support other displays but not ek79007 and ili9881c.   Can I add it to  the default support of esp32_p4_function_ev_board component?    I can make it into a Component and push it to the ESP component Registry.  so that the esp32_p4_function_ev_board component can be directly invoked?  I will always maintain these component projects.  If you need samples of these screens, I think we'd be happy to provide them free of charge

### Describe the solution you'd like

It is hoped that in future esp32-p4 demo will be added to support screen types other than ek79007 or ili9881c. For example, optional :ek79007, ili9881c, xxx display

### Describe alternatives you've considered

If that doesn't work, I'll fork off a new component package to maintain other third-party screens, such as esp32_p4_function_ev_board_pro?

### Additional context

[image]

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

igrr, 2024-08-23:
Hi @Y1hsiaochunnn,

For the time being, this repo supports only development boards sold by Espressif (and now also M5Stack). Since these two types of display controllers are the one which Espressif currently uses, the component only supports them.

If you'd like to add support for custom hardware, you are most welcome to publish your component in the Registry. If you run into any issue, please feel free to ask here or at esp32.com.

If there is sufficient user demand, we can revisit this question and try to make it easier to support custom combinations of P4 EV board BSP with other displays.

Y1hsiaochunnn, 2024-08-23:
Hi @igrr,
Thanks for your reply, I will re-maintain a new component to support other displays,Please ignore this problem, I just want to make it easier for the user

closed 2024-08-23 by Y1hsiaochunnn