espressif/esp-bsp #391: 能否更新I2C驱动？

AaronPMM, 2024-09-12:
### Related area

Driver

### Hardware specification

ESP32-S3-BOX-3

### Is your feature request related to a problem?

我在 ESP-ADF 中使用 BSP 中的 ESO-BOX-3，其中的 I2C 都使用的都是旧的驱动，包括示例也是，能否更新新的I2C驱动？

### Describe the solution you'd like

在 ESP-ADF 中调用初始化接口函数

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

ammaree, 2024-09-25:
@espressif2022 thank for the response, but not sure I understand.

Will I2C v2 driver functionality be added to all the devboard BSP's?
If so, when can we expect to have this?

-or-

Will I2C v2 driver functionality be added specifically to the ESP32-S3-BOX-3 BSP, as this issue requests?
If so then of no value to us since we have replaced the ESP32-S3-BOX with the ESP32-S3-LCD-EV-BOARD[2]

espzav, 2024-09-26:
Hello @ammaree,
we are planning add support I2C v2 driver to all supported BSPs.

ammaree, 2024-10-08:
Hi @espzav 

Thanks for the response, and sorry to bother again, but just trying to get clarity on specifics.

Our order of priority would be:
1. ESP32-S3-LCD-EV-BOARD (800x480 and 480x480)
2. M5Stack CoreS3 (not M5Stack Core2)
3. ESP32-S3-BOX (not ESP32-S3-BOX-3)
4. ESP32-WROVER-KIT

We understand and accept that your priorities might be different but would really appreciate if you can:
a) give an indication your internal top 3 or 4 board priorities; and
b) an indication of the estimated timelines for those top priorities.

Thanks in advance, the info will help a lot.

tore-espressif, 2024-10-29:
Update
* IO expander I2C driver https://github.com/espressif/esp-bsp/pull/424
* esp-s3-box-3 https://github.com/espressif/esp-bsp/pull/414

ManelCS, 2024-11-27:
Has the ESP32-S3-LCD-EV-BOARD been updated to support I2C version 2?

tore-espressif, 2024-11-27:
> Has the ESP32-S3-LCD-EV-BOARD been updated to support I2C version 2?

Not yet, it is the next one on our list

tore-espressif, 2024-11-29:
@ManelCS I2C version 2 is supported in esp32_s3_lcd_ev_board v3
https://components.espressif.com/components/espressif/esp32_s3_lcd_ev_board/versions/3.0.0

closed 2024-12-11 by tore-espressif