espressif/esp-bsp #395: ST7735S esp_lcd support

TeriyakiGod, 2024-09-21:
### Related area

Display driver

### Hardware specification

ESP32

### Is your feature request related to a problem?

I want to use a popular 128x128 tft display st7735s driver with esp_lcd and lvgl but can't because its not supported.

### Describe the solution you'd like

It would solve my problem if there was an esp_lcd_st7735 component.

### Describe alternatives you've considered

I managed to run the display using TFT eSPI library and arduino for esp32, but i want to use esp idf.

### Additional context

This can be implemented by modifying the esp_lcd_st7796 component.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

espzav, 2024-09-23:
Hello @TeriyakiGod,
we are supporting mainly LCD drivers, which are on Espressif's or M5Stack hardware. You can use any LCD driver from [IDF Component Registry](https://components.espressif.com/components?q=esp_lcd) or add your driver to IDF Component Registry too. I am sorry.

TeriyakiGod, 2024-09-23:
It's okay I found this:
https://github.com/lvgl/lvgl_esp32_drivers

closed 2024-09-23 by TeriyakiGod