espressif/esp-bsp #398: Garbled output on esps2 mini with OLED

mado89, 2024-09-26:
### Board

ESP32 S2 Mini

### Hardware Description

ESP32 S2 Mini board with an OLED display 
[image]

### IDE Name

VSCode

### Operating System

Ubuntu

### Description

When using the i2c_oled_example from the examples directory the display does not show a text but only moving pixels as shown in the attached image.
[image]

### Sketch

```cpp
I2C_oled_example_main.c from the examples directory

idf_component.yml:
## IDF Component Manager Manifest File
dependencies:
  lvgl/lvgl: "^9.2.0"
  espressif/esp_lvgl_port: "^2.3.0"
  esp_lcd_sh1107: "^1"
  ## Required IDF version
  idf:
    version: ">=4.1.0"
  # # Put list of dependencies here
  # # For components maintained by Espressif:
  # component: "~1.0.0"
  # # For 3rd party components:
  # username/component: ">=1.0.0,<2.0.0"
  # username2/component2:
  #   version: "~1.0.0"
  #   # For transient dependencies `public` flag can be set.
  #   # `public` flag doesn't have an effect dependencies of the `main` component.
  #   # All dependencies of `main` are public by default.
  #   public: true
```

### I have checked existing issues, README.md and ESP32 Forum

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

espzav, 2024-09-26:
Hello @mado89, This example was mainly for LVGL8. We found the issue, when it is used with LVGL9. We have pull request here, which should solve the issue: https://github.com/espressif/esp-bsp/pull/392 There is readme, how to do it.

mado89, 2024-09-26:
@espzav I will give this example another try with a clean source tree and report back

mado89, 2024-09-26:
@espzav Happy to report that it now works.

espzav, 2024-10-01:
@mado89 Yes, you can close the issue.

closed 2024-10-01 by mado89