espressif/esp-bsp #400: esp_lvgl_port on ESP32-P4 with ILI9881C display

dmartauz, 2024-09-27:
### Board

ESP32-P4-Function-EV-Board

### Hardware Description

ILI9881C touchscreen display (8" 800 x 1280 pixels)

### IDE Name

esp-idf/master

### Operating System

Windows 10

### Description

The display natively works in portrait orientation and it is displaying simple tabview. In this orientation it works well.

[image]

Because I want to use it in landscape orientation and it seems that ILI9881C does not support hardware rotation I am attempting to use `.sw_rotate = true` and after uncommenting last line of the code it seems that first rotated frame is drawn correctly but following frames have some offset (shifted LVGL FPS tracker).

[image]

After clicking on the tabview also tabview buttons get shifted.

[image]

Is master branch of esp_lvgl_port already using PPA in case of ESP32-P4?

### Sketch

```cpp
const lvgl_port_display_cfg_t disp_cfg = {
        .io_handle = mipi_dbi_io,
        .panel_handle = mipi_dpi_panel,
        .buffer_size = EXAMPLE_MIPI_DSI_LCD_V_RES * EXAMPLE_MIPI_DSI_LCD_H_RES, //EXAMPLE_LVGL_DRAW_BUF_LINES
        .double_buffer = true,
        .hres = EXAMPLE_MIPI_DSI_LCD_H_RES,
        .vres = EXAMPLE_MIPI_DSI_LCD_V_RES,
        .monochrome = false,
        .color_format = LV_COLOR_FORMAT_RGB565,
        .rotation = {
            .swap_xy = false,
            .mirror_x = false,
            .mirror_y = false,
        },
        .flags = {
            .buff_dma = false,
            .buff_spiram = true,
            .sw_rotate = true,
            .swap_bytes = false,
            .full_refresh = false,
            .direct_mode = false,
        }
    };

    const lvgl_port_display_dsi_cfg_t disp_dsi_cfg;

    lvgl_display = lvgl_port_add_disp_dsi(&disp_cfg, &disp_dsi_cfg);
    //lv_display_set_rotation(lvgl_display, LV_DISPLAY_ROTATION_270);
```

### I have checked existing issues, README.md and ESP32 Forum

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

espzav, 2024-09-30:
Hi @dmartauz thank you for reporting this issue. I will check it soon and fix it. The master branch is not using PPA yet, because after performance measurement, the SW rotation was faster than PPA because we are rotationg only small pieces of the whole screen. We are still working on improving performance and looking for the best solution. The PPA rotation is used in this PR: https://github.com/espressif/esp-bsp/pull/352

dmartauz, 2024-10-06:
@espzav any update on the fix? In the meantime I am porting my application from S3 with 800x480 parallel display.

espzav, 2024-10-24:
Hello @dmartauz, sorry for delay. I tried in on some screens and examples. In some cases, I see the issue and in some cases it was fixed by this change in LVGL port component:
[image]
Please, could you try it? I need more time for investigation and check all cases.

espzav, 2024-10-24:
@dmartauz thank you for test it. I am happy that it is working. I will do more tests and I will release it soon. 
The SW rotation is slower because there is one additional buffer and copying. The PPA should be faster, when there will be redraw of whole screen, not for parts. But there should be used estarnal RAM instead of internal and it is slower too. 
We are working on speed up and more usage of PPA. You can try rotation with PPA in this PR: https://github.com/espressif/esp-bsp/pull/352 but it is still slower.

dmartauz, 2024-10-24:
@espzav I have double buffer with size of V*H resolution (1280x800px) and `.trans_size` set to 1/10 of the main buffer. I like this setup as from users perspective the screen is redrawn at once. In this configuration using PPA on the whole screen buffer might make sense, right? PPA rotation in the PR would rotate the screen at once in my case?

espzav, 2024-10-24:
@dmartauz Oh, have you got buffer in SPIRAM (`.buff_spiram = true`)? The PPA can help you for full buffer, but still it will be called in every flush and it depends on LVGL display settings (e.g. https://github.com/espressif/esp-bsp/blob/c90025b5198c1d0c392cfd16e837a681a31d68d6/components/esp_lvgl_port/include/esp_lvgl_port_disp.h#L64). But I think, if you set buffer to smaller and try to set `.buff_dma = true` and `.buff_spiram = false`, I think it can be faster. 
For example we are using this size: `1280*50` in 1280x720 in internal RAM (not double buffer).

dmartauz, 2024-10-24:
@espzav I have `.buff_spiram = false` configured but I guess it must allocate the buffer in SPIRAM internally as for 1280x800px and RGB565 the buffer size needed is 1280x800x2bytes = 2048000bytes, correct?

dmartauz, 2024-10-24:
@espzav In case I set `.buff_dma = true` then I get "Not enough memory for LVGL buffer allocation" either for buf1 or for rotation buffer. My application probably already consumes most of internal SRAM.

espzav, 2024-10-24:
@dmartauz Oh yes, the DMA capable internal RAM is smaller. Then you can try to set smaller buffer and use DMA. It should be still faster.

dmartauz, 2024-10-24:
@espzav rotation buffer is allocated with the same size as primary buffer?

espzav, 2024-10-29:
@dmartauz yes

dmartauz, 2024-11-06:
Yes. I did full clean of my project and tested with 2.4.2. Rotation works as expected.

closed 2024-11-07 by espzav