espressif/esp-bsp #403: Programming with ESP-IDF results in invalid handle for TCA9554 driver

karunt, 2024-10-10:
### Board

Qualia ESP32-S3

### Hardware Description

Qualia ESP32-S3 board, equipped with PCA9554 I/O extender, connected to 3.4" 480x480 RGB565 TFT (no touchscreen) LCD (https://www.adafruit.com/product/5825#description), driven by ST7701 panel driver.

### IDE Name

VSCode with Espressif-IDF extension

### Operating System

Windows 11

### Description

Begin code with trying to generate handle for TCA9554 extender. But code crashes and reboots after producing invalid handle error. Following errors are produced:

> E (416) i2c: i2c_master_cmd_begin(1478): i2c driver not installed
> E (426) tca9554: write_direction_reg(133): Write direction reg failed
> E (436) tca9554: reset(151): Write dir reg failed
> E (436) tca9554: esp_io_expander_new_i2c_tca9554(81): Reset failed
> E (446) io_expander: esp_io_expander_set_dir(35): Invalid handle
> E (446) io_expander: esp_io_expander_set_level(63): Invalid handle

Why do I get an `invalid handle` error for the I/O expander?

### Sketch

```cpp
#define ESP_IO_EXPANDER_I2C_TCA9554_ADDRESS_NEW    (0x3F)
#define EXAMPLE_LCD_IO_RGB_VSYNC        42
#define EXAMPLE_LCD_IO_RGB_HSYNC        41
#define EXAMPLE_LCD_IO_RGB_DE           2
#define EXAMPLE_LCD_IO_RGB_PCLK         1
#define EXAMPLE_LCD_IO_RGB_DATA0        40
#define EXAMPLE_LCD_IO_RGB_DATA1        39
#define EXAMPLE_LCD_IO_RGB_DATA2        38
#define EXAMPLE_LCD_IO_RGB_DATA3        0
#define EXAMPLE_LCD_IO_RGB_DATA4        45
#define EXAMPLE_LCD_IO_RGB_DATA5        48
#define EXAMPLE_LCD_IO_RGB_DATA6        47
#define EXAMPLE_LCD_IO_RGB_DATA7        21
#define EXAMPLE_LCD_IO_RGB_DATA8        14
#define EXAMPLE_LCD_IO_RGB_DATA9        13
#define EXAMPLE_LCD_IO_RGB_DATA10       12
#define EXAMPLE_LCD_IO_RGB_DATA11       11
#define EXAMPLE_LCD_IO_RGB_DATA12       10
#define EXAMPLE_LCD_IO_RGB_DATA13       9
#define EXAMPLE_LCD_IO_RGB_DATA14       46
#define EXAMPLE_LCD_IO_RGB_DATA15       3

#define EXAMPLE_LCD_IO_SPI_CS           (IO_EXPANDER_PIN_NUM_5)
#define EXAMPLE_LCD_IO_SPI_SCL          (IO_EXPANDER_PIN_NUM_4)
#define EXAMPLE_LCD_IO_SPI_SDA          (IO_EXPANDER_PIN_NUM_12)
#define EXAMPLE_LCD_IO_RST              (IO_EXPANDER_PIN_NUM_6)
#define EXAMPLE_PIN_NUM_BK_LIGHT        (IO_EXPANDER_PIN_NUM_9)
#define EXAMPLE_LCD_BK_LIGHT_ON_LEVEL   1
#define EXAMPLE_LCD_BK_LIGHT_OFF_LEVEL  !EXAMPLE_LCD_BK_LIGHT_ON_LEVEL

static const char *TAG = "display_example";

IRAM_ATTR static bool rgb_lcd_on_vsync_event(esp_lcd_panel_handle_t panel, const esp_lcd_rgb_panel_event_data_t *edata, void *user_ctx)
{
    return lvgl_port_notify_rgb_vsync();
}

void display_start()
{
  esp_io_expander_handle_t io_expander_tft = NULL;
  esp_io_expander_new_i2c_tca9554(I2C_NUM_0, ESP_IO_EXPANDER_I2C_TCA9554_ADDRESS_NEW, &io_expander_tft);
  esp_io_expander_set_dir(io_expander_tft, IO_EXPANDER_PIN_NUM_0 | IO_EXPANDER_PIN_NUM_1, IO_EXPANDER_OUTPUT);
  esp_io_expander_set_level(io_expander_tft, IO_EXPANDER_PIN_NUM_0 | IO_EXPANDER_PIN_NUM_1, 0);

//More code follows, but upon flashing, code fails to go past the esp_io_expander_set_level function above
}
```

### I have checked existing issues, README.md and ESP32 Forum

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

tore-espressif, 2024-10-10:
The first line of your log says

> E (416) i2c: i2c_master_cmd_begin(1478): i2c driver not installed

have you installed the I2C driver with `i2c_driver_install()`

karunt, 2024-10-11:
Thanks for catching that. I have now done that and the code is moving forward, but still aborting at a different point, probably unrelated to this driver. But another question - the readme document calls for setting pins 0 and 1 direction to output and levels to low. What's the purpose for doing that?

tore-espressif, 2024-10-29:
@karunt The settings you can see in README are just example use. In your application, of course, you must set it according to your needs :)

I'll close this issues as the original problem was solved. Feel free to reopen if you have more questions

closed 2024-10-29 by tore-espressif